rzeldent/esp32cam-rtsp #1: Failing to initialize....

ThiagoFacchini, 2022-09-06:
Hi,
First of all would like to thank you for the share this project.

I'm having this issue and I'm not really sure how to solve it.
Long story short, everything went well, from upload to configuration... but I couldn't see the video stream.

So I decided to jump into the monitor to watch it initialising...

I tried the 3 different camera configs, and all of them errored similarly.

AI THINKER:
```
21:52:24.896 > E (47879) cam_hal: cam_dma_config(293): frame buffer malloc failed
21:52:24.900 > E (47879) cam_hal: cam_config(377): cam_dma_config failed
21:52:24.904 > E (47879) camera: Camera config failed with error 0xffffffff
21:52:24.912 > [ 34724][E][main.cpp:142] start_rtsp_server(): Failed to initialize camera. Type: AI THINKER, frame size: XGA (1024x768), frame rate: 20 ms, jpeg quality: 12

```

ESP32CAM:
```
21:55:48.740 > E (47607) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
21:55:48.744 > [ 34448][E][main.cpp:142] start_rtsp_server(): Failed to initialize camera. Type: ESP32CAM, frame size: XGA (1024x768), frame rate: 20 ms, jpeg quality: 12
```

TTGO T-CAM:
```
21:57:35.888 > E (48561) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
21:57:35.896 > [ 35401][E][main.cpp:142] start_rtsp_server(): Failed to initialize camera. Type: TTGO T-CAM, frame size: XGA (1024x768), frame rate: 20 ms, jpeg quality: 12
```

So I had a look on the module to see if I could identify what type of module I had, but found it:
[image]

Any idea of what is going wrong?

Thanks!
TF

ThiagoFacchini, 2022-09-06:
Forgot to mention: 

I previously tested this camera, with an Arduino sketch and it works.

rzeldent, 2022-09-06:
Do you have the settings for the camera from that sketch? Are they
different?

ThiagoFacchini, 2022-09-06:
Yes I do, which settings should I look into?

ThiagoFacchini, 2022-09-06:
These are the PIN configs I'm using with the arduino sketch:

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

config.xclk_freq_hz = 20000000;
config.pixel_format = PIXFORMAT_JPEG;
config.frame_size = FRAMESIZE_SVGA;
config.jpeg_quality = 12; 
config.fb_count = 2; 

I basically used this sketch: https://github.com/bnbe-club/rtsp-video-streamer-diy-14/tree/master/diy-e14

ThiagoFacchini, 2022-09-07:
Any ideas?

rzeldent, 2022-09-07:
No not yet. have to look into this issue but have to find the time...

ThiagoFacchini, 2022-09-07:
Just a little extra information:
Uploading the Arduino Sketch I mentioned above, I can confirm the camera works just fine, which leads me to believe it's something with the code.

rzeldent, 2022-09-07:
Hi,

The config matches AI_THINKER and looks okay.... I Googled and it might
have something to do with the psram. See the following thread: entry
0x400805f0 E (134) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (134) cam_hal: cam_config(364): cam_dma_config failed · Issue #6209 ·
espressif/arduino-esp32 (github.com)
<https://github.com/espressif/arduino-esp32/issues/6209>.
I made a new branch feature/enable_psram and updated the compiler settings,
This looks like it might be PlatformIo specific, the core library that is
used.

I cannot reproduce it here but could you compile this branch and test if
this fixes the issue?

Kind regards,

Rene

ThiagoFacchini, 2022-09-09:
Hi Rene, I can confirm that it was the issue....
I couple days ago I found the solution....

Here's my platformio.ini file

`platform = espressif32
board = esp-wrover-kit
framework = arduino
monitor_speed = 115200
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = huge_app.csv
upload_speed = 115200
monitor_filters = 
	log2file
	time
	default
build_flags = 
	-O2
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
	-D LED_BUILTIN=4
	-D BOARD_HAS_PSRAM
lib_deps = 
	prampec/IotWebConf @ ^3.2.1
	geeksville/Micro-RTSP @ ^0.1.6
`

Hopefully it will help others.

closed 2022-09-09 by ThiagoFacchini

rzeldent, 2022-09-09:
Hi Thiago,

Thanks for testing; for some reason I was unable to reproduce this issue.
I'll merge the fix back!

Kind regards,

Rene

psi-4ward, 2023-02-28:
I've also problems.

1. `pio run -t erase` and  `pio run -t upload`
2. reboot with released gpio0
3. connecting to the AP
  * It says PSRAM=4MB; Board type=AI THINKER; **Camera was initialized successfully!**
  * Setup new wifi settings:
```
12:44:29.860 > [ 67085][V][main.cpp:329] on_config_saved(): on_config_saved
12:44:29.860 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
12:44:29.860 >
12:44:29.860 > Core  1 register dump:
12:44:29.860 > PC      : 0x400d5ffd  PS      : 0x00060530  A0      : 0x800d6176  A1      : 0x3ffb2530
12:44:29.904 > A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc6c74  A5      : 0x00000010
12:44:29.904 > A6      : 0x3ffc6c74  A7      : 0x00000019  A8      : 0x800d5ff6  A9      : 0x3ffc6238
12:44:29.904 > A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000007  A13     : 0x0000003f
12:44:29.904 > A14     : 0x3f40059f  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
12:44:29.904 > EXCVADDR: 0x00000048  LBEG    : 0x4008c701  LEND    : 0x4008c711  LCOUNT  : 0xfffffffc
12:44:29.951 >
12:44:29.951 >
12:44:29.951 > Backtrace: 0x400d5ffa:0x3ffb2530 0x400d6173:0x3ffb2550 0x40179015:0x3ffb2570 0x400dfc1a:0x3ffb2590 0x400e3efa:0x3ffb25b0 0x400e402b:0x3ffb25f0 0x400d46eb:0x3ffb2650 0x400dfc1a:0x3ffb2680 0x400dfcc1:0x3ffb26a0 0x400dfda7:0x3ffb26e0 0x400dffe6:0x3ffb2750 0x401797f5:0x3ffb27c0 0x400e4a21:0x3ffb27e0 0x400d6972:0x3ffb2800 0x400ea059:0x3ffb2820
12:44:29.951 >
12:44:29.951 >
12:44:29.951 >
12:44:29.951 >
12:44:29.951 > ELF file SHA256: 58451ce837d94057
12:44:29.951 >
12:44:30.129 > Rebooting...

...
12:44:42.596 > WiFi connected
...
12:44:55.810 > E (36867) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
12:44:55.811 > [ 25300][E][main.cpp:300] start_rtsp_server(): Failed to initialize camera: 0x105. Type: AI THINKER, frame size: VGA (640x480), frame rate: 100 ms, jpeg quality: 12
```

Now I open the webui-settings and change the type to `ESP32CAM`:

```
12:47:31.292 > [180805][V][main.cpp:329] on_config_saved(): on_config_saved
12:47:31.292 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
12:47:31.292 >
12:47:31.292 > Core  1 register dump:
12:47:31.335 > PC      : 0x400d5ffd  PS      : 0x00060530  A0      : 0x800d6176  A1      : 0x3ffb2530
12:47:31.335 > A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc6c74  A5      : 0x00000010
12:47:31.335 > A6      : 0x3ffc6c74  A7      : 0x00000019  A8      : 0x800d5ff6  A9      : 0x3ffc6238
12:47:31.335 > A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000007  A13     : 0x0000003f
12:47:31.335 > A14     : 0x3f40059f  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
12:47:31.335 > EXCVADDR: 0x00000048  LBEG    : 0x4008c701  LEND    : 0x4008c711  LCOUNT  : 0xfffffffc
12:47:31.388 >
12:47:31.388 >
12:47:31.388 > Backtrace: 0x400d5ffa:0x3ffb2530 0x400d6173:0x3ffb2550 0x40179015:0x3ffb2570 0x400dfc1a:0x3ffb2590 0x400e3efa:0x3ffb25b0 0x400e402b:0x3ffb25f0 0x400d46eb:0x3ffb2650 0x400dfc1a:0x3ffb2680 0x400dfcc1:0x3ffb26a0 0x400dfda7:0x3ffb26e0 0x400dffe6:0x3ffb2750 0x401797f5:0x3ffb27c0 0x400e4a47:0x3ffb27e0 0x400d6972:0x3ffb2800 0x400ea059:0x3ffb2820
12:47:31.388 >
12:47:31.388 >
12:47:31.388 >
12:47:31.388 >
12:47:31.388 > ELF file SHA256: 58451ce837d94057
12:47:31.388 >
12:47:31.585 > Rebooting..

...
12:47:46.638 > E (26291) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
12:47:46.670 > [ 14704][E][main.cpp:300] start_rtsp_server(): Failed to initialize camera: 0x105. Type: ESP32CAM, frame size: VGA (640x480), frame rate: 100 ms, jpeg quality: 12
```

If I know try to refresh the WebUI: 

```
12:48:44.917 > [ 72961][V][Parsing.cpp:237] _parseRequest(): Request: /
12:48:44.917 > [ 72966][V][Parsing.cpp:238] _parseRequest():  Arguments:
12:48:44.917 > [ 72972][V][main.cpp:81] handle_root(): Handle root
12:48:44.917 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
12:48:44.960 >
12:48:44.960 > Core  1 register dump:
12:48:44.960 > PC      : 0x4009391f  PS      : 0x00060d30  A0      : 0x80094956  A1      : 0x3ffb2080
12:48:44.960 > A2      : 0x0000084f  A3      : 0x400949a4  A4      : 0x3ffb20e4  A5      : 0x3ffb2100
12:48:44.960 > A6      : 0xfffffffc  A7      : 0x00000001  A8      : 0x00000854  A9      : 0x00000009
12:48:44.960 > A10     : 0x00000000  A11     : 0xc8418501  A12     : 0xc8418501  A13     : 0x3ffc6ef4
12:48:44.960 > A14     : 0x00060d20  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
12:48:45.003 > EXCVADDR: 0x00000853  LBEG    : 0x4008b450  LEND    : 0x4008b45b  LCOUNT  : 0x00000000
12:48:45.003 >
12:48:45.003 >
12:48:45.003 > Backtrace: 0x4009391c:0x3ffb2080 0x40094953:0x3ffb20a0 0x40094be3:0x3ffb20c0 0x40113eaa:0x3ffb20e0 0x400e6b7a:0x3ffb2120 0x400d7056:0x3ffb2160 0x40179015:0x3ffb2660 0x400dfc1a:0x3ffb2680 0x400dfcc1:0x3ffb26a0 0x400dfda7:0x3ffb26e0 0x400dffe6:0x3ffb2750 0x401797f5:0x3ffb27c0 0x400e4a47:0x3ffb27e0 0x400d6972:0x3ffb2800 0x400ea059:0x3ffb2820
```

rzeldent, 2023-02-28:
Hi Psi,

Could you make sure you've got the latest version and updates for platformio?
Could you try it with platformio + Gui. Add the following line to platformio.ini:
monitor_filters = esp32_exception_decoder
to see where it goes wrong.... This will decode the crash stack

Kind regards.

psi-4ward, 2023-02-28:
I don't know the GUI - you mean the IDE based on VSCode? 
Can I receive the infos also using pio-core (on cli?)

rzeldent, 2023-02-28:
Yes platform io on vs code

rzeldent, 2023-02-28:
You can also try to clear the file existing data, run task PlatformIO: clear flash

psi-4ward, 2023-02-28:
```

  #0  0x400d5ffa:0x3ffb2530 in update_camera_settings() at src/main.cpp:269
  #1  0x400d6173:0x3ffb2550 in on_config_saved() at src/main.cpp:333
  #2  0x40179015:0x3ffb2570 in std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at /home/psi/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #3  0x400dfc1a:0x3ffb2590 in std::function<void ()>::operator()() const at /home/psi/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
  #4  0x400e3efa:0x3ffb25b0 in iotwebconf::IotWebConf::saveConfig() at .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.cpp:196
  #5  0x400e402b:0x3ffb25f0 in iotwebconf::IotWebConf::handleConfig(iotwebconf::WebRequestWrapper*) at .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.cpp:346
  #6  0x400d46eb:0x3ffb2650 in iotwebconf::IotWebConf::handleConfig() at .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.h:262
      (inlined by) operator() at src/main.cpp:400
      (inlined by) _M_invoke at /home/psi/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #7  0x400dfc1a:0x3ffb2680 in std::function<void ()>::operator()() const at /home/psi/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
  #8  0x400dfcc1:0x3ffb26a0 in FunctionRequestHandler::handle(WebServer&, http_method, String) at /home/psi/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/detail/RequestHandlersImpl.h:45
  #9  0x400dfda7:0x3ffb26e0 in WebServer::_handleRequest() at /home/psi/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.cpp:652
  #10 0x400dffe6:0x3ffb2750 in WebServer::handleClient() at /home/psi/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.cpp:319
  #11 0x401797f5:0x3ffb27c0 in iotwebconf::StandardWebServerWrapper::handleClient() at .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.h:137
  #12 0x400e4a21:0x3ffb27e0 in iotwebconf::IotWebConf::doLoop() at .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.cpp:558
  #13 0x400d6972:0x3ffb2800 in loop() at src/main.cpp:437
  #14 0x400ea059:0x3ffb2820 in loopTask(void*) at /home/psi/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

rzeldent, 2023-02-28:
Hi Psi,

Something I suspected. The pointer to the sensor cannot be obtained (maybe because camera settings invalid?).
However, made a fix for this and of you do an pull of the code from GitHub it might fix the issue.
Details: https://github.com/rzeldent/esp32cam-rtsp/pull/43/commits/266d40fbf9db96e7dd4a62684c11d56a0ec87625

Kind regards,

Rene

psi-4ward, 2023-02-28:
Yes this fixes the panic and I'll get now `update_camera_settings(): Unable to get camera sensor`

Probably this could be related: https://github.com/espressif/esp32-camera/issues/450#issuecomment-1434915386

rzeldent, 2023-02-28:
Yes, this is the issue. Possibly the setting is not correct or a faulty camera.
You get an error that the camera is not found:
E (36867) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

Have you connected it correctly, not faulty?

psi-4ward, 2023-02-28:
Yes should be fine. I've a bunch and tried 3 different ones but they are all from the same aliexpress order.

I entered my AP data but cannot access the ui:

```
[ 4078][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[ 40733][V][main.cpp:81] handle_root(): Handle root
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4009391f  PS      : 0x00060d30  A0      : 0x80094956  A1      : 0x3ffb2080  
A2      : 0x0000084f  A3      : 0x400949a4  A4      : 0x3ffb20e4  A5      : 0x3ffb2100  
A6      : 0xfffffffc  A7      : 0x00000001  A8      : 0x00000854  A9      : 0x00000009  
A10     : 0x00000000  A11     : 0x70d61e3e  A12     : 0x70d61e3e  A13     : 0x3ffc6ef4  
A14     : 0x00060d20  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000853  LBEG    : 0x4008b450  LEND    : 0x4008b45b  LCOUNT  : 0x00000000  

Backtrace: 0x4009391c:0x3ffb2080 0x40094953:0x3ffb20a0 0x40094be3:0x3ffb20c0 0x40113ee6:0x3ffb20e0 0x400e6ba6:0x3ffb2120 0x400d708a:0x3ffb2160 0x40179051:0x3ffb2660 0x400dfc46:0x3ffb2680 0x400dfced:0x3ffb26a0 0x400dfdd3:0x3ffb26e0 0x400e0012:0x3ffb2750 0x40179831:0x3ffb27c0 0x400e4a73:0x3ffb27e0 0x400d69a6:0x3ffb2800 0x400ea085:0x3ffb2820

  #0  0x4009391c:0x3ffb2080 in block_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_tlsf_block_functions.h:75 (discriminator 4)
      (inlined by) block_is_last at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_tlsf_block_functions.h:86 (discriminator 4)
      (inlined by) tlsf_walk_pool at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_tlsf.c:603 (discriminator 4)
  #1  0x40094953:0x3ffb20a0 in multi_heap_get_info_impl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap.c:412
  #2  0x40094be3:0x3ffb20c0 in multi_heap_get_info at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:364
  #3  0x40113ee6:0x3ffb20e0 in heap_caps_get_info at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:535
  #4  0x400e6ba6:0x3ffb2120 in EspClass::getPsramSize() at /home/psi/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:158
  #5  0x400d708a:0x3ffb2160 in handle_root() at src/main.cpp:159 (discriminator 14)
  #6  0x40179051:0x3ffb2660 in std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at /home/psi/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #7  0x400dfc46:0x3ffb2680 in std::function<void ()>::operator()() const at /home/psi/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
  #8  0x400dfced:0x3ffb26a0 in FunctionRequestHandler::handle(WebServer&, http_method, String) at /home/psi/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/detail/RequestHandlersImpl.h:45
  #9  0x400dfdd3:0x3ffb26e0 in WebServer::_handleRequest() at /home/psi/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.cpp:652
  #10 0x400e0012:0x3ffb2750 in WebServer::handleClient() at /home/psi/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.cpp:319
  #11 0x40179831:0x3ffb27c0 in iotwebconf::StandardWebServerWrapper::handleClient() at .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.h:137
  #12 0x400e4a73:0x3ffb27e0 in iotwebconf::IotWebConf::doLoop() at .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.cpp:578
  #13 0x400d69a6:0x3ffb2800 in loop() at src/main.cpp:443
  #14 0x400ea085:0x3ffb2820 in loopTask(void*) at /home/psi/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 53f7970a3ff0cccf

Rebooting...
```

psi-4ward, 2023-02-28:
[image]
[image]

rzeldent, 2023-02-28:
Hi Christophe,

How are you compiling? Have you included the build flags?

build_flags =
-DBOARD_HAS_PSRAM
-mfix-esp32-psram-cache-issue

psi-4ward, 2023-02-28:
Yes:

```ini
$ cat platformio.ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino

#upload_protocol = espota
#upload_port = 192.168.50.222
#upload_flags =
#    --auth='ESP32CAM-RTSP'

# Partition scheme for OTA
board_build.partitions = min_spiffs.csv

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = esp32_exception_decoder

build_flags =
    -O2
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    -D LED_FLASH=4
    -D LED_BUILTIN=33
    -D BOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -D IOTWEBCONF_PASSWORD_LEN=64

lib_deps =
    prampec/IotWebConf @ ^3.2.1
    geeksville/Micro-RTSP @ ^0.1.6
    rzeldent/micro-moustache@^1.0.1
```

```shell
pio run -t upload
```

rzeldent, 2023-02-28:
Does the ESP32-CAM demo works?

psi-4ward, 2023-03-07:
> Does the ESP32-CAM demo works?

Yes, got `arduino-esp32/libraries/ESP32/examples/Camera/CameraWebServer` working and also this project works now. Tbh, I don't know why.

rzeldent, 2023-03-25:
See more information in the thread at https://github.com/espressif/esp32-camera/issues/450.
Might be a faulty PSRAM...