rzeldent/esp32cam-rtsp #4: Failed to start, keep rebooting

widewing, 2022-09-11:
I have followed the steps in the readme, it flashed successfully, but keep rebooting after that.
```
01:03:43.291 > Rebooting...
01:03:43.296 > ets Jul 29 2019 12:21:46
01:03:43.296 > 
01:03:43.296 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
01:03:43.302 > configsip: 0, SPIWP:0xee
01:03:43.302 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:03:43.312 > mode:DIO, clock div:2
01:03:43.312 > load:0x3fff0030,len:1184
01:03:43.312 > load:0x40078000,len:13132
01:03:43.318 > load:0x40080400,len:3036
01:03:43.318 > entry 0x400805e4
01:03:43.701 > [     2][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
01:03:43.707 > [     3][V][WebServer.cpp:85] WebServer():␃`�U���Y.W�.'�U���Y.W�.%\K␗WJj��␅		␒␒��W+QW�Y.\&�KZX�K,\]�,'�LMW��Y](\]��equencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
01:03:44.164 > [   461][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
01:03:44.195 > [   495][I][main.cpp:166] setup(): CPU Freq = 240 Mhz
01:03:44.202 > [   496][I][main.cpp:167] setup(): Free heap: 249416 bytes
01:03:44.206 > [   496][I][main.cpp:168] setup(): Starting ESP32CAM-RTSP...
01:03:44.211 > Config version: 1.0
01:03:44.211 > Config size: 319
01:03:44.211 > Wrong config version. Applying defaults.
01:03:44.217 > [iwcAll]
01:03:44.217 > |-- [iwcSys]
01:03:44.217 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
01:03:44.222 > |   |-- 'iwcApPassword' with value: <hidden>
01:03:44.227 > |   |-- [iwcWifi0]
01:03:44.227 > |   |   |-- 'iwcWifiSsid' with value: ''
01:03:44.233 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
01:03:44.238 > |   \-- 'iwcApTimeout' with value: '30'
01:03:44.243 > |-- [iwcCustom]
01:03:44.243 > |   \-- [settings]
01:03:44.243 > |       |-- 'config' with value: 'AI THINKER'
01:03:44.249 > |       |-- 'fr' with value: '20'
01:03:44.249 > |       |-- 'fs' with value: 'SVGA (800x600)'
01:03:44.254 > |       \-- 'q' with value: '12'
01:03:44.260 > \-- [hidden]
01:03:44.260 > [   556][E][ESPmDNS.cpp:65] begin(): Failed starting MDNS
01:03:44.265 > [   556][E][ESPmDNS.cpp:148] addService(): Failed adding service http.tcp.
01:03:44.270 > 
01:03:44.270 > AP password was not set in configuration
01:03:44.286 > State changing from: 0 to 1
01:03:44.293 > [   591][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
01:03:44.335 > Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
01:03:44.340 > 
01:03:44.340 > Core  0 register dump:
01:03:44.340 > PC      : 0x4015507b  PS      : 0x00060c30  A0      : 0x80155182  A1      : 0x3ffd3360  
01:03:44.351 > A2      : 0x80155182  A3      : 0x80155192  A4      : 0x00000006  A5      : 0x3ffd0a7e  
01:03:44.356 > A6      : 0x0c080400  A7      : 0x3ffc723c  A8      : 0x8015503c  A9      : 0x3ffd3300  
01:03:44.367 > A10     : 0x3ffd3360  A11     : 0x3ffd3376  A12     : 0x80155182  A13     : 0x00000000  
01:03:44.372 > A14     : 0x00000050  A15     : 0x00000075  SAR     : 0x00000007  EXCCAUSE: 0x0000001c  
01:03:44.383 > EXCVADDR: 0x80155192  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
01:03:44.389 > 
01:03:44.389 > 
01:03:44.389 > Backtrace:0x40155078:0x3ffd33600x4015517f:0x3ffd33b0 0x40152516:0x3ffd33d0 0x40152f8e:0x3ffd33f0 0x4011f102:0x3ffd3420 0x4011f179:0x3ffd3450 0x40135b02:0x3ffd3470 0x401362d9:0x3ffd3490 0x401349a0:0x3ffd34b0 0x40171d69:0x3ffd34d0 
01:03:44.410 > 
01:03:44.410 > 
01:03:44.410 > 
01:03:44.410 > 
01:03:44.410 > ELF file SHA256: 0000000000000000
01:03:44.410 > 
01:03:44.416 > Rebooting...
01:03:44.416 > ets Jul 29 2019 12:21:46
01:03:44.416 > 
01:03:44.416 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
01:03:44.422 > configsip: 0, SPIWP:0xee
01:03:44.427 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:03:44.432 > mode:DIO, clock div:2
01:03:44.432 > load:0x3fff0030,len:1184
01:03:44.432 > load:0x40078000,len:13132
01:03:44.438 > load:0x40080400,len:3036
01:03:44.438 > entry 0x400805e4
```

widewing, 2022-09-11:
it seems 3.3v is too low. when I connected 5v this issue is gone.

closed 2022-09-11 by widewing

rzeldent, 2022-09-11:
Thanks!

Yes, there is a brownout but this is disabled to have issues when the power
is a bit too low.
However, 3.3v is really too low...

Thanks for resolving this issue; hops it helps others with similar problems!

Regards,

Rene