rzeldent/esp32cam-rtsp #5: Failing to initialize.... #2

WSDsuperace, 2022-09-12:
Hi,
Your project looks exactly like something I really need. But... I can not get my camera to init.

Same board as reported in Failing to initialize.... #1 and I just can not get it to work. I know the PCB is good, it is working with arduino sketches.
[image]

Any ideas?

rzeldent, 2022-09-12:
Have you checked if you're using the same IO settings as the sketch that did work?
You could try to set  the framebuffers to 1 instead of 2. You can change this in the file include/camera_config.h for AI_Thinker.
  config.fb_count = 1;

This because maybe you do not have PSRAM?
If you set lower resolutions, do you get an image? 

Also make sure you have a good power supply....

rzeldent, 2022-09-12:
Made selection of PSRam automatic and fb_count configurable.
Also added two new boards; wrover and m5 stack.

WSDsuperace, 2022-09-15:
Brilliant!!!!!
I do not know what magic you pulled but now my boards are happy and have PSRAM. Your suggestions to change buffer to 1 did not make any difference and the IF PSRAM always retured False.  But your new build totally works!

Thank you soooooo soooo much.

On a different note, maybe you should add in the documentation (unless it is already there and I have missed it) that the only way to totally reset saved passwords is to do an erase of the flash. Build and Download does not clear previously saved passwords, at least not for me so I nuked it with
esptool.py --chip esp32 --port /dev/ttyUSB0 erase_flash 
and all was well :)
Again thank your for your help.

closed 2022-09-15 by rzeldent

WSDsuperace, 2022-09-15:
All my boards have PSRAM, works perfectly with Micropython but just too slow fps. With the old code I never got the camera to init and something crashed on the way, regardless where the frame buffer was allocated. So thank your for your magic.

Do you want more feature suggestions? If yes, how? Open them as an Issue? What do you suggest?