rzeldent/esp32cam-rtsp #11: Advertised mDNS-based link on web info page not working

davygravy, 2022-09-24:
First, nicely done!  Echt geweldig, eigenlijk!  _Echte!!!_  Thank you for your work on this project - next time I'm in NL visiting in-laws, I'd be happy to buy you a few rounds of beer or coffee.   The quality of the image, the granularity of the adjustments/config, and the look of the webpage are best-in-class, from what I've seen.

Now to the point, when I flashed my AI-Thinker module, I had to initially use the ip address in order to view it, rather than using the mDNS link that is shown near the bottom of the webpage (port 80)  - after the words "The camera stream can be found at the following location:".    Neither VLC nor HA could resolve the mDNS link to the correct IP.  After experiencing this, I went to the webpage and changed the "ThingName" to something else & saved the new config, and then after that my *nix systems (MacOS, Debian) would correctly resolve the host using the new mDNS alias, ie. using VLC or avahi-browser, for instance.

My hunch is that it is not a peculiarity in my network here... I peek around in the code and couldn't find anything obvious that would cause it, but maybe its something between the (new-ish) hostname on the webpage feature and the hand-off/call to mDNS or rtsp server.

**_Please close this issue if you can't reproduce it._**   If you'd like me to test something, I'll be happy to do that. [I have 3 unflashed/new AI-Thinker modules that I could use as "guinea-pig"/test cases...]

Again, thanks!

rzeldent, 2022-09-24:
Hi Dave,

Thanks for your report. Fortunately I was able to reproduce this quickly
and it is caused by a change due to OTA (Over The Air Programming).
There the mDNS is also set and composed of the device and the mac. This is
the value hostname in the config.
This was not updated in the HTM.

I corrected the HTML so ut should be fixed now. I already merged the branch
(Changed mDNS value to HostName by rzeldent · Pull Request #12 ·
rzeldent/esp32cam-rtsp (github.com)
<https://github.com/rzeldent/esp32cam-rtsp/pull/12>) so should be fixed now.

Would be nice if you can confirm this works!

Kind regards,

Rene

davygravy, 2022-09-25:
I performed a git pull, erased my device, rebuilt and uploaded.  It seems to have fixed the "Camera stream" mDNS link.    :+1: :+1: 

I tested the link behavior on MacOS with Chrome and VLC - works as expected.   

On my Debian + Xfce box, VLC seems to not be happy with something.  I know that the mDNS alias on the camera's web page does resolve correctly to the proper IP.  For instance, the Parole Media Player does play the stream correctly when supplied with the mDNS link.  I suspect that it may be a problem with my system or perhaps a problem w/ Debian + VLC.  

Will reboot my system and/or try on another Linux box later today, but wanted to post this now.

rzeldent, 2022-09-25:
Hi Dave,

Have you installed avahi on Debian? This is not installed by default ( avahi
- mDNS/DNS-SD <https://www.avahi.org/>_

> sudo apt install avahi

Could you give this a try?

Rene

davygravy, 2022-09-25:
I do have avahi installed already - we have MacOS and other *nix devices that benefit from avahi/bonjour autoconfiguration, so it is part of my "personal default install", even if Debian doesn't install it.

```
apt list --installed | grep avahi

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

avahi-autoipd/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
avahi-daemon/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
avahi-discover/now 0.8-5 all [installed,upgradable to: 0.8-5+deb11u1]
avahi-utils/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
libavahi-client3/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
libavahi-common-data/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
libavahi-common3/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
libavahi-core7/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
libavahi-glib1/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
python3-avahi/now 0.8-5 amd64 [installed,upgradable to: 0.8-5+deb11u1]
```

VLC (on Bullseye) does also correctly resolve the link, but it is just not able to play rtsp it seems.

ZoneMinder users have commented on this Debian+VLC problem, for instance:  < https://forums.zoneminder.com/viewtopic.php?p=125003&sid=362bfc7ce56e1e272b66442dbf2f13eb#p125003 >

Seems that "free software policy strictness" can have some unintended dis-incentives...  :^(

rzeldent, 2022-10-01:
Hi Dave,

Any progress / hints?

davygravy, 2022-10-02:
Hi Rene,

As far as I can see, the issue seems fixed with the last commit that you made. Link resolves correctly where ever the client supports it.  

👍 

Thanks for fixing this! 

Cheer, regards & respect,

Dave

closed 2022-10-02 by davygravy