rzeldent/esp32cam-rtsp #16: Enhancement needed--RTSP stream authentication.

Doublefire-Chen, 2022-10-16:
I would be appreciated if the authentication for RTSP stream is added for safety concern. Thank you in advance, bro.

rzeldent, 2022-10-16:
Hi Chen,

That indeed would be a nice feature. I'm currently using a library for RTSP and would require a branch.
Will keep it in mind!

alessandrobrugnera, 2022-12-07:
@Doublefire-Chen I don't think authentication for RTSP will be a good solution for protecting the stream since with tools like wireguard you can pretty much instantly sniff the username and password since they are being sent in plain text. I think the best approach will be to implement SRTP which is the secure version of RT(S)P. Then a username and password will make sense since they will be encrypted and the stream itself will be too.

Doublefire-Chen, 2022-12-10:
> @Doublefire-Chen I don't think authentication for RTSP will be a good solution for protecting the stream since with tools like wireshark you can pretty much instantly sniff the username and password since they are being sent in plain text. I think the best approach will be to implement SRTP which is the secure version of RT(S)P. Then a username and password will make sense since they will be encrypted and the stream itself will be too.

Yes, you are right. security level: SRTP > RTSP with authentication > RTSP without authentication.

rzeldent, 2026-08-28:
Is implemented in the current develop branch. Needs to be enabled by a define but is really slow and did not get it to work with VLC.
But let me know if this works on your SRTP client!