rzeldent/esp32cam-rtsp #18: AP Restart on Wifi Client Connect

codefaux, 2022-10-17:
Fresh setup, first time attempting to use this firmware. Device is [ESP32-CAM-MB - CH340G - OV2640](https://www.amazon.com/gp/product/B092ZXT1XG)

Previously running Tasmota firmware OK. Sequence was `git clone git@github.com:rzeldent/esp32cam-rtsp.git`  `pio run`  `pio run -t erase`  then  `pio run -t upload`

I've tried from scratch a few times, always same result; flashes OK, boots OK. AP appears, but as soon as I connect, the AP closes for a few seconds, and reappears. Logs from one run follow;

```
15:13:45.847 > [     3][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
15:13:45.850 > [     9][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
15:13:46.298 > [   461][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
15:13:46.313 > [   475][I][main.cpp:231] setup(): CPU Freq: 240 Mhz
15:13:46.315 > [   476][I][main.cpp:232] setup(): Free heap: 246464 bytes
15:13:46.320 > [   476][I][main.cpp:233] setup(): Starting ESP32CAM-RTSP...
15:13:46.326 > Config version: 1.1
15:13:46.328 > Config size: 326
15:13:46.328 > Wrong config version. Applying defaults.
15:13:46.334 > [iwcAll]
15:13:46.334 > |-- [iwcSys]
15:13:46.334 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
15:13:46.340 > |   |-- 'iwcApPassword' with value: <hidden>
15:13:46.343 > |   |-- [iwcWifi0]
15:13:46.345 > |   |   |-- 'iwcWifiSsid' with value: ''
15:13:46.348 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
15:13:46.354 > |   \-- 'iwcApTimeout' with value: '30'
15:13:46.356 > |-- [iwcCustom]
15:13:46.359 > |   \-- [settings]
15:13:46.359 > |       |-- 'config' with value: 'ESP32CAM'
15:13:46.365 > |       |-- 'fd' with value: '20'
15:13:46.368 > |       |-- 'fs' with value: 'SVGA (800x600)'
15:13:46.370 > |       |-- 'fb' with value: '2'
15:13:46.373 > |       \-- 'q' with value: '12'
15:13:46.376 > \-- [hidden]
15:13:46.379 > AP password was not set in configuration
15:13:46.381 > State changing from: 0 to 1
15:13:46.389 > [   553][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
15:13:46.483 > Setting up AP: ESP32CAM-RTSP
15:13:46.483 > With default password: <hidden>
15:13:46.486 > [   647][E][WiFiAP.cpp:153] softAP(): passphrase too short!
15:13:46.491 > [   647][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
15:13:46.497 > AP IP address: [   652][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
15:13:46.505 > 192.168.4.1
15:13:46.508 > AP password was not set.
15:13:46.511 > WiFi SSID was not set.
15:13:46.511 > Will stay in AP mode.
15:13:46.514 > State changed from: 0 to 1
15:13:51.921 > Connection to AP.
15:13:51.927 > [  6089][V][WiFiGeneric.cpp:405] _arduino_event_cb(): AP Station Connected: MAC: 1e:d8:de:6b:a9:68, AID: 1
15:13:51.936 > [  6090][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
15:13:52.038 > [  6201][V][WiFiGeneric.cpp:419] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
15:13:52.044 > [  6201][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
15:13:52.200 > [  6362][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
15:13:52.205 > [  6369][V][Parsing.cpp:122] _parseRequest(): method: GET url: /generate_204 search:
15:13:52.214 > [  6369][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
15:13:52.219 > [  6373][V][Parsing.cpp:227] _parseRequest(): headerValue: close
15:13:52.225 > [  6379][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
15:13:52.231 > [  6385][V][Parsing.cpp:227] _parseRequest(): headerValue: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.32 Safari/537.36
15:13:52.247 > [  6399][V][Parsing.cpp:226] _parseRequest(): headerName: Host
15:13:52.253 > [  6405][V][Parsing.cpp:227] _parseRequest(): headerValue: connectivitycheck.gstatic.com
15:13:52.258 > [  6413][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
15:13:52.267 > [  6419][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip
15:13:52.272 > [  6425][V][Parsing.cpp:254] _parseArguments(): args:
15:13:52.278 > [  6430][V][Parsing.cpp:237] _parseRequest(): Request: /generate_204
15:13:52.283 > [  6436][V][Parsing.cpp:238] _parseRequest():  Arguments:
15:13:52.289 > [  6441][E][WebServer.cpp:649] _handleRequest(): request handler not found
15:13:52.294 > Request for connectivitycheck.gstatic.com redirected to 192.168.4.1:80
15:13:52.300 > [  6459][W][WebServer.cpp:436] send(): content length is zero
15:13:58.332 > [ 12495][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
15:13:58.338 > [ 12496][V][Parsing.cpp:122] _parseRequest(): method: GET url: / search:
15:13:58.347 > [ 12499][V][Parsing.cpp:226] _parseRequest(): headerName: Host
15:13:58.352 > [ 12504][V][Parsing.cpp:227] _parseRequest(): headerValue: 192.168.4.1
[... 284 lines cut ...]
Relevent section appears to be;
```
15:13:58.893 > [ 13046][V][Parsing.cpp:254] _parseArguments(): args:
15:13:58.895 > [ 13051][V][Parsing.cpp:237] _parseRequest(): Request: /snapshot
15:13:58.904 > [ 13056][V][Parsing.cpp:238] _parseRequest():  Arguments:
15:13:58.907 > [ 13062][V][main.cpp:145] handle_snapshot(): handle_jpg
15:13:58.912 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
15:13:58.921 >
15:13:58.921 > Core  1 register dump:
15:13:58.923 > PC      : 0x400e20a7  PS      : 0x00060d30  A0      : 0x800d3468  A1      : 0x3ffb25f0
15:13:58.929 > A2      : 0x00000000  A3      : 0x3ffc543f  A4      : 0x0000003f  A5      : 0x0000003a
15:13:58.937 > A6      : 0x3f400155  A7      : 0x00000001  A8      : 0x800e20a4  A9      : 0x3ffb25d0
15:13:58.946 > A10     : 0x3ffc49dc  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
15:13:58.954 > A14     : 0x3ffb8a28  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
15:13:58.962 > EXCVADDR: 0x00000004  LBEG    : 0x4008c5d5  LEND    : 0x4008c5e5  LCOUNT  : 0xfffffffb
15:13:58.970 >
15:13:58.970 >
15:13:58.970 > Backtrace:0x400e20a4:0x3ffb25f00x400d3465:0x3ffb2610 0x40170ea5:0x3ffb2660 0x400dc546:0x3ffb2680 0x400dc5ed:0x3ffb26a0 0x400dc6d3:0x3ffb26e0 0x400dc912:0x3ffb2750 0x40171199:0x3ffb27c0 0x400e13c1:0x3ffb27e0 0x400d43ca:0x3ffb2800 0x400e66e5:0x3ffb2820
15:13:58.993 >
15:13:58.993 >
15:13:58.993 >
15:13:58.993 >
15:13:58.993 > ELF file SHA256: 0000000000000000
15:13:58.996 >
```

codefaux, 2022-10-18:
Hello @rzeldent,

Great, glad to hear you solved it.
This has mitigated my issue, and I was able to get the device online.

One significant thing to report:
- The UI indicated that the camera was initialized. It was set to the wrong camera config (ESP32CAM default, mine is AI THINKER) so camera config initialization checking/validating seems broken. I believe this is related to the bug you just mitigated; possibly it believes the camera is correct and attempts to capture a frame, then breaks

Thank you for your attention
I'll leave the Issue open in case you wish to communicate regarding my report; feel free to close if you don't need it

rzeldent, 2022-10-19:
Hi CodeFaux,

This is possible. I do check the result of the camera initialization and
this is ESP_OK, otherwise the code would not continue.
I was able to reproduce it myself with the right settings for the camera so
something else is going on.
Something to look into in the future... Please keep the ticket open so I'll
not forget it!

Thanks for your feedback!

Rene

closed 2022-10-29 by rzeldent