rzeldent/esp32cam-rtsp #20: Stream not working in HA

smartroad, 2022-10-23:
Hi!

I have been able to get this installed on a generic ESP32CAM board and get it logged into HA, showing a picture (using "Generic Camera" integration). Works fine to show a (slowly updating) picture but when I click and it switches to showing the stream it shows an error. It flickers with the player and has occasionally the error: "Error with media stream contents". VLC shows the stream perfectly though :)

Tried with Firefox and Edge

rzeldent, 2022-10-23:
Hi SmartRoad,

What is HA?

Rene

crivera8602, 2022-10-29:
I believe HA means Home assistant

rzeldent, 2022-11-01:
Hi smartroad / crivera,

Thanks for clarifying HA! Home Assistant!

I looked at the documentation of HA https://www.home-assistant.io/integrations/generic/ and was wondering if the completer tsp URL was entered. If this is the case, is there some error message or logging?

Thanks and kind regards,

Rene

rzeldent, 2022-11-02:
Are there any warnings or logging? Did you use the DNS name or the IP address?

crivera8602, 2022-11-04:
it integrates fine on my end  [image]

only thing i noticed was the image disconnecting randomly 

https://user-images.githubusercontent.com/63074950/199900343-4c2dc388-aedf-4de9-bc3b-cd785b996cae.mp4

rzeldent, 2022-11-04:
HI Crivera,

Thanks for investigating this. I saw the glitch in the stream. Maybe the resolution is too high or the framerate too fast (or wifi connectivity?) so sometimes frames are skipped. Is this maybe the case?

Regards,

Rene

rzeldent, 2022-11-24:
It looks like it is working in Home Assistent. So will close this issue.

closed 2022-11-24 by rzeldent

bjoernmayer, 2023-01-30:
@smartroad did you get it working? I am facing the same issue