rzeldent/esp32cam-rtsp #21: compile error

WolfgangAipperspach, 2022-10-25:
src/main.cpp: In function 'esp_err_t initialize_camera()':
src/main.cpp:188:17: error: 'struct camera_config_t' has no member named 'fb_location'
   camera_config.fb_location = psramFound() ? CAMERA_FB_IN_PSRAM : CAMERA_FB_IN_DRAM;
                 ^
src/main.cpp:188:46: error: 'CAMERA_FB_IN_PSRAM' was not declared in this scope
   camera_config.fb_location = psramFound() ? CAMERA_FB_IN_PSRAM : CAMERA_FB_IN_DRAM;
                                              ^
src/main.cpp:188:67: error: 'CAMERA_FB_IN_DRAM' was not declared in this scope
   camera_config.fb_location = psramFound() ? CAMERA_FB_IN_PSRAM : CAMERA_FB_IN_DRAM;
                                                                   ^

rzeldent, 2022-10-26:
Hi Wolfgang,

Are you using the last version of the EspressIf currentlt at 5.1.1?
Try to go to Platforms and check the updates in PlatforIO...

Hope this helps, let me know if this was the issue

wlm2048, 2022-10-28:
I had the same issue and a platform update resolved it for me.

rzeldent, 2022-10-29:
Thanks Bill, for confirming this.

closed 2022-10-29 by rzeldent

WolfgangAipperspach, 2022-11-05:
thanks, platform update resolved the issue for me, as well