rzeldent/esp32cam-rtsp #22: FR: Allow controlling/setting LED lamp from web

RichyHBM, 2022-10-30:
As per the title, many of these boards come with an integrated flash led that can be used in low light situations, would be great to have control of this from the web front end

rzeldent, 2022-10-31:
Hi Richy,

I can implement a configuration setting, so the light is on/or off when starting. Would that be sufficient?

RichyHBM, 2022-10-31:
I had used https://github.com/easytarget/esp32-cam-webserver in the past which provided a 0%-100% setting which would be great, but an on off setting would be a great start!

rzeldent, 2022-11-01:
Hi Richy,

I made a branch; feature/light to set the led to a fixed value (0-100%). Can you take a look and is this what you had in mind?

RichyHBM, 2022-11-01:
Hi, tested the new branch and seems to work great! My only other request would be if there could be an api to change the value without needing to go through the config page? That way you could detect motion in home assistant and turn the light on that way. Not sure if that would be possible?

RichyHBM, 2022-11-01:
Also not related to the original request but would it be possible to allow for a restart without having changed the config? Not needed hi side but even when issuing a curl command it still won't work unless I have a config change made. Reason for this is I can track in home assistant when the camera stops recording (say it freezes or crashes) and issue a reboot.
Let me know if you'd rather a separate issue for that?

rzeldent, 2022-11-01:
Hi Richy,

I think the items you mention are resolved:
- There is a possibility to set the value using a GET request
- Restarting does not check of changes anymore.

Can you give it a spin?

rzeldent, 2022-11-02:
> Tested it out, seems to work great!!

Hi Richy,

I think it is a nice new feature so I will merge this in the main branch
and available for all!
Thanks for testing!

Kind regards,

Rene

rzeldent, 2022-11-02:
Merged feature/light into main

RichyHBM, 2022-11-03:
Sorry to bother about this again but I just noticed something, both of the new apis restart/flash have no authentication, these may want to be behind the same HTTP basic auth as the config so you have to pass the Authorisation header when accessing them

rzeldent, 2022-11-04:
Hi Richy,

Yes, restarting without authorization is not a nice feature. I implemented basic authentication using the user name admin and the access point password. I did this also for the flash LED because this is normally protected by the configuration.
See feature/basic_auth.
Can you test this to see it is working as expected?

Kind regards,

Rene

RichyHBM, 2022-11-04:
Seems to work great!

rzeldent, 2022-11-04:
Fix merged

closed 2022-11-04 by rzeldent