rzeldent/esp32cam-rtsp #24: Possible to update esp32cam via PlatformIO OTA?

RichyHBM, 2022-11-03:
As per the title, tried to update my cameras with latest develop using OTA, it prints that authentication was ok but then fails, this is using my local Wi-Fi, not connected directly to the esp access point. Is OTA supported with this project?

rzeldent, 2022-11-04:
Hi Richy,

You are right. I added this in the past but got a similar error and needed to take a look at it, so the code was still lingering.
I checked it today and the issue was that the partition scheme is/was not compatible with OTA; two versions need to be able to be on the esp32. This is fixed in the branch bugfix/ota.
Another thing that might prevent uploading is DNS. I suggest to use the IP address...

I managed to upload using OTA by just adding the line for the partition. Can you try it to see if your problem is resolved?

N.B. first you have to upload it (once) to have the new partition scheme. From then on OTA should work!

Regards,

Rene

RichyHBM, 2022-11-04:
I think it's working ok, I have 2 devices, the first it worked fine first time and I can OTA it with out an issue, the second it seems to reboot as soon as the uploading process starts, but I feel like that may be a device or power issue

rzeldent, 2022-11-04:
Fix merged

closed 2022-11-04 by rzeldent