rzeldent/esp32cam-rtsp #27: Getting only old snapshot, stream is working and real time

joeheaven, 2022-11-06:
Hi there, I managed to get this up and running. By accident I bought a devive with too low RAM for most applications, but this one works and in principle this is exactly what I wanted to do. So first of all: Many, many thanks for providing this!

When I startet the flashed firmware and did all configuration everything was fine: The ESP-CAM connected to my AP, I can access the webpage and I could do everything as it was supposed to be: Turning LED on and off, getting a stream in VLC and take s snapshot of the current field of view.

However, after that first image all is still working except the snapshot: LED works, stream works, webpage works, but when I open the snapshot URL I always get an old picture. Cleared the cache, disabled all blockers whatsoever, used a frash browser, used private mode, anything that could cause this to be done by the browser, but the cam is just sending one old (somehow buffered) picture.

Any ideas what might be causing that? 

After a powercylce the first image is again correct, but any later attempt yields the first taken picture again. Or is that supposed to work that way?

Best regards,
Jochen

rzeldent, 2022-11-06:
Hi Jochen,

Thanks for reporting this. Might be an issue indeed. Looked at the code and caching from the browser should not occur.
I was thinking, can you start a stream and see if the snapshot is updated in this case? 
It might be that the frame is not grabbed again...?

Rene

rzeldent, 2022-11-06:
Made a possible fix to grab a frame before getting the buffer:
[#](https://github.com/rzeldent/esp32cam-rtsp/tree/bugfix/snapshot_update).
Does this fixes the issue?

Regards,

Rene

joeheaven, 2022-11-06:
Dear Rene, Sunday morning and yet the answer and a fix comes at the speed of light. Wow, thanks very much.

Just for clarification regarding the previous behaviour: Activating the stream or not did not influence the outcome, the image was always the same.

I just checked the new version and observe the following: Getting a snapshot yields an image. Refreshing the browser with Ctrl+F5 (should reduce potential interference with cache) I get again the same picture, a second refresh then yields the new picture. So currently I would use a double cURL or so to get the real current picture. Thats already much better and manageable. But maybe not exactly the intended behaviour.

So: Thanks for a now (kind of) working solution, if you have other variants to test, please let me know.

Again: Thanks!!!!

rzeldent, 2022-11-06:
Hi Jochen,

Think I found the issue and, in a way also logical/explainable. See this post from EspressIf: [https://github.com/espressif/esp32-camera/issues/357](https://github.com/espressif/esp32-camera/issues/357).

What happens is that if you use a framebuffer count > 1, the old images remain in the framebuffer and will be retrieved first. So this explains why you need to refresh multiple times (depends on frame buffer size).
The initial fix got just one framebuffer but this works only of the frame buffer count equals 1; this is not the default.
One could set the framebuffer count to 0 but this might impact the peformance.

I fixed this issue by reading the fb framebuffer count times before sending the snapshot. I gave it a quick test and this is the solution.

Please give it a test so I can merge this fix in the main!

Kind regards,

Rene

joeheaven, 2022-11-07:
Dear Rene,

it works perfectly as expected now! Brilliant!

Thanks very much!

Best regards,
Jochen

rzeldent, 2022-11-07:
Hi Jochen,

Thanks for verifying. I merged the branch!

closed 2022-11-07 by rzeldent