rzeldent/esp32cam-rtsp #30: trouble accessing the webpage

sochartgit, 2022-11-15:
I've tried several ESP32CAM boards. What works:
1) I send the code using Platform.IO.
2) I get the ESP32CAM-RTSP AP with 192.168.4.1.
3) I'm able to fill the form, basically giving AP a password, SSID and SSID password.
4) Unit reboots and gets an IP from DHCP server.
5) Ping from my network

Now what does not work:
6) As soon as I try to open the webpage, unit reboots and goes back into AP mode.
7) I'm coming back to step 2 but SSID is already filled and camera config has been conserved too.

rzeldent, 2022-11-16:
Hi sochartgit,

Is it possible to make some logging? Just do the setup and save the serial output... This might possibly clarify some things.
Anyone else having these issues?

ianhodgson, 2022-12-06:
I had a similar issue with this coming off of other firmware until resetting with 'pio run -t erase'

rzeldent, 2023-02-03:
Hi Ian,

Thanks for mentioning. No answer so think it has been resolved...

closed 2023-02-03 by rzeldent