rzeldent/esp32cam-rtsp #31: Trouble initializing the camera

AdrianBanulsA, 2022-11-17:
Hi @rzeldent

I have an ESP32 camera mounted with the ESP32-CAM-MB USB programmer. I have successfully uploaded and configurated the camera WiFi SSID and WiFi password. However, when I monitor the serial output I receive the following error (it's at the end of the log):

09:18:21.856 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
09:18:21.861 > configsip: 0, SPIWP:0xee
09:18:21.864 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:18:21.870 > mode:DIO, clock div:2
09:18:21.872 > load:0x3fff0030,len:1184
09:18:21.875 > load:0x40078000,len:13132
09:18:21.877 > load:0x40080400,len:3036
09:18:21.877 > entry 0x400805e4
09:18:22.284 > [     2][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=�␘D␂rrrJj
09:18:22.290 > [     3][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)        
09:18:22.295 > [     9][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
09:18:22.743 > [   462][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
09:18:22.777 > [   496][I][main.cpp:267] setup(): CPU Freq: 240 Mhz
09:18:22.781 > [   497][I][main.cpp:268] setup(): Free heap: 245772 bytes
09:18:22.786 > [   497][I][main.cpp:269] setup(): Starting ESP32CAM-RTSP...
09:18:22.792 > Config version: 1.2
09:18:22.803 > Config size: 330
09:18:22.803 > Loading configurations
09:18:22.803 > [iwcAll]
09:18:22.803 > |-- [iwcSys]
09:18:22.803 > |   |-- 'iwcThingName' with value: 'raspicart03_cam'
09:18:22.803 > |   |-- 'iwcApPassword' with value: <hidden>
09:18:22.808 > |   |-- [iwcWifi0]
09:18:22.808 > |   |   |-- 'iwcWifiSsid' with value: 'TP-Link_84EE'
09:18:22.813 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
09:18:22.819 > |   \-- 'iwcApTimeout' with value: '30'
09:18:22.822 > |-- [iwcCustom]
09:18:22.824 > |   \-- [settings]
09:18:22.824 > |       |-- 'config' with value: 'ESP32CAM'
09:18:22.830 > |       |-- 'fd' with value: '20'
09:18:22.833 > |       |-- 'fs' with value: 'SVGA (800x600)'
09:18:22.836 > |       |-- 'fb' with value: '2'
09:18:22.838 > |       |-- 'q' with value: '12'
09:18:22.841 > |       \-- 'li' with value: '1'
09:18:22.844 > \-- [hidden]
09:18:22.847 > State changing from: 0 to 2
09:18:22.869 > [   588][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
09:18:22.960 > Setting up AP: raspicart03_cam
09:18:22.963 > Use password: <hidden>
09:18:22.963 > [   679][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
09:18:22.968 > [   680][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
09:18:23.528 > AP IP address: 192.168.4.1
09:18:23.529 > [  1250][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
09:18:23.534 > [  1250][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
09:18:23.542 > [  1251][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
09:18:23.548 > [  1257][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
09:18:23.557 > AP timeout (ms): 30000
09:18:23.557 > State changed from: 0 to 2
09:18:53.547 > State changing from: 2 to 3
09:18:53.559 > [ 31279][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
09:18:53.565 > [ 31279][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
09:18:53.571 > [ 31283][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
09:18:53.577 > E (44951) wifi_init_default: esp_wifi_get_mac failed with 12289
09:18:53.582 > Connecting to [TP-Link_84EE] (password is hidden)
09:18:53.588 > WiFi timeout (ms): 30000
09:18:53.590 > [ 31292][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
09:18:53.737 > [ 31447][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
09:18:53.746 > [ 31466][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.0.152
09:18:53.752 > [ 31466][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
09:18:53.760 > [ 31469][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.0.152, MASK: 255.255.255.0, GW: 192.168.0.1
09:18:53.768 > WiFi connected
09:18:53.771 > IP address: 192.168.0.152
09:18:53.774 > State changing from: 3 to 4
09:18:56.873 > Active mDNS services: 1
09:18:56.874 > Accepting connection
09:18:56.876 > [ 34593][V][main.cpp:238] on_connected(): on_connected
09:18:56.886 > [ 34605][I][ArduinoOTA.cpp:141] begin(): OTA server at: esp32-5443b2abcb1c.local:3232
09:18:56.892 > [ 34605][V][main.cpp:221] start_rtsp_server(): start_rtsp_server
09:18:56.898 > [ 34607][V][main.cpp:200] initialize_camera(): initialize_caera
09:18:56.903 > [ 34613][I][main.cpp:201] initialize_camera(): Camera config: ESP32CAM
09:18:56.911 > [ 34619][I][main.cpp:203] initialize_camera(): Frame size: SVGA (800x600)
09:18:56.917 > [ 34625][I][main.cpp:205] initialize_camera(): Frame buffers: 2
09:18:56.921 > [ 34631][I][main.cpp:207] initialize_camera(): JPEG quality: 12
09:18:56.928 > [ 34637][I][main.cpp:209] initialize_camera(): Frame rate: 20 ms
09:18:56.956 > E (48340) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
09:18:56.962 > [ 34676][E][main.cpp:225] start_rtsp_server(): Failed to initialize camera: 0x105. Type: ESP32CAM, frame size: SVGA (800x600), frame buffers: 2, frame rate: 20 ms, jpeg quality: 12
09:18:56.980 > State changed from: 3 to 4

I would appreciate any help. Thank you so much in advance!

rzeldent, 2022-11-17:
Hi Andrian,

The camera cannot be initialized; it looks like something hardware related. Is the power supply OK? Otherwise a faulty camera or flat cable connection? It this every time or sometimes?

This happens more often but there was no solution. [https://forum.arduino.cc/t/esp32-cam-error-camera-init-failed/959287/10](https://forum.arduino.cc/t/esp32-cam-error-camera-init-failed/959287/10)

Let me know if if works on another esp32 cam or maybe you have a spare camera....

rhaamo, 2023-01-16:
I've got two cheap ESP32-CAM with both an OV2640 and getting the same problem with both too

rzeldent, 2023-02-09:
I looked at the build and saw this line:
Flash: [=====     ]  54.4% (used 1069225 bytes from 1966080 bytes)

So maybe too big to swap?

WarHawk8080, 2023-03-14:
Put a 22uf capacitor on the power...might be having issues with power sag and brownouts...these little modules are VERY susceptible to voltage spikes and ripple
https://electronics.stackexchange.com/questions/602540/esp32-power-supply-do-i-need-input-capacitors

rzeldent, 2023-03-25:
Added a setting to enable using the PSRAM module. If there are problems, try without the PSRAM.
https://github.com/rzeldent/esp32cam-rtsp/pull/53

I read that there are modules around with no/faulty PSRAM

regystro, 2023-07-09:
Same just happened to me. After setting it up, I ended with the same error. That's probably caused because you have something wrong in your config that prevents camera initialization, but even if you reflash the firmware, the configuration is kept, so nothing to do here. Until I found https://github.com/rzeldent/esp32cam-rtsp/issues/30#issuecomment-1338746449 (thank you, @ianhodgson)

Solution:
- Use command `pio run -t erase` to reset the settings
- Then reflash the firmware and you'll be able to connect again to the AP
- In my case problem was I choosed ESP32CAM as camera model. That was causing the initialization error. The default AI THINKER works flawlessly (again, in my case)
- Also do not forget to set the AP password

@rzeldent , I think it would be worth mentioning this in the README for troubleshooting

closed 2024-02-02 by rzeldent