rzeldent/esp32cam-rtsp #32: Unable to auth wifi with +32 chars

sirovermark, 2022-11-23:
Hi guys,

I can't authenticate to my AP when the WiFi password is longer than 32 characters:

```
00:29:45.090 > [ 51937][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
00:29:45.096 > [ 51969][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
00:29:45.104 > [ 51972][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
00:29:45.110 > [ 51986][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
00:29:45.120 > [ 51984][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
00:29:45.126 > [ 51995][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
00:29:45.136 > State changed from: 2 to 3
00:29:48.215 > [ 55094][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: TestWifi, BSSID: *************, Reason: 15
00:29:48.224 > [ 55095][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
00:29:48.232 > [ 55102][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
00:30:15.073 > Giving up.
```

I've tried to remove all the characters of the same password until ending with 32 and it works perfectly.

If it's impossible to fix this and reach the maximum length (63 characters), I think this should be documented somewhere :)

Thanks for your work!

rzeldent, 2022-11-24:
Hi sirovermark,

Yes, you are right. The password is limited to 32 characters. Basically this is just a define in another library used: iotwebconf. There in file iotwebconfsettings.h at line 25 the following is defined:
```
// -- Maximal length of password used in IotWebConfig configuration.
#ifndef IOTWEBCONF_PASSWORD_LEN
# define IOTWEBCONF_PASSWORD_LEN 33
#endif
```
This should be resolved in the library but for now I added a define to the build to support passwords up to 63 characters.
This should fix this issue. Unfortunately this changes the configuration size so, after updating, the configuration will be lost.
Could you give it a try to see if this solves the issue? It is located in the branch bugfix_wifi_password_lenght.

Regards,

Rene

sirovermark, 2022-11-24:
Hi @rzeldent ,

I'm unable to build it using the new code. I've cleaned everything before running it and I've tried many times:

```
Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (5.2.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1     
|   |-- DNSServer @ 2.0.0  
|   |   |-- WiFi @ 2.0.0   
|   |-- WebServer @ 2.0.0  
|   |   |-- WiFi @ 2.0.0   
|   |   |-- FS @ 2.0.0     
|   |-- WiFi @ 2.0.0       
|   |-- EEPROM @ 2.0.0     
|   |-- ESPmDNS @ 2.0.0    
|   |   |-- WiFi @ 2.0.0   
|-- Micro-RTSP @ 0.1.6     
|   |-- WiFi @ 2.0.0       
|-- micro-moustache @ 1.0.0
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- RTSPServer @ 1.0.0
|   |-- arduino-timer @ 2.3.1
|   |-- Micro-RTSP @ 0.1.6
|   |   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32cam\src\main.cpp.o
Building .pio\build\esp32cam\bootloader.bin
Generating partitions .pio\build\esp32cam\partitions.bin
esptool.py v4.2.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32cam\libdfd\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32cam\lib0f8\DNSServer\DNSServer.cpp.o
Compiling .pio\build\esp32cam\libacc\FS\FS.cpp.o
Compiling .pio\build\esp32cam\libacc\FS\vfs_api.cpp.o
Compiling .pio\build\esp32cam\lib07a\WebServer\Parsing.cpp.o
Compiling .pio\build\esp32cam\lib07a\WebServer\WebServer.cpp.o
Compiling .pio\build\esp32cam\lib07a\WebServer\detail\mimetable.cpp.o
[... 56 lines cut ...]
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-rgb-led.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32cam\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32cam\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32cam\libFrameworkArduino.a
*** [.pio\build\esp32cam\firmware.elf] Implicit dependency `C:\Users\********\.platformio\platforms\espressif32\builder\64' not found, needed by target `.pio\build\esp32cam\firmware.elf'.
============================================================================================================ [FAILED] Took 46.25 seconds ============================================================================================================
```

No problems with the main one, of course :)

Thank you!

rzeldent, 2022-11-24:
Hi sirovermark,

Was a typo. Should compile now...

sirovermark, 2022-11-24:
Looks good @rzeldent !! :)

I have now been able to log into the WIFI using a long password.

Thank you so much for your fast response!

rzeldent, 2022-11-24:
Nice.

I will merge it for the other users and made a bug report for the library.
[https://github.com/prampec/IotWebConf/issues/273](https://github.com/prampec/IotWebConf/issues/273)

Thanks for confirming.

closed 2022-11-24 by rzeldent