rzeldent/esp32cam-rtsp #34: Trouble compiling in VSCode

KingBain, 2023-01-11:
[image]

Compiling .pio\build\esp32cam\src\main.cpp.o
In file included from src\main.cpp:9:0:
include/frame_size.h:18:17: error: 'FRAMESIZE_240X240' was not declared in this scope
     {"240x240", FRAMESIZE_240X240},
                 ^
include/frame_size.h:21:24: error: 'FRAMESIZE_HVGA' was not declared in this scope
     {"HVGA (480x320)", FRAMESIZE_HVGA},
                        ^
include/frame_size.h:25:23: error: 'FRAMESIZE_HD' was not declared in this scope
     {"HD (1280x720)", FRAMESIZE_HD},
                       ^
src\main.cpp: In function 'void stream_text_file_gzip(const unsigned char*, size_t, const char*)':
src\main.cpp:55:73: error: no matching function for call to 'WebServer::sendContent(const char*, size_t&)'
   web_server.sendContent(reinterpret_cast<const char *>(content), length);
                                                                         ^
In file included from .pio\libdeps\esp32cam\IotWebConf\src/IotWebConf.h:25:0,
                 from src\main.cpp:4:
C:\Users\Bain\.platformio\packages\framework-arduinoespressif32\libraries\WebServer\src/WebServer.h:130:8: note: candidate: void WebServer::sendContent(const String&)
   void sendContent(const String& content);
        ^
C:\Users\Bain\.platformio\packages\framework-arduinoespressif32\libraries\WebServer\src/WebServer.h:130:8: note:   candidate expects 1 argument, 2 provided
src\main.cpp: In function 'void handle_root()':
src\main.cpp:80:25: error: 'class EspClass' has no member named 'getChipModel'

KingBain, 2023-01-11:
[image]

looks like updating modules in PIO fixed the issue.

closed 2023-01-11 by KingBain

rzeldent, 2023-01-12:
Hi John,

Thanks for sharing your experience and resolution; this might help others having the same issue!

Kind  regards,

Rene