rzeldent/esp32cam-rtsp #36: Cannot compile

doudz, 2023-01-29:
Hi !

I didn't manage to compile it : 

```
doudz@big:~/esp32cam-rtsp$ pio run 
Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (1.12.4) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10004.210126 (1.0.4) 
 - tool-esptoolpy @ 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 @ 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|   |-- DNSServer @ 1.1.0
|   |   |-- WiFi @ 1.0
|   |-- WebServer @ 1.0
|   |   |-- WiFi @ 1.0
|   |   |-- FS @ 1.0
|   |-- WiFi @ 1.0
|   |-- EEPROM @ 1.0.3
|   |-- ESPmDNS @ 1.0
|   |   |-- WiFi @ 1.0
|-- Micro-RTSP @ 0.1.6
|   |-- WiFi @ 1.0
|-- micro-moustache @ 1.0.0
|-- ArduinoOTA @ 1.0
|   |-- Update @ 1.0
|   |-- WiFi @ 1.0
|   |-- ESPmDNS @ 1.0
|   |   |-- WiFi @ 1.0
|-- ESPmDNS @ 1.0
|   |-- WiFi @ 1.0
|-- RTSPServer @ 1.0.0
|   |-- arduino-timer @ 2.3.1
|   |-- Micro-RTSP @ 0.1.6
|   |   |-- WiFi @ 1.0
|   |-- ESPmDNS @ 1.0
|   |   |-- WiFi @ 1.0
|   |-- WiFi @ 1.0
Building in release mode
Compiling .pio/build/esp32cam/src/main.cpp.o
Generating partitions .pio/build/esp32cam/partitions.bin
Compiling .pio/build/esp32cam/lib9fc/WiFi/ETH.cpp.o
Compiling .pio/build/esp32cam/lib9fc/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32cam/lib9fc/WiFi/WiFiAP.cpp.o
In file included from src/main.cpp:9:0:
include/frame_size.h:18:17: error: 'FRAMESIZE_240X240' was not declared in this scope
     {"240x240", FRAMESIZE_240X240},
                 ^
include/frame_size.h:21:24: error: 'FRAMESIZE_HVGA' was not declared in this scope
     {"HVGA (480x320)", FRAMESIZE_HVGA},
                        ^
include/frame_size.h:25:23: error: 'FRAMESIZE_HD' was not declared in this scope
     {"HD (1280x720)", FRAMESIZE_HD},
                       ^
src/main.cpp: In function 'void stream_text_file_gzip(const unsigned char*, size_t, const char*)':
src/main.cpp:55:73: error: no matching function for call to 'WebServer::sendContent(const char*, size_t&)'
   web_server.sendContent(reinterpret_cast<const char *>(content), length);
                                                                         ^
In file included from .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.h:25:0,
                 from src/main.cpp:4:
/home/doudz/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:130:8: note: candidate: void WebServer::sendContent(const String&)
   void sendContent(const String& content);
        ^
/home/doudz/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:130:8: note:   candidate expects 1 argument, 2 provided
src/main.cpp: In function 'void handle_root()':
src/main.cpp:80:25: error: 'class EspClass' has no member named 'getChipModel'
       {"ChipModel", ESP.getChipModel()},
                         ^
src/main.cpp:83:31: error: 'class EspClass' has no member named 'getChipCores'
       {"CpuCores", String(ESP.getChipCores())},
                               ^
src/main.cpp: In function 'void handle_snapshot()':
src/main.cpp:168:36: error: no matching function for call to 'WebServer::sendContent(const char*&, unsigned int&)'
   web_server.sendContent(fb, fb_len);
                                    ^
In file included from .pio/libdeps/esp32cam/IotWebConf/src/IotWebConf.h:25:0,
                 from src/main.cpp:4:
/home/doudz/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:130:8: note: candidate: void WebServer::sendContent(const String&)
   void sendContent(const String& content);
        ^
/home/doudz/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:130:8: note:   candidate expects 1 argument, 2 provided
src/main.cpp: In function 'void handle_flash()':
src/main.cpp:185:27: error: 'analogWrite' was not declared in this scope
   analogWrite(LED_FLASH, v);
                           ^
src/main.cpp: In function 'void on_config_saved()':
src/main.cpp:194:55: error: 'analogWrite' was not declared in this scope
   analogWrite(LED_FLASH, atoi(flash_led_intensity_val));
                                                       ^
src/main.cpp: In function 'esp_err_t initialize_camera()':
src/main.cpp:213:17: error: 'struct camera_config_t' has no member named 'fb_location'
   camera_config.fb_location = psramFound() ? CAMERA_FB_IN_PSRAM : CAMERA_FB_IN_DRAM;
                 ^
src/main.cpp:213:46: error: 'CAMERA_FB_IN_PSRAM' was not declared in this scope
   camera_config.fb_location = psramFound() ? CAMERA_FB_IN_PSRAM : CAMERA_FB_IN_DRAM;
                                              ^
src/main.cpp:213:67: error: 'CAMERA_FB_IN_DRAM' was not declared in this scope
   camera_config.fb_location = psramFound() ? CAMERA_FB_IN_PSRAM : CAMERA_FB_IN_DRAM;
                                                                   ^
src/main.cpp: In function 'void on_connected()':
src/main.cpp:242:55: error: 'analogWrite' was not declared in this scope
   analogWrite(LED_FLASH, atoi(flash_led_intensity_val));
                                                       ^
src/main.cpp: In function 'void setup()':
src/main.cpp:260:27: error: 'analogWrite' was not declared in this scope
   analogWrite(LED_FLASH, 0);
                           ^
Compiling .pio/build/esp32cam/lib9fc/WiFi/WiFiClient.cpp.o
Compiling .pio/build/esp32cam/lib9fc/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/esp32cam/lib9fc/WiFi/WiFiMulti.cpp.o
*** [.pio/build/esp32cam/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.53 seconds ==========================
```

rzeldent, 2023-01-29:
Hi Sébastien,

Have you tried to update all the packages?

If compiling via the command line, maybe ```pio pkg update``` might help?!

Regards,

Rene

rzeldent, 2023-02-03:
Hi Sebastian,

Did this solve this issue?

doudz, 2023-02-03:
Yes it works !

Thanks

closed 2023-02-03 by doudz