rzeldent/esp32cam-rtsp #37: Error UDP Packet

bloop16, 2023-02-01:
Hello,
I can`t  run the stream, always get the Error:

`17:11:43.778 > [ 46825][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
17:11:43.783 > error sending udp packet`

`17:10:56.510 > 
17:10:56.510 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:10:56.515 > configsip: 0, SPIWP:0xee
17:10:56.518 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:10:56.523 > mode:DIO, clock div:2
17:10:56.526 > load:0x3fff0030,len:1184
17:10:56.529 > load:0x40078000,len:13104
17:10:56.532 > load:0x40080400,len:3036
17:10:56.532 > entry 0x400805e4
17:10:56.948 > [     2][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
17:10:56.954 > [     3][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
17:10:56.977 > [    31][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
17:10:57.426 > [   479][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
17:10:57.451 > [   505][I][main.cpp:267] setup(): CPU Freq: 240 Mhz
17:10:57.454 > [   505][I][main.cpp:268] setup(): Free heap: 238040 bytes
17:10:57.460 > [   506][I][main.cpp:269] setup(): Starting ESP32CAM-RTSP...
17:10:57.465 > Config version: 1.3
17:10:57.468 > Config size: 392
17:10:57.468 > Loading configurations
17:10:57.471 > [iwcAll]
17:10:57.471 > |-- [iwcSys]
17:10:57.474 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
17:10:57.476 > |   |-- 'iwcApPassword' with value: <hidden>
17:10:57.482 > |   |-- [iwcWifi0]
17:10:57.482 > |   |   |-- 'iwcWifiSsid' with value: 'TP-Link_2.5Ghz'
17:10:57.488 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
17:10:57.493 > |   \-- 'iwcApTimeout' with value: '30'
17:10:57.496 > |-- [iwcCustom]
17:10:57.499 > |   \-- [settings]
17:10:57.499 > |       |-- 'config' with value: 'AI THINKER'
17:10:57.504 > |       |-- 'fd' with value: '20'
17:10:57.507 > |       |-- 'fs' with value: 'SVGA (800x600)'
17:10:57.510 > |       |-- 'fb' with value: '2'
17:10:57.513 > |       |-- 'q' with value: '12'
17:10:57.515 > |       \-- 'li' with value: '0'
17:10:57.518 > \-- [hidden]
17:10:57.521 > State changing from: 0 to 2
17:10:57.541 > [   595][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
17:10:57.641 > Setting up AP: ESP32CAM-RTSP
17:10:57.641 > Use password: <hidden>
17:10:57.643 > AP IP address: [   695][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
17:10:57.652 > 192[   695][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START        
17:10:57.657 > .168.4.1
17:10:57.660 > AP timeout (ms): 30000
17:10:57.660 > State changed from: 0 to 2
17:11:27.653 > State changing from: 2 to 3
17:11:27.667 > [ 30721][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
17:11:27.672 > [ 30722][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
17:11:27.678 > [ 30725][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Stated
17:11:27.683 > E (43814) wifi_init_default: esp_wifi_get_mac failed with 12289
17:11:27.689 > Connecting to [TP-Link_2.5Ghz] (password is hidden)
17:11:27.694 > WiFi timeout (ms): [ 30735][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped       
17:11:27.700 > [ 30729][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
17:11:27.708 > [ 30754][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
17:11:27.714 > 30000
17:11:27.731 > [ 30786][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
17:11:27.737 > [ 30792][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
17:11:27.743 > [ 30793][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:11:27.754 > [ 30794][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
17:11:27.762 > State changed from: 2 to 3
17:11:27.798 > [ 30852][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: TP-Link_2.5Ghz, BSSID: 0c:9d:92:cf:59:20, Channel: 6, Auth: WPA2_PSK
17:11:27.809 > [ 30854][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED       
17:11:28.395 > [ 31450][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.0.49
17:11:28.401 > [ 31450][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
17:11:28.409 > [ 31453][D][WiFiGeneric.cpp:996] _eventCallback(): STA IP: 192.168.0.49, MASK: 255.255.255.0, GW: 192.168.0.2
17:11:28.418 > WiFi connected
17:11:28.421 > IP address: 192.168.0.49
17:11:28.421 > State changing from: 3 to 4
17:11:31.521 > [ 34575][W][ESPmDNS.cpp:222] queryService(): No results found!
17:11:31.526 > Active mDNS services: 0
17:11:31.526 > Accepting connection
17:11:31.529 > [ 34577][V][main.cpp:238] on_connected(): on_connected
17:11:31.537 > [ 34591][I][ArduinoOTA.cpp:141] begin(): OTA server at: esp32-30c6f7207d4c.local:3232     
17:11:31.542 > [ 34592][V][main.cpp:221] start_rtsp_server(): start_rtsp_server
17:11:31.548 > [ 34594][V][main.cpp:200] initialize_camera(): initialize_camera
17:11:31.554 > [ 34599][I][main.cpp:201] initialize_camera(): Camera config: AI THINKER
17:11:31.562 > [ 34606][I][main.cpp:203] initialize_camera(): Frame size: SVGA (800x600)
17:11:31.567 > [ 34612][I][main.cpp:205] initialize_camera(): Frame buffers: 2
17:11:31.573 > [ 34618][I][main.cpp:207] initialize_camera(): JPEG quality: 12
17:11:31.578 > [ 34624][I][main.cpp:209] initialize_camera(): Frame rate: 20 ms
17:11:31.796 > [ 34851][I][main.cpp:229] start_rtsp_server(): Camera initialized
17:11:31.802 > [ 34851][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=554, ...)
17:11:31.807 > [ 34853][I][rtsp_server.cpp:9] rtsp_server(): Starting RTSP server
17:11:31.815 > State changed from: 3 to 4
17:11:38.886 > Creating TSP streamer
17:11:38.886 > Created streamer width=800, height=600
17:11:38.888 > Creating RTSP session
17:11:39.885 > RTSP received OPTIONS
17:11:40.930 > RTSP received DESCRIBE
17:11:42.576 > RTSP received SETUP
17:11:43.642 > RTSP received PLAY
17:11:43.764 > [ 46818][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
17:11:43.769 > error sending udp packet`

What am I doing wrong?

Thank`s for helping!

rzeldent, 2023-02-01:
Hi bloob,

First time I see this.  See [the list of error codes](https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip2/include/lwip/errno.h)

However, values for size etc.. seem reasonable so unfortunately cannot tell you what you're doing wrong...

Error 12 is ENOMEM, but the heap looks good. So maybe lowering the image size or buffer might help?!
Maybe an update of the arduino esp32 environment?

Let me know or update the issue if you find something?!

bloop16, 2023-02-02:
Hi rzeldent,
I think I could enclose the issue. 
Sry, first time I’m using a Sketch like this. 

I’m triggering the rtsp stream with motioneye, after erasing the board compliantly and build the sketch again it seams to work. Also I had to set the camera options in motioneye to tcp. Has been set to udp. 

But I haven’t been able to open the connection over the browser and iFrame. 

A mother problem i have is that the camera ssh log is writhing a message that the picture is send about every 170ms and writing an error that it’s running out off memory. (Sry not at home) But stays working. 
I’m not able to change the frame rate. The camera isn’t reacting on a change in motioneye. 
Is it possible to change the rate off sending frames?

thy for helping
bloop

rzeldent, 2023-02-03:
Hi bloob,

The framerate can be changed in the web GUI. See the section about [setting up](https://github.com/rzeldent/esp32cam-rtsp#setting-up-the-esp32cam-rtsp).

You could consider, as the rate is not 170 ms, to set it to 200ms, so 5 frames per second. Maybe this gives the esp some time to "breathe"

 Kind regards,

Rene

bloop16, 2023-02-05:
Thanks again!
I´ve tried to reach the /config site, but always getting the answer Server already running.
What can i do to reach the config site again?

rzeldent, 2023-02-05:
I suspect you're connecting to motion eye web server. Is this possible?

You can see the IP of the ESP32CAM in the logging when it is booting and
connected to a serial terminal.
Look for the line: [  8056][D][WiFiGeneric.cpp:991] _eventCallback(): STA
IP: *192.168.1.194*, MASK: 255.255.255.0, GW: 192.168.50.1
Here in bold is the address.of the ESP32 CAM webserver, probably another
address in your setup.

Connect using a browser on the same subnet to this address with a
web browser using HTTP. Here you should be able to log in and configure the
device.

bloop16, 2023-02-05:
No, I’m connecting to the Cam IP with http://192.168.0.49/config. The motioneye connection is closed. A reboot off the cam also didn’t help.

SteveWiFi, 2023-02-05:
Just stopping by to say that I'm also affected by this...

I installed the latest PlatformIO using pip and set it all up using the command line (not that I think it'll make a difference).

I am wondering if there's been a change in the ESP32/Arduino code which has caused this?

rzeldent, 2023-02-05:
Hi Steve/bloop,

I just did a complete rebuild (using platformIo), retested the current develop with the latest espressIf/Arduino code and I do not have any issues. Tested the build-in snapshot url and tested with VLC. No problems with both.

Could you explain a bit more what you're doing with motioneye? Can you confirm it works with VLC?

Kind regards,

Rene

SteveWiFi, 2023-02-05:
I'm not doing anything with Motioneye ... 

I built the code and installed it to my ESP32-CAM (ESP32-S, AI Thinker Camera), used VLC to connect to the RTSP stream and got several pages of the UDP errors ( [ 46818][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12 ) as per the initial issue.

I have two of them and can confirm that it happens on both, so I can rule it out being a board issue.

rzeldent, 2023-02-05:
Hi Steve,

Yes, I also saw a few of them but everything works normal; streaming is
working fine.
I do not know where they originate from but it does not seem to be a
problem... Is it?

Rene

SteveWiFi, 2023-02-05:
Hi Rene,

Given that I replicated it on both boards, I could only guess that there's a bug in one of the dependencies I used to build it.

I was hoping we could compare which packages we built it with to narrow it down.

Drizzt321, 2023-02-06:
So I'm seeing the exact same issue, recent checkout and using VSCode/PIO as per your instructions on the README. Trying to play with VLC, getting lots of lines of that same UDP error.

My CPU model `ESP32-D0WDQ5 rev. 3`, 4MB PSRAM, 640x480 resolution, 100ms frame rate (10fps), 2 buffers, JPEG quality 12, using WROVER KIT camera type. Free Heap is 131.20 KB, Max free block 75.99 KB. 

When I hit the `/snapshot` endpoint, I get a JPG, but it sometimes takes multiple seconds to load up. According to the Network tab in debug mode, it can be >1s to upwards of 8 or 9 seconds to completely receive/display the image. Seems split between Waiting for server response, and Content download.  So I'm thinking something is just taking waaaaay too long, maybe the JPG library? Maybe initiating the read from the camera? 

I'm watching the log output via Serial Monitor, not seeing any errors.

PlatformIO is v3.0.0
MS C/C++ for Visual Studio Code, v1.13.0
PIO esp32cam dependencies `C:\Users\drizzt\.platformio\penv\Scripts\platformio.exe pkg list --environment esp32cam `
```
Platform espressif32 @ 6.0.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20006.221224 (required: platformio/framework-arduinoespressif32 @ ~3.20006.0) 
├── tool-esptoolpy @ 1.40400.0 (required: platformio/tool-esptoolpy @ ~1.40400.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5)

Libraries
├── IotWebConf @ 3.2.1 (required: prampec/IotWebConf @ ^3.2.1)
├── Micro-RTSP @ 0.1.6 (required: geeksville/Micro-RTSP @ ^0.1.6)
└── micro-moustache @ 1.0.0 (required: rzeldent/micro-moustache @ ^1.0.0)
```

Any more I can do to help or check, let me know so this can be tracked down.

rzeldent, 2023-02-06:
Hi Drizz,

Strange, did you do a pull the repo? The automatic CI build is working fine and just retested it with a fresh clone.
But yes, some code has been replaced recently that was already available in a library. The current develop should compile with micro-moustache 1.0.1.

Kind regards,

Rene

Drizzt321, 2023-02-06:
@rzeldent I hadn't updated from repo to local, just did, and that fixed the compile issue. 

Hasn't fixed the UDP error unfortunately, though.

Drizzt321, 2023-02-07:
So, got an RTSP stream to work, 1 buffer, 640x480, quality 35 (is 1 low or high?), 100ms frame rate (10fps). 

However it tends to stutter, seems that the CPU can't keep up when the inter-frame changes seem to be a lot?

Strangely, when I tried 1280x1024 q=15 (lower seems to be better quality), I did get that to manage to work. It started to connect, gave the UDP error, but I let VLC continue, and it re-initialized and then started to playback the video. Something really strange seems to be going on.

While it's running, I checked, Free Heap: 128.42 KB, Max free blocks: 75.99 KB. Does run kinda slow/jerky. Trying at 33ms frame rate (30.3fps), sometimes slightly smoother, mostly about as jerky. I wonder if it's the CPU readout, or if the current compile only is using 1 of the 2 cores rather than spreading the load, or what. 

When I switched back to 2 buffers, I got the UDP error spamming the output. 

Just some more data, hope this helps some. Really would love to get this to work, this is much better sized than an RPi Zero 2 W + HQ camera module for my needs. 

```
00:27:31.404 > Rebooting...
00:27:31.404 > ets Jul 29 2019 12:21:46
00:27:31.406 >
00:27:31.406 > rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
00:27:31.412 > configsip: 0, SPIWP:0xee
00:27:31.412 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:27:31.420 > mode:DIO, clock div:2
00:27:31.420 > load:0x3fff0030,len:1184
00:27:31.423 > load:0x40078000,len:13104
00:27:31.426 > load:0x40080400,len:3036
00:27:31.430 > entry 0x400805e4
00:27:31.843 > [     2][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
00:27:31.849 > [     3][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
00:27:31.872 > [    31][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
00:27:32.321 > [   480][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
00:27:32.348 > [   507][I][main.cpp:267] setup(): CPU Freq: 240 Mhz
00:27:32.351 > [   507][I][main.cpp:268] setup(): Free heap: 238040 bytes
00:27:32.356 > [   507][I][main.cpp:269] setup(): Starting ESP32CAM-RTSP...
00:27:32.362 > Config version: 1.3
00:27:32.364 > Config size: 392
00:27:32.364 > Loading configurations
00:27:32.367 > [iwcAll]
00:27:32.367 > |-- [iwcSys]
00:27:32.370 > |   |-- 'iwcThingName' with value: 'FrontDoorCam'
00:27:32.373 > |   |-- 'iwcApPassword' with value: <hidden>
00:27:32.378 > |   |-- [iwcWifi0]
00:27:32.378 > |   |   |-- 'iwcWifiSsid' with value: 'darkwireless-iot'
00:27:32.384 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
00:27:32.389 > |   \-- 'iwcApTimeout' with value: '20'
00:27:32.392 > |-- [iwcCustom]
00:27:32.395 > |   \-- [settings]
00:27:32.395 > |       |-- 'config' with value: 'WROVER KIT'
00:27:32.400 > |       |-- 'fd' with value: '100'
00:27:32.403 > |       |-- 'fs' with value: 'SXGA (1280x1024)'
00:27:32.406 > |       |-- 'fb' with value: '1'
00:27:32.409 > |       |-- 'q' with value: '15'
00:27:32.414 > |       \-- 'li' with value: '0'
00:27:32.417 > \-- [hidden]
00:27:32.417 > State changing from: 0 to 2
00:27:32.438 > [   597][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
00:27:32.528 > Setting up AP: FrontDoorCam
00:27:32.528 > Use password: <hidden>
00:27:32.531 > [   687][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
00:27:32.536 > [   688][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
00:27:33.097 > AP IP address: 192.168.4.1
00:27:33.097 > [  1258][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
00:27:33.102 > [  1258][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
00:27:33.110 > [  1259][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
00:27:33.116 > [  1266][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
00:27:33.125 > AP timeout (ms): 20000
00:27:33.125 > State changed from: 0 to 2
00:27:53.115 > State changing from: 2 to 3
00:27:53.127 > [ 21287][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
00:27:53.133 > [ 21288][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
00:27:53.138 > [ 21291][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
00:27:53.144 > E (24945) wifi_init_default: esp_wifi_get_mac failed with 12289
00:27:53.149 > Connecting to [[ 21301][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
[... 40 lines cut ...]
00:28:02.409 > Creating RTSP session
00:28:03.403 > RTSP received OPTIONS
00:28:04.508 > RTSP received DESCRIBE
00:28:05.612 > RTSP received SETUP
00:28:06.718 > RTSP received PLAY
00:28:09.326 > [ 37487][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
00:28:09.333 > error sending udp packet
00:28:17.342 > Creating TSP streamer
00:28:17.342 > Created streamer width=1280, height=1024
00:28:17.348 > Creating RTSP session
00:28:18.343 > RTSP received TEARDOWN
00:28:19.343 > RTSP received OPTIONS
00:28:19.344 > closing TCP socket
00:28:19.346 > closing UDP socket
00:28:19.346 > closing UDP socket
00:28:20.450 > RTSP received DESCRIBE
00:28:21.555 > RTSP received SETUP
00:28:22.659 > RTSP received PLAY
00:28:48.544 > [ 76707][E][WiFiClient.cpp:422] write(): fail on fd 55, errno: 104, "Connection reset by peer"
00:28:48.552 > client has closed the socket
00:28:48.552 > client closed socket, exiting
00:28:48.555 > closing TCP socket
00:28:48.558 > closing UDP socket
00:28:48.562 > closing UDP socket
```

rzeldent, 2023-02-09:
So it is the line:
17:11:43.764 > [ 46818][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12

12 should be ENOMEM but there is enough heap space available so strange. However, mention at [https://github.com/espressif/esp-idf/issues/6852](https://github.com/espressif/esp-idf/issues/6852) it states that an error 12 is returned when the client is disconnected. So more investigation needed?!

int WiFiUDP::endPacket(){
  struct sockaddr_in recipient;
  recipient.sin_addr.s_addr = (uint32_t)remote_ip;
  recipient.sin_family = AF_INET;
  recipient.sin_port = htons(remote_port);
  int sent = sendto(udp_server, tx_buffer, tx_buffer_len, 0, (struct sockaddr*) &recipient, sizeof(recipient));
  if(sent < 0){
    log_e("could not send data: %d", errno);
    return 0;
  }
  return 1;
}

rzeldent, 2023-02-09:
> @rzeldent <https://github.com/rzeldent> so that's ready for me to do a
> pull and recompile?
>
> Would it be helpful for me to do a Wireshark recording?

Hi Drizz,

I just merged 10 minutes ago so you can use the develop branch. So, yes,
that would be sufficient.
I'm now updating some names in the config and later this evening will merge
that also.

Enjoy!

rzeldent, 2023-02-09:
What client are you using to receive the stream? VLC?

Drizzt321, 2023-02-09:
Yes, VLC to test, then I'll be using [Frigate NVR](https://frigate.video/) for long term usage. But validating it works and positioning/view/etc with VLC first.

rzeldent, 2023-02-09:
Strange, over here it works fine with VLC on the same WiFi network. I see in the logging that your device has [PSRAM](https://thingpulse.com/esp32-how-to-use-psram/) (There is some variation in the ESP32CAM modules) so this is fine.

I see in the logging a gap of 26 seconds where no data is sent. This is strange especially because streaming should start after the play. Do you have maybe some UDP filtering in place?

00:28:22.659 > RTSP received PLAY
00:28:48.544 > [ 76707][E][WiFiClient.cpp:422] write(): fail on fd 55, errno: 104, "Connection reset by peer"

Corton, 2023-02-11:
Hey, I am also having the UDP packet issue. Camera initializes, I can access the web gui, change the settings, but if I try and connect to the rtsp stream via VLC I get the error below. I originally thought it was the new pfsense router I had setup in the last week, but I swapped back to my old router and the same thing. I can connect to all my other, non-esp32 cameras via RTSP just fine as well.

Creating TSP streamer
Created streamer width=640, height=480
Creating RTSP session
RTSP received OPTIONS
RTSP received DESCRIBE
RTSP received SETUP
RTSP received PLAY
[1938812][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
error sending udp packet
[1938813][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
error sending udp packet
[1938818][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
error sending udp packet
[1938826][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
error sending udp packet
RTSP received TEARDOWN
closing TCP socket
closing UDP socket
closing UDP socket

lejibxl, 2023-02-24:
I had the same problem, I added an external antenna and it greatly reduced the problem

WarHawk8080, 2023-03-14:
Either it's a power issue (need a 22uf capacitor on power) or too weak signal from wifi

rzeldent, 2024-02-06:
I removed the code to check for OTA updates. This is in the develop branch.
This might help to increase the performance of the camera (packets sent)
Can you notice any improvements?

rottaran, 2024-05-06:
I can add an observation: We use the LwIP networkstack on ESP32-S3 with ESP-IDF 5.1.2 but have a custom communication backend that is not related to the ESP-IDF Wi-Fi driver. Our transmit callback that handles outgoing messages from LwIP down to the phy interface always succeeds. But the iperf UDP benchmark gets this ENOMEM from sendto() after 4 datagrams and succeeds in the following calls. It fails very rarely a second time. The issue only appears if the datagram fits into one Wi-Fi frame, for example with 100 Byte datagrams and also 1400 Byte datagrams. The issue disappears if the datagram is too big and IP fragmentation is used. My conclusion is, that the UDP ENOMEM is caused by LwIP and is not related to the Wi-Fi driver and all its settings.

I walked down the sendto implementation and placed debug messages on all places where an allocation could fail and ENOMEM is generated. This lead to no addiotinal outputs so far. I must have missed something.

rzeldent, 2026-08-28:
The default buffers in the LwIP are to  small. The current  development branch contains some fixes.
The issue is not that there is no memory but resending too aggressively; should have been resolved

rzeldent, 2026-08-29:
It turns out it is not fragmented or too little heap. The issue is the LwIP buffers. They are too small and cannot be changed as the SDK is precompiled on Arduino/platformIO (sdkconfig).
I rewrote the streaming engine, added audio and srtp did extensive optimalization; see https://github.com/rzeldent/esp32cam-rtsp/pull/107  to not overflow the buffers and the issue occurs rarely on the current develop branch. If I got some positive feedback, I'll merge this to main!!!!