rzeldent/esp32cam-rtsp #44: Cannot build esp32cam-rtsp

BogdanDIA, 2023-03-02:
I am using Ubuntu 20.04 and followed the build instructions to build the esp32cam-rtsp image.
I am using the following packages:
- python 3.8.10
- PlatformIO Core, version 6.1.6

**I get the following build error on pio run:**
processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/bogdan/.platformio/packages/tool-scons/scons.py", line 88, in <module>
    import SCons.Script
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/__init__.py", line 71, in <module>
    import SCons.Action
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Action.py", line 114, in <module>
    import SCons.Errors
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Errors.py", line 31, in <module>
    import SCons.Util
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 37, in <module>
    from typing import Optional, Union
  File "/home/bogdan/.local/lib/python3.8/site-packages/typing.py", line 1359, in <module>
    class Callable(extra=collections_abc.Callable, metaclass=CallableMeta):
  File "/home/bogdan/.local/lib/python3.8/site-packages/typing.py", line 1007, in __new__
    self._abc_registry = extra._abc_registry
AttributeError: type object 'Callable' has no attribute '_abc_registry'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 72, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 32, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 12, in <module>
    import os, glob, subprocess, os.path, time, pwd, sys, requests_unixsocket
  File "/usr/lib/python3/dist-packages/requests_unixsocket/__init__.py", line 1, in <module>
    import requests
  File "/home/bogdan/.local/lib/python3.8/site-packages/requests/__init__.py", line 48, in <module>
    from charset_normalizer import __version__ as charset_normalizer_version
  File "/home/bogdan/.local/lib/python3.8/site-packages/charset_normalizer/__init__.py", line 24, in <module>
    from .api import from_bytes, from_fp, from_path, normalize
  File "/home/bogdan/.local/lib/python3.8/site-packages/charset_normalizer/api.py", line 3, in <module>
    from typing import BinaryIO, List, Optional, Set
  File "/home/bogdan/.local/lib/python3.8/site-packages/typing.py", line 1359, in <module>
    class Callable(extra=collections_abc.Callable, metaclass=CallableMeta):
  File "/home/bogdan/.local/lib/python3.8/site-packages/typing.py", line 1007, in __new__
    self._abc_registry = extra._abc_registry
AttributeError: type object 'Callable' has no attribute '_abc_registry'

Original exception was:
Traceback (most recent call last):
  File "/home/bogdan/.platformio/packages/tool-scons/scons.py", line 88, in <module>
    import SCons.Script
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/__init__.py", line 71, in <module>
    import SCons.Action
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Action.py", line 114, in <module>
    import SCons.Errors
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Errors.py", line 31, in <module>
    import SCons.Util
  File "/home/bogdan/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 37, in <module>
    from typing import Optional, Union
  File "/home/bogdan/.local/lib/python3.8/site-packages/typing.py", line 1359, in <module>
    class Callable(extra=collections_abc.Callable, metaclass=CallableMeta):
  File "/home/bogdan/.local/lib/python3.8/site-packages/typing.py", line 1007, in __new__
    self._abc_registry = extra._abc_registry
AttributeError: type object 'Callable' has no attribute '_abc_registry'
============================================================================= [FAILED] Took 0.13 seconds =============================================================================

Found some info about this error here:
https://stackoverflow.com/questions/55833509/attributeerror-type-object-callable-has-no-attribute-abc-registry
and I uninstalled typing package.

**Without typing package, I got the following error:**
File "/home/bogdan/.local/lib/python3.8/site-packages/asyncio/base_events.py", line 296

    future = tasks.async(future, loop=self)

                   ^

SyntaxError: invalid syntax

Any solution on how to fix the error?
Thanks

rzeldent, 2023-03-02:
Hi Bogdan,

I suggest to install Visual Stdio code on Ubuntu and install the PlatformIO plugin.

Regards,

Rene

BogdanDIA, 2023-03-03:
Well, was trying to do plain old CLI but it seems not to be possible.

Build and Upload worked after installing VS Code and PlatformIO plugin.

Thanks

rzeldent, 2023-03-03:
Nice! Enjoy!

closed 2023-03-03 by rzeldent