rzeldent/esp32cam-rtsp #45: Exception was unhandled

Fusseldieb, 2023-03-07:
Reboots as soon as GET /snapshot. Bummer.

```
[19:45:15]ets Jun  8 2016 00:22:57
[19:45:15]
[19:45:15]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:45:15]configsip: 0, SPIWP:0xee
[19:45:15]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:45:15]mode:DIO, clock div:2
[19:45:15]load:0x3fff0018,len:4
[19:45:15]load:0x3fff001c,len:1044
[19:45:15]load:0x40078000,len:8896
[19:45:15]load:0x40080400,len:5828
[19:45:15]entry 0x400806ac
[19:45:15][     3][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[19:45:15][     4][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
[19:45:15]E (446) esp_core_dump_flash:[19:45:15]E (450) esp_core_dump_flash: No core dump partition found!
[19:45:15][    20][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[19:45:16][   473][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[19:45:16][   494][I][main.cpp:361] setup(): CPU Freq: 240 Mhz
[19:45:16][   494][I][main.cpp:362] setup(): Free heap: 237124 bytes
[19:45:16][   494][I][main.cpp:363] setup(): Starting ESP32CAM-RTSP...
[19:45:16]Config version: 1.4
[19:45:16]Config size: 450
[19:45:16]Wrong config version. Applying defaults.
[19:45:16][iwcAll]
[19:45:16]|-- [iwcSys]
[19:45:16]|   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
[19:45:16]|   |-- 'iwcApPassword' with value: <hidden>
[19:45:16]|   |-- [iwcWifi0]
[19:45:16]|   |   |-- 'iwcWifiSsid' with value: ''
[19:45:16]|   |   \-- 'iwcWifiPassword' with value: <hidden>
[19:45:16]|   \-- 'iwcApTimeout' with value: '30'
[19:45:16]|-- [iwcCustom]
[19:45:16]|   |-- [board]
[19:45:16]|   |   \-- 'bt' with value: 'AI THINKER'
[19:45:16]|   |-- [camera]
[19:45:16]|   |   |-- 'fd' with value: '100'
[19:45:16]|   |   |-- 'fs' with value: 'VGA (640x480)'
[19:45:16]|   |   |-- 'q' with value: '12'
[19:45:16]|   |   |-- 'b' with value: '0'
[19:45:16]|   |   |-- 'c' with value: '0'
[19:45:16]|   |   |-- 's' with value: '0'
[19:45:16]|   |   |-- 'e' with value: 'Normal'
[19:45:16]|   |   |-- 'wb' with value: '1'
[19:45:16]|   |   |-- 'awbg' with value: '1'
[19:45:16]|   |   |-- 'wbm' with value: 'Auto'
[19:45:16]|   |   |-- 'ec' with value: '1'
[19:45:16]|   |   |-- 'aec2' with value: '1'
[19:45:16]|   |   |-- 'ael' with value: '0'
[19:45:16]|   |   |-- 'aecv' with value: '300'
[19:45:16]|   |   |-- 'gc' with value: '1'
[19:45:16]|   |   |-- 'agcg' with value: '0'
[19:45:16]|   |   |-- 'gcl' with value: '2X'
[19:45:16]|   |   |-- 'bpc' with value: '0'
[19:45:16]|   |   |-- 'wpc' with value: '1'
[19:45:16]|   |   |-- 'rg' with value: '1'
[19:45:16]|   |   |-- 'lenc' with value: '1'
[19:45:16]|   |   |-- 'hm' with value: '0'
[19:45:16]|   |   |-- 'vm' with value: '0'
[19:45:16]|   |   |-- 'dcw' with value: '1'
[19:45:16]|   |   \-- 'cb' with value: '0'
[19:45:16]|   \-- [io]
[19:45:16]|       \-- 'li' with value: '0'
[19:45:16]\-- [hidden]
[19:45:16]AP password was not set in configuration
[19:45:16]State changing from: 0 to 1
[19:45:16][   642][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[19:45:16]Setting up AP: ESP32CAM-RTSP
[19:45:16]With default password: <hidden>
[19:45:16]AP IP address: [   735][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[19:45:16]192.[   740][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
[19:45:16][   742][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
[19:45:16]168.4.1
[19:45:16][   749][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[19:45:16][   749][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
[19:45:16][   763][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
[19:45:16]AP password was not set.
[19:45:16]WiFi SSID was not set.
[19:45:16]Will stay in AP mode.
[19:45:16]State changed from: 0 to 1
[19:45:21]Connection to AP.
[19:45:21][  5380][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: 24:18:1d:2c:e0:66, AID: 1
[19:45:21][  5381][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[19:45:21][  5670][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[19:45:21][  5670][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[19:45:22][  7174][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[19:45:22][  7198][V][Parsing.cpp:122] _parseRequest(): method: GET url: /snapshot search: 
[19:45:22][  7198][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[19:45:22][  7200][V][Parsing.cpp:227] _parseRequest(): headerValue: 192.168.4.1
[19:45:22][  7206][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[19:45:22][  7212][V][Parsing.cpp:227] _parseRequest(): headerValue: keep-alive
[19:45:22][  7218][V][Parsing.cpp:226] _parseRequest(): headerName: Cache-Control
[19:45:22][  7224][V][Parsing.cpp:227] _parseRequest(): headerValue: max-age=0
[19:45:22][  7231][V][Parsing.cpp:226] _parseRequest(): headerName: Upgrade-Insecure-Requests
[19:45:22][  7238][V][Parsing.cpp:227] _parseRequest(): headerValue: 1
[19:45:22][  7244][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[19:45:22][  7249][V][Parsing.cpp:227] _parseRequest(): headerValue: Mozilla/5.0 (Linux; Android 10; SM-G965N) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Mobile Safari/537.36
[19:45:22][  7265][V][Parsing.cpp:226] _parseRequest(): headerName: Accept
[19:45:22][  7271][V][Parsing.cpp:227] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
[19:45:22][  7288][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
[19:45:22][  7294][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip, deflate
[19:45:22][  7301][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Language
[19:45:22][  7307][V][Parsing.cpp:227] _parseRequest(): headerValue: pt-BR,pt;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6
[19:45:23][  7316][V][Parsing.cpp:254] _parseArguments(): args: 
[19:45:23][  7321][V][Parsing.cpp:237] _parseRequest(): Request: /snapshot
[19:45:23][  7327][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[19:45:23][  7332][V][main.cpp:190] handle_snapshot(): handle_snapshot
[19:45:23]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[19:45:23]
[19:45:23]Core  1 register dump:
[19:45:23]PC      : 0x400e5533  PS      : 0x00060f30  A0      : 0x800d389c  A1      : 0x3ffb25f0  
[19:45:23]A2      : 0x00000000  A3      : 0x400d3845  A4      : 0x00000056  A5      : 0x000000be  
[19:45:23]A6      : 0x3f400de7  A7      : 0x003fffff  A8      : 0x800e5530  A9      : 0x3ffb25d0  
[19:45:23]A10     : 0x3ffc5c94  A11     : 0x3ffc6cdb  A12     : 0x0000003f  A13     : 0x0000003f  
[19:45:23]A14     : 0x3f40013b  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[19:45:23]EXCVADDR: 0x00000004  LBEG    : 0x4008c701  LEND    : 0x4008c711  LCOUNT  : 0xfffffffc  
[19:45:23]
[19:45:23]
[19:45:23]Backtrace: 0x400e5530:0x3ffb25f0 0x400d3899:0x3ffb2610 0x40179055:0x3ffb2660 0x400dfc4a:0x3ffb2680 0x400dfcf1:0x3ffb26a0 0x400dfdd7:0x3ffb26e0 0x400e0016:0x3ffb2750 0x40179835:0x3ffb27c0 0x400e4a51:0x3ffb27e0 0x400d69aa:0x3ffb2800 0x400ea089:0x3ffb2820
[19:45:23]
[19:45:23]
[19:45:23]
[19:45:23]
[19:45:23]ELF file SHA256: 69ab4b83473c45e6
[19:45:23]
[19:45:23]E (7555) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[19:45:23]Rebooting...
```

rzeldent, 2023-03-08:
Hi Valentino,

Yes indeed a bummer. Reason is that normally the camera is only initialized when the internet connection to the AP is established.
Looking at the logging, the ESP32-CAM was not yet connected to the AP but in configuration mode. So, unhandled.
If you take a look at the status page you can see that there is no IP address present.

However, I looked at it and do not see a reason to initialize the camera earlier. I changed the code in the branch https://github.com/rzeldent/esp32cam-rtsp/tree/bugfix/camera_initialization. I gave it a quick test but please try if this solved your problem.

Kind regards,

Rene

Fusseldieb, 2023-03-09:
Thank you for your response.

I forgot mentioning that I **also** tried connecting it to an AP and also got an unhandled exception with a restart.

Thank you for a quick fix, however, I don't have an ESP32Cam nearby now, therefore I can't test it.

rzeldent, 2023-03-11:
Hi Valentino,

I committed the pull request and think this problem is now resolved.
Thanks for finding and mentioning this issue!

closed 2023-03-11 by rzeldent