rzeldent/esp32cam-rtsp #47: LED Flash Brightness Issue

jkykayex, 2023-03-18:
Caveat: I Could be mistaken.

I ultimately intend to use this with HomeAssistant and integrate the LED brighness control over there, however, in my initial testing, I used chrome with the link in the instructions (http://192.168.x.xxx/flash?v=100 as the parameter (100) and the[ following test tool](https://reqbin.com/)

Both these methods returned the following responses: 

[image]

rzeldent, 2023-03-19:
Hi jkykayes,

I retested it with the current develop and at my side it is working as it should. I used RapidApi but this is yet another restclient.
The light turns on/off.

[image]

Is it possible the password you are using is not correct. The password that you entered (yourself) for the AP password in the System configuration when configuring the WiFi....

[image]

Could you please retry it?

Good luck!

rzeldent, 2023-03-19:
I checked the code and added a messager: 401 Unauthorized so it will be clearer that there is an authentication problem.
This has just been merged in the main branch PR #48

closed 2023-03-22 by rzeldent

jkykayex, 2023-03-23:
> I checked the code and added a message: 401 Unauthorized so it will be clearer that there is an authentication problem. This has just been merged in the main branch PR #48

Hey, Thanks for the replies and help - Unfortunately I failed! 

First, I changed the password here to something super easy (12345678):

[image]

Then went on to using the web tool which failed and then I tried using VS code like you with that simple password...

[image]

This also failed and in checking the LED brightness settings after doing this the LED was still sitting at 0: 
[image]