rzeldent/esp32cam-rtsp #49: Can't view stream with VLC and RTSPtoWeb

Hatry1337, 2023-03-22:
Hello. Firstly after flashing and setup I tried to open stream in VLC, but have only black screen with VLC logo in player. Logs says "error sending udp packet":
```
14:10:35.038 > Creating TSP streamer
14:10:35.038 > Created streamer width=1280, height=720
14:10:35.046 > Creating RTSP session
14:10:36.040 > RTSP received OPTIONS
14:10:37.145 > RTSP received DESCRIBE
14:10:38.250 > RTSP received SETUP
14:10:39.355 > RTSP received PLAY
14:10:52.104 > [169517][E][WiFiClient.cpp:422] write(): fail on fd 52, errno: 104, "Connection reset by peer"
14:10:52.113 > client has closed the socket
14:10:52.113 > client closed socket, exiting
14:10:52.115 > closing TCP socket
14:10:52.118 > closing UDP socket
14:10:52.123 > closing UDP socket
14:11:08.807 > Creating TSP streamer
14:11:08.807 > Created streamer width=1280, height=720
14:11:08.815 > Creating RTSP session
14:11:09.809 > RTSP received SETUP
14:11:10.916 > RTSP received PLAY
14:11:14.380 > [191793][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.386 > error sending udp packet
14:11:14.386 > [191794][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.394 > error sending udp packet
14:11:14.394 > [191799][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.400 > error sending udp packet
14:11:14.707 > [192120][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.712 > error sending udp packet
14:11:14.712 > [192121][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.721 > error sending udp packet
14:11:14.721 > [192125][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.726 > error sending udp packet
14:11:14.729 > [192134][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.735 > error sending udp packet
14:11:14.737 > [192142][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.743 > error sending udp packet
14:11:14.746 > [192154][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.751 > error sending udp packet
14:11:14.754 > [192158][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.760 > error sending udp packet
14:11:14.762 > [192166][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.768 > error sending udp packet
14:11:14.771 > [192174][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.776 > error sending udp packet
14:11:14.779 > [192182][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.785 > error sending udp packet
14:11:14.787 > [192190][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.793 > error sending udp packet
14:11:14.793 > [192198][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.801 > error sending udp packet
14:11:14.801 > [192206][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.807 > error sending udp packet
14:11:14.810 > [192214][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.815 > error sending udp packet
14:11:14.818 > [192222][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:14.823 > error sending udp packet
14:11:15.350 > [192763][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:15.355 > error sending udp packet
14:11:15.355 > [192763][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:15.363 > error sending udp packet
14:11:15.363 > [192768][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:15.369 > error sending udp packet
14:11:15.372 > [192777][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:15.377 > error sending udp packet
14:11:15.380 > [192785][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:15.386 > error sending udp packet
14:11:15.388 > [192794][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:15.394 > error sending udp packet
14:11:15.397 > [192802][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
14:11:15.402 > error sending udp packet
[... 145 lines cut ...]
]
2023/03/22 11:30:33 [RTSP/1.0 200 OK
CSeq: 4
Date: Thu, Jan 01 1970 00:22:31 GMT
Range: npt=0.000-
Session: -2147458888
RTP-Info: url=rtsp://127.0.0.1:8554/mjpeg/1/track1

]
time="2023-03-22T11:30:33Z" level=info msg="Success connection RTSP" call=Start channel=0 func=StreamServerRunStream module=core stream=2ff849bc-0244-4477-8cee-bf2020f56f78
2023/03/22 11:30:49 [TEARDOWN rtsp://192.168.13.89:554/mjpeg/1/ RTSP/1.0
CSeq: 5
User-Agent: Lavf58.76.100
Session: -2147458888

]
2023/03/22 11:30:49 [RTSP Client RTP ReadFull <nil>]
2023/03/22 11:30:49 [RTSP Client Close <nil>]
time="2023-03-22T11:30:49Z" level=error msg="Stream error restart stream stream no video" call=Restart
time="2023-03-22T11:30:51Z" level=info msg="Run stream" call=Run channel=0 func=StreamServerRunStreamDo module=core stream=2ff849bc-0244-4477-8cee-bf2020f56f78
time="2023-03-22T11:30:51Z" level=info msg="Stop stream no client" call=ClientHas channel=0 func=StreamServerRunStreamDo module=core stream=2ff849bc-0244-4477-8cee-bf2020f56f78
```

Idk if I'm doing something wrong, or this is some protocols mismatch or something like this. Snapshots working fine, also I have success on recording some stream through ffmpeg.
I used VLC 3.0.18 on Arch Linux and latest docker image of RTSPtoWeb.

rzeldent, 2023-03-25:
Hi Thomas,

I added the option to stream directly from the esp32-cam module. See the latest release.
I think this is much easier!

Kind regards,

Rene

Hatry1337, 2023-03-28:
Hi Rene, JPEG stream works like a charm, thank you! But still cannot play RTSP from VLC unfortunately :(
Also some things changed: 
- No more errors in log, only RTSP OPTIONS requests:
```
21:32:48.557 > RTSP received OPTIONS
21:33:44.604 > RTSP received OPTIONS
21:34:40.652 > RTSP received OPTIONS
21:35:36.699 > RTSP received OPTIONS
```
- Controls in VLC become clickable.

\--------------------------------------------------------
Flashed latest code from `develop` branch.

LarsNorgaard, 2023-06-07:
Same problem here. RTSP does not work.

rzeldent, 2023-06-08:
Hi Thomas/Lars,

I retested RTSP and it does work. However, if the bandwidth is too low this causes stability issues.
I scaled down the initial settings to limit the data sent by the ESP32 cam See #76. Maybe this helps.

I suggest to use a low resolution and low frame rate and crank this up slowly.

Hope this works!

[image]