rzeldent/esp32cam-rtsp #51: ESP32-Wrover Module

chanyork, 2023-03-22:
Hello,

I am trying to get this to work with my ESP-Wrover Module but does not seem to work.  Please help.
https://www.amazon.com.au/Freenove-ESP32-WROVER-Bluetooth-Compatible-Tutorials/dp/B09BC5CNHM/ref=asc_df_B09BC5CNHM/?tag=googleshopdsk-22&linkCode=df0&hvadid=463598473184&hvpos=&hvnetw=g&hvrand=5905131068816457203&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9071325&hvtargid=pla-1599847483814&psc=1

the RTSP in VLC 
[image]

rzeldent, 2023-03-25:
Hi Chanyork,

Could you please provide more details like logging and the overview screen?

Regards,

Rene

SirUli, 2023-09-17:
hi @rzeldent

i have chosen "WROVER KIT" as the pins were the same as for the Freenove Kit.

This is the stacktrace from the commandline for a snapshot. I didn't change the default settings (i.e. i didn't enable PSRAM yet):

```
23:33:48.021 > [ 39028][V][main.cpp:204] handle_snapshot(): handle_snapshot
23:33:56.017 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
23:33:56.022 >
23:33:56.022 > Core  1 register dump:
23:33:56.025 > PC      : 0x400e646b  PS      : 0x00060f30  A0      : 0x800d39f2  A1      : 0x3ffd5130
23:33:56.033 > A2      : 0x00000000  A3      : 0x00000056  A4      : 0x3f402056  A5      : 0x000000cc  
23:33:56.042 > A6      : 0x3f402fbf  A7      : 0x003fffff  A8      : 0x800e6468  A9      : 0x3ffd5110
23:33:56.047 > A10     : 0x3ffc5334  A11     : 0x3ffc63f7  A12     : 0x0000003f  A13     : 0x0000003f  
23:33:56.056 > A14     : 0x3f40205f  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
23:33:56.064 > EXCVADDR: 0x00000004  LBEG    : 0x4008b2e0  LEND    : 0x4008b2f6  LCOUNT  : 0xffffffff  
23:33:56.072 >
23:33:56.072 >
23:33:56.072 > Backtrace: 0x400e6468:0x3ffd5130 0x400d39ef:0x3ffd5150 0x4017b305:0x3ffd51a0 0x400e0a76:0x3ffd51c0 0x400e0b1d:0x3ffd51e0 0x400e0c03:0x3ffd5220 0x400e0e32:0x3ffd5290 0x4017babd:0x3ffd52f0 0x400e5989:0x3ffd5310 0x400d6dd6:0x3ffd5330 0x400eaf6d:0x3ffd5350
23:33:56.094 >
23:33:56.094 >
23:33:56.094 >
23:33:56.094 >
23:33:56.094 > ELF file SHA256: 6b1c9c5a6dc074c2
```

And then it reboots. This is the stacktrace from the commandline for a jpeg stream. I didn't change the default settings (i.e. i didn't enable PSRAM yet):

```
23:36:42.440 > [165710][V][main.cpp:234] handle_stream(): handle_stream
23:36:42.446 > [165715][V][main.cpp:241] handle_stream(): starting streaming
23:36:50.447 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
23:36:50.452 >
23:36:50.452 > Core  1 register dump:
23:36:50.455 > PC      : 0x400e646b  PS      : 0x00060130  A0      : 0x800d558c  A1      : 0x3ffd5110
23:36:50.463 > A2      : 0x00000000  A3      : 0x3f40248d  A4      : 0x0000000b  A5      : 0x000000f1
23:36:50.472 > A6      : 0x3f402fb1  A7      : 0x003fffff  A8      : 0x800e6468  A9      : 0x3ffd50f0
23:36:50.477 > A10     : 0x3ffc5334  A11     : 0x3f40248d  A12     : 0x3ffd50e4  A13     : 0x00000032  
23:36:50.485 > A14     : 0x3ffd5144  A15     : 0x00040000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
23:36:50.494 > EXCVADDR: 0x00000004  LBEG    : 0x4008b2e0  LEND    : 0x4008b2f6  LCOUNT  : 0xffffffff  
23:36:50.502 >
23:36:50.502 >
23:36:50.502 > Backtrace: 0x400e6468:0x3ffd5110 0x400d5589:0x3ffd5130 0x4017b305:0x3ffd51a0 0x400e0a76:0x3ffd51c0 0x400e0b1d:0x3ffd51e0 0x400e0c03:0x3ffd5220 0x400e0e32:0x3ffd5290 0x4017babd:0x3ffd52f0 0x400e5989:0x3ffd5310 0x400d6dd6:0x3ffd5330 0x400eaf6d:0x3ffd5350
23:36:50.524 >
23:36:50.524 >
23:36:50.524 >
23:36:50.524 >
23:36:50.524 > ELF file SHA256: 6b1c9c5a6dc074c2
23:36:50.528 >
23:36:50.706 > Rebooting...
```

Overview page:

```
ESP32
SDK Version: v4.4.5
CPU model: ESP32-D0WD-V3 rev. 3
CPU speed: 240 Mhz
CPU cores: 2
RAM size: 230.80 KB
PSRAM size: 4 MB
Flash size: 4 MB

Diagnostics
Uptime: 00:00:49
Chip temperature: 67.78 °C
RTSP sessions: RTSP server disabled
Free heap: 56.17 KB
Max free block: 35.99 KB

Peripheral
Board type: WROVER KIT
LED intensity: 0 [0-100]

Network
Host name: esp32-244cabf45330.local
Mac address: 24:4C:AB:F4:53:30
Wifi mode: AP
Access point:
  Signal strength: 0 dbm
  IPv4 address: 192.168.4.1
  IPv6 address: 
```

rzeldent, 2024-02-06:
The modules have been updated in the develop branch; also the WROVER module.
Please take a look to see if this solved the problem!

closed 2026-09-04 by github-actions[bot]