rzeldent/esp32cam-rtsp #56: WEB does not start. Module rebooting

ivanshateev, 2023-03-27:
At the first inclusion, there was a web form with parameters. I entered the name of the wifi network with a password and changed the mode to ESP32CAM. After reloading the web form does not open and is not connected to the point. It feels like something is incorrectly stored in the EEROM

Connection to AP.
[  7218][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: 80:e6:50:17:69:58, AID: 1
[  7219][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[  7247][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[  7248][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[  7395][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[  7398][V][Parsing.cpp:122] _parseRequest(): method: GET url: /connecttest.txt search: 
[  7400][V][Parsing.cpp:226] _parseRequest(): headerName: Cache-Control
[  7406][V][Parsing.cpp:227] _parseRequest(): headerValue: no-cache
[  7412][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[  7418][V][Parsing.cpp:227] _parseRequest(): headerValue: Close
[  7424][V][Parsing.cpp:226] _parseRequest(): headerName: Pragma
[  7429][V][Parsing.cpp:227] _parseRequest(): headerValue: no-cache
[  7436][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[  7441][V][Parsing.cpp:227] _parseRequest(): headerValue: Microsoft NCSI
[  7448][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[  7454][V][Parsing.cpp:227] _parseRequest(): headerValue: www.msftconnecttest.com
[  7461][V][Parsing.cpp:254] _parseArguments(): args: 
[  7466][V][Parsing.cpp:237] _parseRequest(): Request: /connecttest.txt
[  7472][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[  7477][E][WebServer.cpp:649] _handleRequest(): request handler not found
Request for www.msftconnecttest.com redirected to 192.168.4.1:80
[  7495][W][WebServer.cpp:436] send(): content length is zero
[  7761][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[  7764][V][Parsing.cpp:122] _parseRequest(): method: GET url: /connecttest.txt search: 
[  7766][V][Parsing.cpp:226] _parseRequest(): headerName: Cache-Control
[  7772][V][Parsing.cpp:227] _parseRequest(): headerValue: no-cache
[  7778][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[  7784][V][Parsing.cpp:227] _parseRequest(): headerValue: Close
[  7790][V][Parsing.cpp:226] _parseRequest(): headerName: Pragma
[  7795][V][Parsing.cpp:227] _parseRequest(): headerValue: no-cache
[  7802][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[  7808][V][Parsing.cpp:227] _parseRequest(): headerValue: Microsoft NCSI
[  7814][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[  7820][V][Parsing.cpp:227] _parseRequest(): headerValue: www.msftconnecttest.com
[  7827][V][Parsing.cpp:254] _parseArguments(): args: 
[  7832][V][Parsing.cpp:237] _parseRequest(): Request: /connecttest.txt
[  7838][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[  7844][E][WebServer.cpp:649] _handleRequest(): request handler not found
Request for www.msftconnecttest.com redirected to 192.168.4.1:80
[  7861][W][WebServer.cpp:436] send(): content length is zero
[ 37875][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[ 37876][V][Parsing.cpp:122] _parseRequest(): method: GET url: /connecttest.txt search: 
[ 37880][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[ 37886][V][Parsing.cpp:227] _parseRequest(): headerValue: Close
[ 37891][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[ 37897][V][Parsing.cpp:227] _parseRequest(): headerValue: Microsoft NCSI
[ 37904][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[ 37909][V][Parsing.cpp:227] _parseRequest(): headerValue: www.msftconnecttest.com
[ 37917][V][Parsing.cpp:254] _parseArguments(): args: 
[ 37922][V][Parsing.cpp:237] _parseRequest(): Request: /connecttest.txt
[ 37928][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[ 37933][E][WebServer.cpp:649] _handleRequest(): request handler not found
Request for www.msftconnecttest.com redirected to 192.168.4.1:80
[ 37951][W][WebServer.cpp:436] send(): content length is zero
[ 38218][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[ 38219][V][Parsing.cpp:122] _parseRequest(): method: GET url: /connecttest.txt search: 
[ 38223][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[ 38229][V][Parsing.cpp:227] _parseRequest(): headerValue: Close
[ 38234][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[ 38240][V][Parsing.cpp:227] _parseRequest(): headerValue: Microsoft NCSI
[ 38247][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[ 38252][V][Parsing.cpp:227] _parseRequest(): headerValue: www.msftconnecttest.com
[ 38260][V][Parsing.cpp:254] _parseArguments(): args: 
[ 38265][V][Parsing.cpp:237] _parseRequest(): Request: /connecttest.txt
[ 38271][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[ 38276][E][WebServer.cpp:649] _handleRequest(): request handler not found
Request for www.msftconnecttest.com redirected to 192.168.4.1:80
[... 143 lines cut ...]
[  4660][V][Parsing.cpp:227] _parseRequest(): headerValue: www.msftconnecttest.com
[  4667][V][Parsing.cpp:254] _parseArguments(): args: 
[  4672][V][Parsing.cpp:237] _parseRequest(): Request: /connecttest.txt
[  4678][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[  4684][E][WebServer.cpp:649] _handleRequest(): request handler not found
Request for www.msftconnecttest.com redirected to 192.168.4.1:80
[  4701][W][WebServer.cpp:436] send(): content length is zero
[  4966][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[  4967][V][Parsing.cpp:122] _parseRequest(): method: GET url: /connecttest.txt search: 
[  4971][V][Parsing.cpp:226] _parseRequest(): headerName: Cache-Control
[  4977][V][Parsing.cpp:227] _parseRequest(): headerValue: no-cache
[  4983][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[  4989][V][Parsing.cpp:227] _parseRequest(): headerValue: Close
[  4995][V][Parsing.cpp:226] _parseRequest(): headerName: Pragma
[  5000][V][Parsing.cpp:227] _parseRequest(): headerValue: no-cache
[  5007][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[  5013][V][Parsing.cpp:227] _parseRequest(): headerValue: Microsoft NCSI
[  5019][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[  5025][V][Parsing.cpp:227] _parseRequest(): headerValue: www.msftconnecttest.com
[  5032][V][Parsing.cpp:254] _parseArguments(): args: 
[  5037][V][Parsing.cpp:237] _parseRequest(): Request: /connecttest.txt
[  5043][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[  5049][E][WebServer.cpp:649] _handleRequest(): request handler not found
Request for www.msftconnecttest.com redirected to 192.168.4.1:80
[  5066][W][WebServer.cpp:436] send(): content length is zero

rzeldent, 2023-03-27:
Also, see that your camera doe not initialize. Try to uncheck the "use PSRAM" option and set the frame buffers to 1.

rzeldent, 2023-03-27:
Could you also enable the exception decoder in Platformio:

https://community.platformio.org/t/how-do-i-use-esp32-exception-decoder/22789

This will hopefully provider some more info...

ivanshateev, 2023-03-27:
I erased the EEROM in this way (via the Arduino IDE):
#include <EEPROM.h>

void setup() {
  pinMode(9, OUTPUT);
  digitalWrite(9, HIGH);
  delay(2000);
  digitalWrite(9, LOW);
  
  EEPROM.begin(512);
  // Записываем 0 во все 512 байт EEPROM
  for (int i = 0; i < 512; i++) {
    EEPROM.write(i, 0);
    
  }

  pinMode(9, OUTPUT);
  digitalWrite(9, HIGH);
  delay(2000);
  digitalWrite(9, LOW);
  EEPROM.end();
}

void loop() {
}

But now the next one. 
JPEG Motion stream - working. 
RTSP camera stream - Error. (VLC cannot open MRL 'rtsp://192.168.4.1:554/mjpeg/1'. Look for more details in the log file.)

ivanshateev, 2023-03-27:
In the diagnostics on the web page it says - RTSP is disabled. How to turn it on?

Connected to point: ESP32CAM-RTSP.

ESP32 SDK Version: v4.4.4
CPU model: ESP32-D0WDQ5 rev. 1
CPU speed: 240 Mhz
CPU cores: 2
RAM size: 264.31 KB
PSRAM size: 4 MB
Flash size: 4 MB

Diagnostics
Uptime: 00:01:57
RTSP sessions: RTSP server disabled
Free heap: 55.88 KB
Max free block: 35.99 KB

rzeldent, 2023-03-27:
RTSP is only enabled if it is connected to an access point. The 192.168.4.1 is only for configuration/testing.

ivanshateev, 2023-03-27:
Everything is working.  Thank you.

closed 2023-03-27 by ivanshateev