rzeldent/esp32cam-rtsp #57: Multiclient

RichMar, 2023-03-31:
I like your solution for this camera. Are you ever planning a multi-client extension?

rzeldent, 2023-04-01:
Hi Rich,

You mean that more than one client can connect and watch at the same time?
This is present for RTSP (not for Motion JPEG). However, the power of the ESP32 has severe limitations and as you can read in the issues, is already reaching its limits...

Enjoy!

Rene

kvikindi, 2023-05-19:
Hey there RichMar - You might be able to use something like [go2rtc](https://github.com/AlexxIT/go2rtc) to _split_ the stream. It would however require go2rtc to be running on a separate machine, and it's a bit of a handful wrapping your head around it.

Let us know how it goes!
Ragnar

closed 2023-08-28 by rzeldent