rzeldent/esp32cam-rtsp #58: Can't get initial webpage to come up

craytb, 2023-04-02:
I have performed the latest build, uploaded via an ESPCAM-MB to an ESP32cam (AI Thinker).  I see the AccessPoint ESP32cam-RSTP,
and connect. Once connected, it times out on connecting to 192.168.4.1.  For the life of me, I don't know what I have done wrong.  Any suggestions?  Used Platform IO (VScode) via a windows computer.

Swat79, 2023-04-06:
Ive got exactly the same issue. The AP starts, and after connection, I just see a blank white page.

Swat79, 2023-04-06:
You can try http://192.168.4.1/config to connect to the configuration page.

rzeldent, 2023-04-06:
Hi swat79,

Thanks for answering to this question!
It depends also on your OS and default browser. When connected to a new AP and the key has been entered your OS uses your browser to go to a few predefined url's on the AP to connect.

tema-mazy, 2023-04-08:
same issue /

`12:33:00.105 > [ 41678][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
12:33:00.113 > [ 41679][V][Parsing.cpp:122] _parseRequest(): method: GET url: /bootstrap.min.css search: 
12:33:00.121 > [ 41683][V][Parsing.cpp:226] _parseRequest(): headerName: Host
12:33:00.127 > [ 41688][V][Parsing.cpp:227] _parseRequest(): headerValue: 192.168.4.1
12:33:00.133 > [ 41694][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
12:33:00.139 > [ 41700][V][Parsing.cpp:227] _parseRequest(): headerValue: keep-alive
12:33:00.145 > [ 41707][V][Parsing.cpp:226] _parseRequest(): headerName: Accept
12:33:00.151 > [ 41712][V][Parsing.cpp:227] _parseRequest(): headerValue: text/css,*/*;q=0.1
12:33:00.158 > [ 41720][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
12:33:00.164 > [ 41725][V][Parsing.cpp:227] _parseRequest(): headerValue: Mozilla/5.0 (iPhone; CPU iPhone OS 16_3_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Mobile/15E148
12:33:00.179 > [ 41740][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Language
12:33:00.185 > [ 41747][V][Parsing.cpp:227] _parseRequest(): headerValue: en-GB,en;q=0.9
12:33:00.192 > [ 41753][V][Parsing.cpp:226] _parseRequest(): headerName: Referer
12:33:00.197 > [ 41759][V][Parsing.cpp:227] _parseRequest(): headerValue: http://192.168.4.1/
12:33:00.204 > [ 41766][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
12:33:00.211 > [ 41772][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip, deflate
12:33:00.217 > [ 41779][V][Parsing.cpp:254] _parseArguments(): args: 
12:33:00.222 > [ 41784][V][Parsing.cpp:237] _parseRequest(): Request: /bootstrap.min.css
12:33:00.229 > [ 41790][V][Parsing.cpp:238] _parseRequest():  Arguments: 
12:33:00.234 > [ 41796][W][WebServer.cpp:436] send(): content length is zero
12:33:10.359 > [ 51932][E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
12:33:10.365 > [ 51933][I][WiFiClient.cpp:549] connected(): Unexpected: RES: 0, ERR: 12
`

just commented out bootstrap.css in html and everything is ok

rzeldent, 2023-04-10:
Hi tema,

Do you think it is a good idea to ditch bootstrap? Not as nice but maybe more practical?

Rene

pchaussalet, 2023-04-11:
@rzeldent there might be a way to make things look good enough without having to rely on the whole bootstrap css. If I find some time in the coming days, I can take a look and see if I can open a PR, if it works for you.
(btw, congrats, at last an embedded ui that doesn't make my eyes bleed... 😄 )

tema-mazy, 2023-04-11:
> Hi tema,
> 
> Do you think it is a good idea to ditch bootstrap? Not as nice but maybe more practical?
> 
> Rene

I think it is overweight. 200kb CSS for such small page ...

rzeldent, 2023-04-16:
Created a PR myself to have the CSS in the HTML.

Does this work for you?

closed 2023-05-06 by rzeldent