rzeldent/esp32cam-rtsp #60: Not saving SSID and pass.

ChuckRockTheReal, 2023-04-05:
I've uploaded to the esp32cam, login in to the AP, I change the SSID and password for my network, apply the new setting. It shows them in the log, but it just goes back to the default and does not request the ip reservation I had created in my router.

any ideas?

rzeldent, 2023-04-06:
Hi Chuck,

Most of the time this is because no connection could be made with the provided credentials. Then it defaults back to AP mode.
Another possibility is that the SSID is on 5Ghz (ESP32 only supports 2.4Ghz) so no connection could be made...

Let me know if this helps/works now...

Kind regards
Rene

alior101, 2023-04-09:
I had the same issues, then I looked carefully at the logs and saw a space added to my AP SSID name... It could be from the html config page. Anyway, making sure there is no spaces in the text box where I put the SSID name and making sure that the SSID is correct in the logs fixed the issue for me.. Good luck..

rzeldent, 2023-04-14:
Chuck? Any progress?

closed 2023-05-06 by rzeldent