rzeldent/esp32cam-rtsp #64: cannot get it to work

koraysels, 2023-04-26:
I cannot connect to the espcam afer the initial configuration is done. these are the logs:

what is going on ? 

```
19:40:32.594 > Rebooting...
19:40:32.597 > ets Jul 29 2019 12:21:46
19:40:32.600 >
19:40:32.600 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:40:32.603 > configsip: 0, SPIWP:0xee
19:40:32.606 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:40:32.611 > mode:DIO, clock div:2
19:40:32.614 > load:0x3fff0030,len:1184
19:40:32.616 > load:0x40078000,len:13132
19:40:32.619 > load:0x40080400,len:3036
19:40:32.623 > entry 0x400805e4
19:40:33.044 > [     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ..��X
19:40:33.049 > [     4][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
19:40:33.054 > [    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
19:40:33.504 > [   462][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
19:40:33.536 > [   496][I][main.cpp:396] setup(): CPU Freq: 240 Mhz
19:40:33.539 > [   497][I][main.cpp:397] setup(): Free heap: 244196 bytes
19:40:33.544 > [   497][I][main.cpp:398] setup(): SDK version: v4.4.2
19:40:33.550 > [   500][I][main.cpp:399] setup(): Starting ESP32CAM-RTSP...
19:40:33.556 > [   506][V][main.cpp:404] setup(): PSRAM found and initialized
19:40:33.564 > Config version: 1.5
19:40:33.564 > Config size: 455
19:40:33.564 > Loading configurations
19:40:33.567 > [iwcAll]
19:40:33.567 > |-- [iwcSys]
19:40:33.567 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
19:40:33.574 > |   |-- 'iwcApPassword' with value: <hidden>
19:40:33.578 > |   |-- [iwcWifi0]
19:40:33.578 > |   |   |-- 'iwcWifiSsid' with value: 'smelskes'
19:40:33.584 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
19:40:33.586 > |   \-- 'iwcApTimeout' with value: '30'
19:40:33.593 > |-- [iwcCustom]
19:40:33.593 > |   |-- [board]
19:40:33.595 > |   |   \-- 'bt' with value: 'ESP32CAM'
19:40:33.598 > |   |-- [camera]
19:40:33.601 > |   |   |-- 'fd' with value: '100'
19:40:33.603 > |   |   |-- 'fs' with value: 'HD (1280x720)'
19:40:33.606 > |   |   |-- 'q' with value: '12'
19:40:33.610 > |   |   |-- 'eps' with value: '1'
19:40:33.611 > |   |   |-- 'fb' with value: '1'
19:40:33.615 > |   |   |-- 'b' with value: '0'
19:40:33.617 > |   |   |-- 'c' with value: '0'
19:40:33.620 > |   |   |-- 's' with value: '0'
19:40:33.622 > |   |   |-- 'e' with value: 'Normal'
19:40:33.628 > |   |   |-- 'wb' with value: '1'
19:40:33.633 > |   |   |-- 'awbg' with value: '1'
19:40:33.634 > |   |   |-- 'wbm' with value: 'Auto'
19:40:33.636 > |   |   |-- 'ec' with value: '1'
19:40:33.639 > |   |   |-- 'aec2' with value: '1'
19:40:33.642 > |   |   |-- 'ael' with value: '0'
19:40:33.645 > |   |   |-- 'aecv' with value: '300'
19:40:33.648 > |   |   |-- 'gc' with value: '1'
19:40:33.653 > |   |   |-- 'agcg' with value: '0'
19:40:33.656 > |   |   |-- 'gcl' with value: '2X'
19:40:33.659 > |   |   |-- 'bpc' with value: '0'
19:40:33.661 > |   |   |-- 'wpc' with value: '1'
19:40:33.665 > |   |   |-- 'rg' with value: '1'
19:40:33.667 > |   |   |-- 'lenc' with value: '1'
19:40:33.670 > |   |   |-- 'hm' with value: '0'
19:40:33.673 > |   |   |-- 'vm' with value: '0'
19:40:33.675 > |   |   |-- 'dcw' with value: '1'
19:40:33.678 > |   |   \-- 'cb' with value: '0'
19:40:33.681 > |   \-- [io]
19:40:33.684 > |       \-- 'li' with value: '0'
19:40:33.686 > \-- [hidden]
19:40:33.686 > [   645][V][main.cpp:282] initialize_camera(): initialize_camera
19:40:33.692 > [   645][I][main.cpp:283] initialize_camera(): Camera config: ESP32CAM
19:40:33.704 > [   650][I][main.cpp:289] initialize_camera(): Frame size: HD (1280x720)
19:40:33.706 > [   656][I][main.cpp:291] initialize_camera(): JPEG quality: 12
19:40:33.712 > [   662][I][main.cpp:292] initialize_camera(): Frame duration: 100 ms
19:40:33.717 > [   668][I][main.cpp:296] initialize_camera(): Enable PSRAM: 1
19:40:33.726 > [   674][I][main.cpp:297] initialize_camera(): Frame buffers: 1
19:40:33.728 > [   679][I][main.cpp:303] initialize_camera(): PSRAM enabled!
19:40:33.758 > E (710) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
19:40:33.763 > [   718][E][main.cpp:452] setup(): Failed to initialize camera: 0x105. Type: ESP32CAM, frame size: HD (1280x720), frame rate: 100 ms, jpeg quality: 12
19:40:33.779 > State changing from: 0 to 2
19:40:33.798 > [   758][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
19:40:33.891 > Setting up AP: ESP32CAM-RTSP
19:40:33.891 > Use password: <hidden>
19:40:33.894 > [   851][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
19:40:33.899 > [   853][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
19:40:34.560 > AP IP address: 192.168.4.1
19:40:34.560 > [  1523][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
19:40:34.566 > [  1523][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
19:40:34.574 > [  1524][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
19:40:34.583 > [  1530][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
19:40:34.592 > AP timeout (ms): 30000
19:40:34.592 > State changed from: 0 to 2
19:41:04.583 > State changing from: 2 to 3
19:41:04.592 > [ 31553][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
19:41:04.599 > [ 31554][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
19:41:04.603 > [ 31558][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
19:41:04.609 > Camera probe failed with error 0x105E (45509) wifi_init_default: esp_wifi_get_mac failed with 12289
19:41:04.617 > Connecting to [smelskes[ 31570][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
19:41:04.625 > [ 31561][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
19:41:04.634 > [ 31584][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
19:41:04.639 > ] (password is hidden)
19:41:04.642 > WiFi timeout (ms): 30000
19:41:04.659 > [ 31620][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
19:41:04.665 > [ 31626][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
19:41:04.670 > [ 31627][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
19:41:04.681 > [ 31628][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
19:41:04.692 > State changed from: 2 to 3
19:41:04.714 > [ 31675][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: smelskes, BSSID: b0:be:76:ea:73:16, Reason: 202
19:41:04.722 > [ 31676][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
19:41:04.730 > [ 31683][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 202 - AUTH_FAIL
19:41:04.739 > [ 31690][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
19:41:04.744 > [ 31698][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
19:41:04.830 > [ 31791][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: smelskes, BSSID: b0:be:76:ea:73:16, Channel: 4, Auth: WPA2_PSK
19:41:04.841 > [ 31792][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
19:41:07.299 > [ 34261][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.1.181
19:41:07.305 > [ 34261][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
19:41:07.313 > [ 34264][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.1.181, MASK: 255.255.255.0, GW: 192.168.1.1
19:41:07.322 > WiFi connected
19:41:07.324 > IP address: 192.168.1.181
19:41:07.329 > State changing from: 3 to 4
19:41:10.425 > Active mDNS services: 3
19:41:10.425 > Accepting connection
19:41:10.428 > [ 37388][V][main.cpp:358] on_connected(): on_connected
19:41:10.439 > [ 37400][I][ArduinoOTA.cpp:141] begin(): OTA server at: esp32-e05a1bacbe30.local:3232
19:41:10.444 > [ 37401][E][main.cpp:365] on_connected(): Not starting RTSP server: camera not initialized
19:41:10.453 > State changed from: 3 to 4
```

rzeldent, 2023-04-27:
Hi Koray,

There is a camera initialization error. If the camera cannot be initialized, the rtsp server will nog start.
There are more people having this issue: https://github.com/rzeldent/esp32cam-rtsp/issues/31.
Are the board settings correct, camera mounted correctly, faulty hardware?

Good luck!

rzeldent, 2023-05-06:
Hi Koray,

Any progress?

closed 2023-05-30 by rzeldent

regystro, 2023-07-09:
As I can see your camera Type is set to ESP32CAM, please, have a look at https://github.com/rzeldent/esp32cam-rtsp/issues/31#issuecomment-1627570916 in case it solves your problem.