rzeldent/esp32cam-rtsp #67: how do I set an permanent ip for the camera

havingfu, 2023-05-22:
in configuration i want to add permanent IP for the cam, so the RTSP stream will run on that URL (and maybe run a couple of cameras with this code)
can I set the IP maybe through the code? can't find a way
thanks

havingfu, 2023-05-23:
thanks for responding,
I prefer the code version so if the camera reboots it will set the IP from the code again

on what file do i put the code above?

havingfu, 2023-05-29:
# finally figured out how to solve this:
I went to the file names `IotWebConf.cpp` and searched for this line: `WiFi.begin(ssid, password);` (it's around the end of the file)

then i changed the entire void function named **connectWifi** to this full code:

```cpp
void IotWebConf::connectWifi(const char* ssid, const char* password)
{
  IPAddress local_IP(192, 168, 130, 44);
  IPAddress gateway(192, 168, 130, 83);
  IPAddress subnet(255, 255, 255, 0);

  // Connect to WiFi
  WiFi.begin(ssid, password);
  WiFi.config(local_IP, gateway, subnet);
}
```
- [x] Then I reuploaded the code and it worked perfectly.

closed 2023-05-29 by havingfu

Wauro21, 2025-06-20:
> # finally figured out how to solve this:
> 
> I went to the file named `IotWebConf.cpp` and searched for this line: `WiFi.begin(ssid, password);` (it's around the end of the file)
> 
> then i changed the entire void function named **connectWifi** to this full code:
> 
> void IotWebConf::connectWifi(const char* ssid, const char* password)
> {
>   IPAddress local_IP(192, 168, ?, ?);
>   IPAddress gateway(192, 168, ?, ?);
>   IPAddress subnet(255, 255, 255, 0);
> 
>   // Connect to WiFi
>   WiFi.begin(ssid, password);
>   WiFi.config(local_IP, gateway, subnet);
> }
> 
> > **make sure you change the local_IP and gateway for your network!**
> 
>     * [x]  Then I reuploaded the code and it worked perfectly.

# Using setWiFiConnectionHandler 

Adding to this, you don't need to redefine the `IotWebConf` `connectWifi` method, the class defines a method to pass your own custom WiFi handler: `setWifiConnectionHandler`, you only need to respect the expected signature for the handler. 

So, you could define a function `staticIPHandle` and macros with the desired default values for IP, gateway and submask. The functions files are added inside the `src/` folder: 

- `WiFiHandle.hpp`:

```cpp
#ifndef WIFIHANDLE_HPP
#define WIFIHANDLE_HPP

#include "IPAddress.h"
#include "WiFi.h"

#ifndef DEFAULT_IP
#define DEFAULT_IP 192, 168, 0, 50
#endif

#ifndef DEFAULT_GATEWAY
#define DEFAULT_GATEWAY 192,168,0,1
#endif

#ifndef DEFAULT_MASK
#define DEFAULT_MASK 255,255,255,0
#endif

void staticIPHandle(const char* ssid, const char* password);

#endif
```

- `WiFiHandle.cpp`:
```cpp
#include "WiFiHandle.hpp"

void staticIPHandle(const char* ssid, const char* password)
{
    IPAddress local_IP(DEFAULT_IP);
    IPAddress gateway(DEFAULT_GATEWAY);
    IPAddress subnet(DEFAULT_SUBMASK);

    // WiFi connection
    WiFi.begin(ssid, password);
    WiFi.config(local_IP, gateway, subnet);

}
```

Then on the `main.cpp` file, just before the `iotWebConf` object calls its init routine: 
https://github.com/rzeldent/esp32cam-rtsp/blob/00635d03b73be82c0795266f299155fbf2386cef/src/main.cpp#L380

Add the following line, to pass your handler (rememeber to add the include): 
```cpp
#include "WiFiHandler.hpp"

/*
Rest of the code 
*/

// Line 380 of the main.cpp
iotWebConf.setWifiConnectionHandler(staticIPHandle); // Pass the handler to the object
// Call the init routine as always.
 iotWebConf.init();
```

This way you don't need to edit the class directly and by using the macros, you could define you own IP, Gateway and submask as compilation flags ( or automate it using pio scripts).