rzeldent/esp32cam-rtsp #68: Can't get it to compile in PlatformIO - call of overloaded 'String(uint64_t, int)' is ambiguous

januxnet, 2023-05-23:
Hi there, just did git clone the project, opened folder in VStudio + PlatformIO, saw that dependecies downloaded and then hit 'build', but it can't compile, messages below. Thank you.

Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (4.2.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20002.220503 (2.0.2) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
Compiling .pio/build/esp32cam/src/main.cpp.o
Archiving .pio/build/esp32cam/lib612/libDNSServer.a
Indexing .pio/build/esp32cam/lib612/libDNSServer.a
src/main.cpp:68:72: error: call of overloaded 'String(uint64_t, int)' is ambiguous
 auto thingName = String(WIFI_SSID) + "-" + String(ESP.getEfuseMac(), 16);
                                                                                                                                  ^
In file included from /home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:159,
                 from src/main.cpp:1:
/home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:75:18: note: candidate: 'String::String(double, unsigned char)'
         explicit String(double, unsigned char decimalPlaces = 2);
                  ^~~~~~
/home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:74:18: note: candidate: 'String::String(float, unsigned char)'
         explicit String(float, unsigned char decimalPlaces = 2);
                  ^~~~~~
/home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:73:18: note: candidate: 'String::String(long unsigned int, unsigned char)'
         explicit String(unsigned long, unsigned char base = 10);
                  ^~~~~~
/home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:72:18: note: candidate: 'String::String(long int, unsigned char)'
         explicit String(long, unsigned char base = 10);
                  ^~~~~~
/home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:71:18: note: candidate: 'String::String(unsigned int, unsigned char)'
         explicit String(unsigned int, unsigned char base = 10);
                  ^~~~~~
/home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:70:18: note: candidate: 'String::String(int, unsigned char)'
         explicit String(int, unsigned char base = 10);
                  ^~~~~~
/home/jvls/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:69:18: note: candidate: 'String::String(unsigned char, unsigned char)'
         explicit String(unsigned char, unsigned char base = 10);
                  ^~~~~~
Archiving .pio/build/esp32cam/lib652/libFS.a
Indexing .pio/build/esp32cam/lib652/libFS.a
Archiving .pio/build/esp32cam/lib34a/libWebServer.a
Indexing .pio/build/esp32cam/lib34a/libWebServer.a
Compiling .pio/build/esp32cam/libcb9/EEPROM/EEPROM.cpp.o
Compiling .pio/build/esp32cam/lib4c2/ESPmDNS/ESPmDNS.cpp.o
Compiling .pio/build/esp32cam/lib366/IotWebConf/IotWebConf.cpp.o
Compiling .pio/build/esp32cam/lib366/IotWebConf/IotWebConfMultipleWifi.cpp.o
*** [.pio/build/esp32cam/src/main.cpp.o] Error 1

rzeldent, 2023-05-23:
Hi januxnet,

I compiled it locally using the latest develop and compiled fine. I checked for platform updates and they were present.
I upgraded to 6.3.0:

Package      Current    Wanted    Latest    Type      Environments
-----------  ---------  --------  --------  --------  --------------
espressif32  6.1.0      6.3.0     6.3.0     Platform  esp32cam

I compiled again and no problem. Can you try to update your platform?

# pio pkg update

Hope this helps,

Rene

januxnet, 2023-05-23:
Perfect, I thought the environment updated automatically. Ran the command and after the update it compiled ok. Know I'm scared of the other projects that I have... hope they still compile :)

rzeldent, 2023-08-28:
Hi DrJeckyll,

Maybe just a clean project would also do the trick. Glad to hear it is resolved.

Enjoy!

closed 2023-08-28 by rzeldent