rzeldent/esp32cam-rtsp #70: timestamp

joopdo, 2023-05-26:
Sorry to raise a ticket for a simple question. 
Would I be able to project a timestamp on my camera feed with esp32cam-rtsp?

rzeldent, 2023-05-26:
Hi Joop,

I think it is not possible to produce an overlay on an image using an
esp32. It simply lacks the processing power.
The RTSP frames do have a timestamp and, together with the start of the
transmission, it should be possible to calculate the TS afterwards.

Will have to look if maybe it is possible to put a real timestamp on the
stream (using NTP)....

Rene

joopdo, 2023-05-27:
Hi Rene,

I see, thank you for explaining!

Let me know if I can test or do something.

Have a good day

rzeldent, 2023-05-28:
Hi Joop,

The timestamp that is given to each frame is taken from the function now(). See rtsp_server.cpp in the function:
bool rtsp_server::client_handler(void *arg)
```
auto now = millis();
	for (const auto &client : self->clients_)
	{
		// Handle requests
		client->session->handleRequests(0);
		// Send the frame. For now return the uptime as time marker, currMs
		client->session->broadcastCurrentFrame(now);
	}
```

This is only an uint32 and as this is in milliseconds, an absolute time is not possible. In the streamer some more magic is going on:

  ```
  uint32_t units = 90000; // Hz per RFC 2435
    m_Timestamp += (units * deltams / 1000);                             // fixed timestamp increment for a frame rate of 25fps
```

It might be possible to change the timestamp here but this is in another library I do not own...

joopdo, 2023-05-28:
Perfect! Just what I needed! Thank you sir!

closed 2023-05-28 by joopdo