rzeldent/esp32cam-rtsp #71: Working config on ESP32-S (Waveshare) with ESP-CAM-MB

saundersrg, 2023-05-27:
Thought I'd leave my experience here after much playing around with settings to get this working on a Waveshare ESP32CAM module (ESP32-S - ESP32-D0WD-V3 rev. 3). I had many of the same faults that people are posting here (reboots on connection, camera not initialised, RTSP not started, snapshot freezing) and this combination seems to work (with some catches).

1. Set the type to AI Thinker, not ESPCAM. This seemed to resolve the reboot on connection issue.
2. Disabled PSRAM. This module apparently has "4MB external PSRAM" but this enabling this seems to be causing the camera not initialized issue.

I am still having UDP issues, but I am confident that when I switch to an external antenna these issues will resolve themselves (also seen on micro-rtsp comments). If I set it to 640x480 and 1 fps, then it's very stable with no errors at some distance from the AP.

One thing I do find is that if I hit the HTTP streams or configs before RTSP, then RTSP doesn't seem to initialise properly and I have to reset the unit to get RTSP to work. If I hit RTSP first before hitting any of the HTTP protocols, RTSP works fine. Hitting HTTP after this point stops the RTSP stream, but once the HTTP request is complete, RTSP works again.

rzeldent, 2024-02-06:
Yes, the JPEG stream takes down the RTSP stream and does not seem to recover. This JPEG stream was more a gimmick than a real option; main purpose was RTSP.
I made new definitions in the develop branch for more boards. Also the OTA is taken out and think this increases the performance.

Thanks for your feedback!

rzeldent, 2024-02-25:
Hi Brett,

In the boards definition for your cam, there is an option
```
 "'-D CAMERA_CONFIG_FB_LOCATION=CAMERA_FB_IN_PSRAM'",
```

If you change that to
```
"'-D CAMERA_CONFIG_FB_LOCATION=CAMERA_FB_IN_DRAM'",
```

it should be disabled

closed 2026-09-04 by github-actions[bot]