rzeldent/esp32cam-rtsp #72: static ip, netmask, gateway and dns

SikaiGGMU, 2023-05-29:
Is it possible to manually config ip, netmask, default gateway and dns server?

rzeldent, 2023-05-29:
Hi Sikai,

No, the application is using a library to configure the networking: IoTWebConf. This is aimed at DHCP.
You can take a look at issue: #67 ;this is similar and has a solution.

Regards,

Rene

havingfu, 2023-05-29:
finally figured out how to solve this:
I went to the file names `IotWebConf.cpp` and searched for this line: `WiFi.begin(ssid, password);` (it's around the end of the file)

then i changed the entire void function named connectWifi to this full code:

`void IotWebConf::connectWifi(const char* ssid, const char* password)
{
  IPAddress local_IP(192, 168, 130, 44);
  IPAddress gateway(192, 168, 130, 83);
  IPAddress subnet(255, 255, 255, 0);

  // Connect to WiFi
  WiFi.begin(ssid, password);
  WiFi.config(local_IP, gateway, subnet);
}`

then I reuploaded the code and it worked perfectly.

rzeldent, 2023-05-29:
Hi HavingFu,

Yes, this is a way to do it and for now the fastest way. Personally I prefer a DNS reservation as this does not require code changes. But great!

Rene

closed 2023-05-29 by rzeldent