rzeldent/esp32cam-rtsp #81: Unable to compile due to "Installing espressif32"

thomas41587, 2023-07-08:
Hey guys,
I am unable to compile the source code in my VS with platformio. It always fails at the "Installing espressif32" point and does not generate an error message. Is there a possibility to manually install the dependency? Or a different log file with more information on the error?
```
Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: Installing espressif32
Error: 

 *  The terminal process "C:\Users\Thomas\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

rzeldent, 2023-07-08:
Hi Thomas,

I have no idea but suggest you remove it and install platformIO again.

Rene

rzeldent, 2023-07-09:
Any progress? Maybe take also a look at issue #68

thomas41587, 2023-07-10:
Thank you for your help. I tried to update my installation with the "pio pkg update" command. The update itself was successful but I am still unable to compile. At least I do have more output now. But still no clue what might be wrong:
```
Tool Manager: Installing platformio/framework-arduinoststm32 @ ~4.10900.0
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoststm32@4.10900.200819 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- IWatchdog @ 1.0.0
|-- Servo @ 1.1.2
|-- Wire @ 1.0
Building in release mode
RuntimeError: deque mutated during iteration:
  File "C:\Users\Thomas\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 182:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\Thomas\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Thomas\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Thomas\Desktop\Anycubic_Vyper_MB_CE_6.1d\buildroot\share\PlatformIO\scripts\offset_and_rename.py", line 24:
    marlin.relocate_vtab(LD_FLASH_OFFSET)
  File "C:\Users\Thomas\Desktop\Anycubic_Vyper_MB_CE_6.1d\buildroot\share\PlatformIO\scripts\marlin.py", line 33:
    replace_define("VECT_TAB_OFFSET", address)
  File "C:\Users\Thomas\Desktop\Anycubic_Vyper_MB_CE_6.1d\buildroot\share\PlatformIO\scripts\marlin.py", line 22:
    for define in env['CPPDEFINES']:
======================================================================================== [FAILED] Took 114.25 seconds ========================================================================================

Environment              Status    Duration
-----------------------  --------  ------------
STM32F103RET6_ac_tri_f1  FAILED    00:01:54.250
==================================================================================== 1 failed, 0 succeeded in 00:01:54.250 ====================================================================================
 *  The terminal process "C:\Users\Thomas\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

thomas41587, 2023-07-10:
Thank you for the hint. Running the update seemed to solve the issue. I am now able to compile.

closed 2023-07-10 by thomas41587