rzeldent/esp32cam-rtsp #82: Feature Request: combined streamed audio and vide

theschles, 2023-07-13:
Hi there!  I've been searching all over the Internet to see how to have combined streamed audio and video -- so that if I attach an electret microphone / breakout board (perhaps with a pre-amp) to my ESP32-CAM, I can then stream both video and audio in a single RTSP URL.

I've found this -- which streams video.

I've found other repos which show how to stream audio.

I have yet to find anything that will combine stream audio and video.

Thank you for your help!

eerison, 2023-08-31:
But is it possible to keep a communication in real-time? Do you have any example @vivekanandbv ,how to add this 2 way channel

rzeldent, 2024-02-02:
Think this is a bit out of scope for the ESP32CAM. Yes, one could add a I2S mems microphone but is already having difficulties streaming.

theschles, 2024-02-05:
If I read the manual correctly, https://github.com/AlexxIT/go2rtc can mux an audio and a video stream together.

I added Go2RTC as the sole consumer of the Esp32Cam feed when I found out the ESP32Cam apparently can only stream to a single recipient at any time.  This became quickly apparent when I tried to have two tablets watch the Esp32Cam's video stream concurrently.

Go2RTC allows multiple clients to connect to a Go2RTC-hosted stream.

rzeldent, 2024-04-16:
Hi,

This should be possible; multiple sockets are accepted... However, think this does not work because of lack of CPU...

rzeldent, 2024-04-16:
Oh, and yes, having both image and sound would be great. Some boards do have an onboard mems microphone...
Will look into this...

Kishlay-notabot, 2024-08-23:
can anyone link some libraries which support realtime *audio only* streaming through rstp or any other protocol?

rzeldent, 2026-01-08:
Hi burakk,

No unfortunately not; started to work on this two years ago (see branch 106-create-new-rtsp-server).
but lacking time...

closed 2026-01-08 by rzeldent