rzeldent/esp32cam-rtsp #86: Xiao ESP32S3 Sense?

SHKaminski, 2023-08-11:
I've been trying to get this installed on a Seeed Studio Xiao ESP32S3 Sense. I can get it to compile using all the original settings, but when I try to upload, I get the wrong chip error (It identifies a ESP32-S3 instead of a ESP32). 

When I try to change the board to the Seeed Studio ESP32S3, I can't get it to compile but am beset with many "file not found" errors for Arduino.h and WiFiServer.h.  The platform.ini is below.

I'll confess that I'm new to PlatformIO; therefore, it's entirely possible that I'm doing things wrong. But after a lot searching and attempts, I thought I'd stop and ask for directions.

`[env:esp32cam]
platform = espressif32
framework = arduino
board = seeed_xiao_esp32s3
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
board_build.partitions = min_spiffs.csv
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = 
	log2file
	time
	default
	esp32_exception_decoder
build_flags = 
	-O2
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
	-D LED_FLASH=4
	-D LED_BUILTIN=33
	-D BOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-D IOTWEBCONF_PASSWORD_LEN=64
lib_ldf_mode = deep+
lib_compat_mode = off
board_build.embed_txtfiles = 
	html/index.min.html
	html/restart.min.html
lib_deps = 
	prampec/IotWebConf @ ^3.2.1
	geeksville/Micro-RTSP @ ^0.1.6
	rzeldent/micro-moustache@^1.0.1`

rzeldent, 2023-08-12:
Hi SHKaminski,

Uploading might fail because you need to build it for the ESP32-S3 chip. This is another type of chip, see https://linuxhint.com/difference-between-esp32-s3/. The S3 has enough power to run the streaming but have never tested it; the camera connection might also be different. If you have problems uploading try setting the configuration to S3. Currently this is defined in the platformio.ini as:

[env:esp32cam]
platform = espressif32
board = esp32cam

Take a look at the choices at https://docs.platformio.org/en/latest/boards/espressif32/esp32s3box.html

Hope this helps getting it running. Please share your experiences!

Rene

BTW nice new board! Will look it I can also get my hands on one...

SHKaminski, 2023-08-14:
Thanks for the guidance. I was able to get it up and running. Here's what I ended up doing:

- I manually added the XIAO ESP32S3 Sense to lookup_camera_config.h (and set it to the default camera while I was there).
- I had to comment out the build_flag -D LED_BUILTIN=33 since the LED_BUILTIN for this board is 21. Even changing this flag to 21 caused a problem. But removing it entirely worked.
- In main.cpp, I had to comment out {"Temperature", String((temperature_sens_read() - 32) / 1.8)}. I'm not entirely sure why this caused a compile error, but it did. (Also, was there a typo in the code here? it was "temprature" in a couple of places?)

So it's working now! I am going through all the normal troubleshooting. Thanks again for you help. Here's the platformio.ini that worked:

`[env:esp32cam]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
board_build.f_cpu = 240000000L
board_build.partitions = min_spiffs.csv
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = 
	log2file
	time
	default
	esp32_exception_decoder
build_flags = 
	-O2
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
	-D LED_FLASH=8
	;-D LED_BUILTIN=21
	-D BOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-D IOTWEBCONF_PASSWORD_LEN=64
board_build.embed_txtfiles = 
	html/index.min.html
	html/restart.min.html
lib_deps = 
	prampec/IotWebConf @ ^3.2.1
	geeksville/Micro-RTSP @ ^0.1.6
	rzeldent/micro-moustache@^1.0.1`

rzeldent, 2023-08-15:
Hi SHKaminski,

Thanks for looking into this! I also ordered this board to integrate this in the build....
Maybe this board is a bit faster so more stable...

SHKaminski, 2023-08-16:
Just one follow-up note on the stability on the XIAO ESP32S3:
For me, I had a lot of stability issues, specifically a lot of dropped UDP packets. It ended up being due to a weak wifi connection, even though the access point was in the same room a couple of meters away.

I replaced the flat antenna that came with the board with a regular stick antenna, and things got much better. It's been very solid since then.

rzeldent, 2023-08-16:
Hi,

Thanks for the feedback. Do you have the changes in the pins?
Just received one but lacking time to work on this....

SHKaminski, 2023-08-16:
Sure. Here's what I have that appears to be working:

constexpr camera_config_t esp32cam_xiaoesp32s3_settings = {
    .pin_pwdn = -1,
    .pin_reset = -1,
    .pin_xclk = 10,
    .pin_sscb_sda = 40,
    .pin_sscb_scl = 39,
    .pin_d7 = 48,
    .pin_d6 = 11,
    .pin_d5 = 12,
    .pin_d4 = 14,
    .pin_d3 = 16,
    .pin_d2 = 18,
    .pin_d1 = 17,
    .pin_d0 = 15,
    .pin_vsync = 38, 
    .pin_href = 47,
    .pin_pclk = 13,
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,
    .pixel_format = PIXFORMAT_JPEG, // for streaming
    //.pixel_format = PIXFORMAT_RGB565, // for face detection/recognition
    .frame_size = FRAMESIZE_UXGA,
    .jpeg_quality = 12,
    .fb_count = 1,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY
};

rzeldent, 2023-08-16:
Hi SHKaminski,

Was reading again what you wrote 2 days ago: " I'm not entirely sure why this caused a compile error, but it did. (Also, was there a typo in the code here? it was "temprature" in a couple of places?)"

This is correct, this typo. There is an undocumented function present "temprature" that was misspelled by the Chinese developers. Correcting this will fail to link :-)

rzeldent, 2023-08-17:
Created a branch: [feature/seeed_xiao_esp32s3](https://github.com/rzeldent/esp32cam-rtsp/tree/feature/seeed_xiao_esp32s3) (Untested)

SHKaminski, 2023-08-17:
Ah, thanks for that! I guess it makes sense.
Thank you!

SHKaminski, 2023-08-17:
I'll try your fork.

FYI: With the released version, when I try to compile with "temprature" or "temperature," I get the error below. When I comment out line 129, it works.

Linking .pio\build\esp32cam\firmware.elf
c:/users/shkam/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32cam/src/main.cpp.o:(.literal._Z11handle_rootv+0x130): undefined reference to `temprature_sens_read'
c:/users/shkam/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32cam/src/main.cpp.o: in function `handle_root()':
D:\Documents\PlatformIO\esp32cam-rtsp-develop/src/main.cpp:129: undefined reference to `temprature_sens_read'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32cam\firmware.elf] Error 1

SHKaminski, 2023-08-17:
I tried the XIAO fork. It compiled and uploaded properly, but it gets stuck in a boot loop with this error:

E (224) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377060
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (196) spi_flash: Detected size(8192k) smaller than the size in the binary image header(16384k). Probe failed.

rzeldent, 2023-08-28:
Yes, I tried it and had the same. Will take a look later what's going on....

blacxcode, 2023-08-29:
does the Xiao ESP32S3 Sense support using OV5640 camera?

SHKaminski, 2023-08-29:
Yes.
https://www.seeedstudio.com/OV5640-Camera-for-XIAO-ESP32S3-Sense-With-Heat-Sink-p-5739.html

rzeldent, 2023-09-21:
Hi SHKaminski,

I'm working on a new build system to support multiple devices. Please take a look.
https://github.com/rzeldent/esp32cam-rtsp/tree/feature/seeed_xiao_esp32s3

SHKaminski, 2023-09-23:
I loaded the latest version to my Xiao ESP32S3 Sense and it appears to be running well so far. No errors or glitches. I'll let you know if anything crops up. 
Thanks for your work on this. It's a useful tool.

jenswes, 2023-10-07:
Hi, thanks for the code :-) it's nice.

I've tried with two cheap ESP32-WROVER-DEV from amazon, but run into stability issues (they are maybe too chep, have a week antenna, if they run, they are slow....)

so the Xiao ESP32S3 Sense look like a promizing solution.

When i tried to build
origin/feature/seeed_xiao_esp32s3

Processing seeed_xiao_esp32s3_sense (platform: espressif32; board: seeed_xiao_esp32s3; framework: arduino)
--------------------------------------------------------------------------------------------------------
UnknownBoard: Unknown board ID 'seeed_xiao_esp32s3'

as written here:
https://community.platformio.org/t/error-during-build-on-seeed-xiao-esp32c3-unknown-board-id/29732/2
i changed my platformio.ini like:

[env:seeed_xiao_esp32s3_sense]
#platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git
board = seeed_xiao_esp32s3
framework = arduino
platform_packages =
framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master

pio run then downloads a lot stuff

Platform Manager: espressif32@6.4.0+sha.f6ec392 has been installed!
Tool Manager: toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch5 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20011.0
Tool Manager: framework-arduinoespressif32@3.20011.230801 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40501.0
Tool Manager: tool-esptoolpy@1.40501.0 has been installed!
Library Manager: Installing prampec/IotWebConf @ ^3.2.1
Library Manager: IotWebConf@3.2.1 has been installed!
Library Manager: Installing geeksville/Micro-RTSP @ ^0.1.6
Library Manager: Micro-RTSP@0.1.6 has been installed!
Library Manager: Installing rzeldent/micro-moustache @ ^1.0.1
Library Manager: micro-moustache@1.0.1 has been installed!
Library Manager: Installing contrem/arduino-timer @ ^2.3.1
Library Manager: arduino-timer@2.3.1 has been installed!

and my build succeed :-)
i'll get Xiao ESP32S3  soon, and report back :-)

jenswes, 2023-10-10:
the Xiao ESP32S3 simply works, exept for the rtsp stream, the console shows a lot
```
[ 22853][E][WiFiUdp.cpp:185] endPacket(): could not send data: 12
error sending udp packet
```

```
Board type:     Seed Xiao ESP32S3 Sense 
SDK Version:   v4.4.5
CPU model:    ESP32-S3 rev. 0
CPU speed:    240 Mhz
CPU cores:     2
RAM size:       303.33 KB
PSRAM size:   8 MB
Flash size:      8 MB
```
ffplay http://<ip>/stream works well and smooth

rzeldent, 2023-10-17:
Have you tried decreasing the framerate? Sounds like a capacity problem...

closed 2023-10-29 by rzeldent