rzeldent/esp32cam-rtsp #87: RTSP stream teardown

curiegupta, 2023-08-14:
Hi,
I am working on a project that uses ESP32S3-EYE board and a OV2640 camera. The camera configuration parameters are kept to default and the PSRAM is enabled.  
Everything works great except that I am observing frequent disconnections on the RTSP stream. I am trying to access the RTSP stream with the help of ‘rtsp’ library in python. I also tried through VLC. At times, the stream works uninterrupted for an hour then disconnects and sometimes it gets disconnected within 2-3 minutes.
Not sure of what is causing the issue. Browsed the open/closed issues and tried a few workarounds, but all in vain.

Any suggestions or help would be greatly appreciated!

Thanks in advance.
:)

JohannesBauer97, 2023-08-25:
I can confirm the same behavior, once the web /stream page got opened, the RTSP connection is not working.

rzeldent, 2024-02-06:
OTA has been taken out in the **develop** branch this might increase performance...
Could you try to see if the esp32S3-eye board is still working correctly with the new easy configurable board definitions?

rzeldent, 2026-08-28:
The develop branch has a new version with improved streaming. Let me know if this fixes the issue!

closed 2026-11-04 by github-actions[bot]