rzeldent/esp32cam-rtsp #88: Failed to initialize camera

mobigroup, 2023-08-16:
When default resolution increased to 800x600 the firmware cannot init camera:
```
10:00:32.260 > [   432][V][main.cpp:289] initialize_camera(): initialize_camera
10:00:32.266 > [   432][I][main.cpp:290] initialize_camera(): Camera config: ESP32CAM
10:00:32.274 > [   439][I][main.cpp:296] initialize_camera(): Frame size: SVGA (800x600)
10:00:32.280 > [   445][I][main.cpp:298] initialize_camera(): JPEG quality: 14
10:00:32.285 > [   451][I][main.cpp:299] initialize_camera(): Frame duration: 200 ms
10:00:32.291 > [   457][I][main.cpp:303] initialize_camera(): Enable PSRAM: 0
10:00:32.296 > [   463][I][main.cpp:304] initialize_camera(): Frame buffers: 1
10:00:32.302 > [   468][I][main.cpp:315] initialize_camera(): PSRAM disabled
10:00:32.332 > E (476) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
10:00:32.338 > [   508][E][main.cpp:459] setup(): Failed to initialize camera: 0x105. Type: ESP32CAM, frame size: SVGA (800x600), frame rate: 200 ms, jpeg quality: 14
```

There is no way to change the resolution because web interface is not accessible.

mobigroup, 2023-08-28:
The ticket is not resolved and it’s still actual. Only when we select “AI THINKERl” as the board the firmware works with ESP32-CAM board. Probably, web interface applies the board type incorrect.

mobigroup, 2023-08-28:
That’s hard to find the actual difference in the boards from multiple manufacturers. I use this board: https://www.aliexpress.com/item/1005002066523822.html

rzeldent, 2023-08-28:
I also have one of those and it matched with the settings for the AI-Thinker. So I assume this is an AI-Thinker.
See also: https://randomnerdtutorials.com/esp32-cam-ai-thinker-pinout/

rzeldent, 2023-08-28:
Maybe the best thing is to document the boards, a picture and how they look, and provide multiple build settings.

This because there are also boards (like the Seeed Studio XIAO ESP32S3 Sense) that uses a different processor.
Cycling might help for a few types but some boards seem to hang if using the wrong configuration and the flash needs to be erased.