rzeldent/esp32cam-rtsp #89: Web access does not work

mobigroup, 2023-08-16:
The logged address 192.168.1.57 is not accessible and "http://192.168.1.57/config" requires username and password and it does not work too with empty username (it does not exists in the initial configuration and that is not documented): 

```
10:04:52.384 > State changed from: 0 to 2
10:05:22.379 > State changing from: 2 to 3
10:05:22.388 > [ 31240][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
10:05:22.394 > [ 31240][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_STOP
10:05:22.468 > [ 31318][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
10:05:22.473 > [ 31320][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
10:05:22.479 > CCamera probe failed with error 0x105E (44998) wifi_init_default: esp_wifi_get_mac failed with 12289
10:05:22.487 > onnecting to [ALEKSEI[ 31330][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
10:05:22.496 > ] (password is hidden)[ 31340][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_STOP
10:05:22.504 > 
10:05:22.504 > WiFi timeout (ms): 30000
10:05:22.521 > [ 31372][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
10:05:22.526 > [ 31379][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
10:05:22.538 > [ 31378][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
10:05:22.543 > [ 31385][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
10:05:22.551 > State changed from: 2 to 3
10:05:23.553 > [ 32404][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ALEKSEI, BSSID: f8:2e:3f:0c:2e:e0, Reason: 2
10:05:23.561 > [ 32405][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
10:05:23.570 > [ 32412][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
10:05:23.578 > [ 32418][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
10:05:23.583 > [ 32427][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
10:05:24.601 > [ 33452][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ALEKSEI, BSSID: f8:2e:3f:0c:2e:e0, Reason: 2
10:05:24.609 > [ 33453][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
10:05:24.617 > [ 33460][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
10:05:24.626 > [ 33466][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
10:05:24.631 > [ 33475][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
10:05:24.707 > [ 33557][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: ALEKSEI, BSSID: f8:2e:3f:0c:2e:e0, Channel: 6, Auth: WPA2_PSK
10:05:24.717 > [ 33559][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
10:05:26.923 > [ 35774][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.1.57
10:05:26.928 > [ 35774][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
10:05:26.937 > [ 35778][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.57, MASK: 255.255.255.0, GW: 192.168.1.1
10:05:26.945 > WiFi connected
10:05:26.948 > IP address: 192.168.1.57
10:05:26.953 > State changing from: 3 to 4
10:05:30.049 > [ 38900][W][ESPmDNS.cpp:222] queryService(): No results found!
10:05:30.054 > Active mDNS services: 0 
10:05:30.054 > Accepting connection
10:05:30.057 > [ 38902][V][main.cpp:365] on_connected(): on_connected
10:05:30.063 > [ 38915][I][ArduinoOTA.cpp:141] begin(): OTA server at: esp32-e05a1bd2b4a8.local:3232
10:05:30.069 > [ 38915][E][main.cpp:372] on_connected(): Not starting RTSP server: camera not initialized
10:05:30.077 > State changed from: 3 to 4
```

```
10:11:29.331 > [178361][V][Parsing.cpp:237] _parseRequest(): Request: /config
10:11:29.336 > [178367][V][Parsing.cpp:238] _parseRequest():  Arguments: 
10:11:29.342 > Requesting authentication.
10:11:29.345 > [178383][W][WebServer.cpp:436] send(): content length is zero
10:12:16.184 > [225222][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.1.57
10:12:16.189 > [225223][V][Parsing.cpp:122] _parseRequest(): method: GET url: /config search: 
10:12:16.197 > [225226][V][Parsing.cpp:226] _parseRequest(): headerName: Host
10:12:16.203 > [225232][V][Parsing.cpp:227] _parseRequest(): headerValue: 192.168.1.57
10:12:16.209 > [225238][V][Parsing.cpp:226] _parseRequest(): headerName: Accept
10:12:16.214 > [225244][V][Parsing.cpp:227] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
10:12:16.225 > [225254][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
10:12:16.231 > [225260][V][Parsing.cpp:227] _parseRequest(): headerValue: keep-alive
10:12:16.236 > [225267][V][Parsing.cpp:226] _parseRequest(): headerName: Upgrade-Insecure-Requests
10:12:16.245 > [225274][V][Parsing.cpp:227] _parseRequest(): headerValue: 1
10:12:16.250 > [225280][V][Parsing.cpp:226] _parseRequest(): headerName: Cookie
10:12:16.256 > [225285][V][Parsing.cpp:227] _parseRequest(): headerValue: ai_user=kMu2D|2023-08-16T03:00:25.342Z
10:12:16.265 > [225294][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
10:12:16.270 > [225300][V][Parsing.cpp:227] _parseRequest(): headerValue: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.4 Safari/605.1.15
10:12:16.286 > [225315][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Language
10:12:16.292 > [225322][V][Parsing.cpp:227] _parseRequest(): headerValue: en-GB,en;q=0.9
10:12:16.300 > [225328][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
10:12:16.306 > [225335][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip, deflate
10:12:16.311 > [225341][V][Parsing.cpp:226] _parseRequest(): headerName: Authorization
10:12:16.320 > [225348][V][Parsing.cpp:227] _parseRequest(): headerValue: Basic OjE0MTU5MjY1Mw==
10:12:16.325 > [225355][V][Parsing.cpp:254] _parseArguments(): args: 
10:12:16.331 > [225360][V][Parsing.cpp:237] _parseRequest(): Request: /config
10:12:16.336 > [225365][V][Parsing.cpp:238] _parseRequest():  Arguments: 
10:12:16.342 > Requesting authentication.
10:12:16.345 > [225381][W][WebServer.cpp:436] send(): content length is zero
```

rzeldent, 2023-08-28:
Hi mobigroup,

This seems to be an issue in the ESP library: See https://github.com/espressif/arduino-esp32/issues/7562

Is it correct you only have digits in your password?

Rene

closed 2023-09-21 by rzeldent

druckgott, 2023-11-20:
I have exaclty the same problem is there any solution?

druckgott, 2023-11-20:
I can access the first time the AP then Set my ssid and pw. The Esp go to. My Networks and get the IP 192.164.0.80 but i cant reach the confuration any more.
Always if the esp. Is in my network its Not any ore availibe. I use 3 more esps on my network with Different Software they work All.
I habe a fritzbox with a ssid with one blank inside and only numeric Passworts aß it is mostly on fritzbox.

rzeldent, 2023-11-20:
Hi druckgott,

Have you also changed the AP password? Can you still access it when using it as an AP on 192.168.4.1?

druckgott, 2023-11-20:
I changed the Passworts and i cant access the AP any more because it always switch to my  network

rzeldent, 2023-11-20:
Thats's no problem; check on the ap for the ip and cannect using that ip

druckgott, 2023-11-21:
This is what I get on console.

AP comes very short. I can connect, but after some seconds AP is not any more avaible:

```
--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
07:39:06.992 > �ets Jul 29 2019 12:21:46
07:39:06.992 >
07:39:06.992 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
07:39:06.992 > configsip: 0, SPIWP:0xee
07:39:06.992 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:39:06.992 > mode:DIO, clock div:2
07:39:06.992 > load:0x3fff0030,len:1184
07:39:06.992 > load:0x40078000,len:13132
07:39:06.992 > load:0x40080400,len:3036
07:39:06.992 > entry 0x400805e4
07:39:06.992 > [     3][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ���)�Hh␅␂␂␂␂�][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
07:39:06.992 > [    10][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
07:39:06.992 > [   462][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
07:39:06.992 > [   489][I][main.cpp:395] setup(): CPU Freq: 240 Mhz
07:39:06.992 > [   489][I][main.cpp:396] setup(): Free heap: 244340 bytes
07:39:06.992 > [   489][I][main.cpp:397] setup(): SDK version: v4.4.2
07:39:06.992 > [   492][I][main.cpp:398] setup(): Starting ESP32CAM-RTSP...
07:39:06.992 > Config version: 1.5
07:39:06.992 > Config size: 455
07:39:06.992 > Loading configurations
07:39:06.992 > [iwcAll]
07:39:06.992 > |-- [iwcSys]
07:39:06.992 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
07:39:06.992 > |   |-- 'iwcApPassword' with value: <hidden>
07:39:06.992 > |   |-- [iwcWifi0]
07:39:06.992 > |   |   |-- 'iwcWifiSsid' with value: 'UNSERNETZ MY'
07:39:06.992 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
07:39:06.992 > |   \-- 'iwcApTimeout' with value: '30'
07:39:06.992 > |-- [iwcCustom]
07:39:06.992 > |   |-- [board]
07:39:06.992 > |   |   \-- 'bt' with value: 'ESP32CAM'
07:39:06.992 > |   |-- [camera]
07:39:06.992 > |   |   |-- 'fd' with value: '100'
07:39:06.992 > |   |   |-- 'fs' with value: 'VGA (640x480)'
07:39:06.992 > |   |   |-- 'q' with value: '12'
07:39:06.992 > |   |   |-- 'eps' with value: '0'
07:39:06.992 > |   |   |-- 'fb' with value: '1'
07:39:06.992 > |   |   |-- 'b' with value: '0'
07:39:06.992 > |   |   |-- 'c' with value: '0'
07:39:06.992 > |   |   |-- 's' with value: '0'
07:39:06.992 > |   |   |-- 'e' with value: 'Normal'
07:39:06.992 > |   |   |-- 'wb' with value: '1'
07:39:06.992 > |   |   |-- 'awbg' with value: '1'
07:39:06.992 > |   |   |-- 'wbm' with value: 'Auto'
07:39:06.992 > |   |   |-- 'ec' with value: '1'
07:39:06.992 > |   |   |-- 'aec2' with value: '1'
07:39:06.992 > |   |   |-- 'ael' with value: '0'
07:39:06.992 > |   |   |-- 'aecv' with value: '300'
07:39:06.992 > |   |   |-- 'gc' with value: '1'
07:39:06.992 > |   |   |-- 'agcg' with value: '0'
07:39:06.992 > |   |   |-- 'gcl' with value: '2X'
07:39:06.992 > |   |   |-- 'bpc' with value: '0'
07:39:06.992 > |   |   |-- 'wpc' with value: '1'
07:39:06.992 > |   |   |-- 'rg' with value: '1'
07:39:06.992 > |   |   |-- 'lenc' with value: '1'
07:39:06.992 > |   |   |-- 'hm' with value: '0'
07:39:06.992 > |   |   |-- 'vm' with value: '0'
07:39:06.992 > |   |   |-- 'dcw' with value: '1'
07:39:06.992 > |   |   \-- 'cb' with value: '0'
07:39:06.992 > |   \-- [io]
07:39:06.992 > |       \-- 'li' with value: '0'
[... 33 lines cut ...]
07:39:12.697 > [  9333][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
07:39:12.697 > [  9334][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
07:39:12.713 > [  9334][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
07:39:12.713 > State changed from: 2 to 3
07:39:12.729 > [  9367][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: UNSERNETZ MY, BSSID: c0:25:06:e4:07:a0, Reason: 202
07:39:12.739 > [  9367][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:39:12.739 > [  9375][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 202 - AUTH_FAIL
07:39:12.739 > [  9381][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
07:39:12.755 > [  9390][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
07:39:12.806 > [  9456][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: AAAAAAAAAAA MY, BSSID: c0:25:06:e4:07:a0, Channel: 11, Auth: WPA2_PSK
07:39:12.822 > [  9458][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
07:39:12.838 > [  9477][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.0.80
07:39:12.839 > [  9477][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
07:39:12.855 > [  9480][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.0.80, MASK: 255.255.255.0, GW: 192.168.0.1
07:39:12.855 > WiFi connected
07:39:12.855 > IP address: 192.168.0.80
07:39:12.855 > State changing from: 3 to 4
07:39:15.962 > [ 12603][W][ESPmDNS.cpp:222] queryService(): No results found!
07:39:15.962 > Active mDNS services: 0
07:39:15.962 > Accepting connection
07:39:15.962 > [ 12604][V][main.cpp:353] on_connected(): on_connected
07:39:15.978 > [ 12618][I][ArduinoOTA.cpp:141] begin(): OTA server at: esp32-08d1f96b1758.local:3232
07:39:15.991 > [ 12618][E][main.cpp:364] on_connected(): Not starting RTSP server: camera not initialized
07:39:15.991 > State changed from: 3 to 4
```