rzeldent/esp32cam-rtsp #92: WiFi disconnection error on RTSP stream

bandaangosta, 2023-09-11:
Hi, @rzeldent . Thanks for a great project.
I'm running it on 3 ESP32-CAM devices, all cheap AI-Thinker clones. I use Onvier android app to monitor several cameras at home, 1 of them a ESP32-CAM with the esp32cam-rtsp firmware. For some reason, I have never been able to view the stream on VLC (3.0.16 on Linux Mint 21), although that is a secondary problem for me.
On one device, it runs very solid and stable (streaming to Onvier for hours until I close the app). On the other two, after seconds/minutes, stream is interrupted with a "Disconnected: RES: 0, ERR: 128" error.

_pio device monitor_ output on failure:

```
22:20:41.117 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:20:41.148 > configsip: 0, SPIWP:0xee
22:20:41.148 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:20:41.148 > mode:DIO, clock div:2
22:20:41.148 > load:0x3fff0030,len:1184
22:20:41.148 > load:0x40078000,len:13232
22:20:41.148 > load:0x40080400,len:3028
22:20:41.148 > entry 0x400805e4
22:20:41.564 > [     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
22:20:41.568 > [␀a␀␂␂␂��eյuUKkU�.W�.i␖␗.'NS␅�UKkU�.W�.�J'HU���Y.W�.'�U���Y.W�.%\K␗WJj��␅		␒R&�W+QW�Y.\&�KZX�K,\]�,'&MMW��Y](\]��equencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
22:20:41.816 > [   239][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
22:20:41.865 > [   266][V][e[   267][I][main.cpp:403] setup(): CPU Freq: 240 Mhz, 2 core(s)
22:20:41.866 > [   267][I][main.cpp:404] setup(): Free heap: 214596 bytes
22:20:41.866 > [   268][I][main.cpp:405] setup(): SDK version: v4.4.5
22:20:41.866 > [   272][I][main.cpp:406] setup(): Starting ESP32CAM-RTSP...
22:20:41.866 > [   277][V][main.cpp:411] setup(): PSRAM found and initialized
22:20:41.866 > Config version: 1.5
22:20:41.866 > Config size: 455
22:20:41.866 > Loading configurations
...
22:21:02.597 > [ 20997][V][main.cpp:356] start_rtsp_server(): start_rtsp_server
22:21:02.597 > [ 20999][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=554, ...)
22:21:02.597 > [ 21006][I][rtsp_server.cpp:9] rtsp_server(): Starting RTSP server
22:21:02.597 > State changed from: 3 to 4
22:21:57.571 > Creating TSP streamer
22:21:57.571 > Created streamer width=320, height=240
22:21:57.587 > Creating RTSP session
22:21:58.578 > RTSP received DESCRIBE
22:22:00.574 > RTSP received SETUP
22:22:02.587 > RTSP received PLAY
22:22:09.600 > RTSP received OPTIONS
22:22:20.607 > RTSP received OPTIONS
22:22:31.613 > RTSP received OPTIONS
22:22:42.619 > RTSP received OPTIONS
22:22:53.627 > RTSP received OPTIONS
22:23:04.633 > RTSP received OPTIONS
22:23:15.639 > RTSP received OPTIONS
22:23:26.661 > RTSP received OPTIONS
22:23:38.658 > RTSP received OPTIONS
22:23:48.673 > RTSP received OPTIONS
22:23:59.679 > RTSP received OPTIONS
22:24:10.685 > RTSP received OPTIONS
22:24:21.707 > RTSP received OPTIONS
22:24:32.713 > RTSP received OPTIONS
22:24:43.718 > RTSP received OPTIONS
22:24:54.724 > RTSP received OPTIONS
22:25:06.736 > RTSP received OPTIONS
22:25:17.743 > RTSP received OPTIONS
22:25:28.749 > RTSP received OPTIONS
22:25:39.771 > RTSP received OPTIONS
22:25:50.776 > RTSP received OPTIONS
22:26:01.783 > RTSP received OPTIONS
22:26:12.788 > RTSP received OPTIONS
22:26:23.794 > RTSP received OPTIONS
22:26:34.800 > RTSP received OPTIONS
22:26:45.820 > RTSP received OPTIONS
22:26:56.825 > RTSP received OPTIONS
22:27:07.830 > RTSP received OPTIONS
22:27:18.837 > RTSP received OPTIONS
22:27:30.850 > RTSP received OPTIONS
22:27:41.855 > RTSP received OPTIONS
22:27:50.864 > [429317][D][WiFiClient.cpp:546] connected(): Disconnected: RES: 0, ERR: 128
22:27:50.889 > client has closed the socket
22:27:50.889 > client closed socket, exiting
22:27:50.889 > closing TCP socket
22:27:50.889 > closing UDP socket
22:27:50.889 > closing UDP socket
...

```
I have tried reducing resolution and fps. It seems to help somehow, but it fails nonetheless. 

The main difference I see is that the ESP32 that works reports 4MB PSRAM, while the failing ones 2MB (although it should be 4MB according to the chip on board). CPU model is also different.

Works:
```
ESP32
SDK Version:
v4.4.5
CPU model:
ESP32-D0WDQ6 rev. 1
CPU speed:
240 Mhz
CPU cores:
2
RAM size:
229.93 KB
PSRAM size:
4 MB
Flash size:
4 MB

```
Does not work:

```
SDK Version:
v4.4.5
CPU model:
ESP32-D0WD-V3 rev. 3
CPU speed:
240 Mhz
CPU cores:
2
RAM size:
229.98 KB
PSRAM size:
2 MB
Flash size:
4 MB

```

All ESP32s are very close to the wifi router, so it's not a signal power issue.

Any clues? Have you seen a similar behavior before?

rzeldent, 2023-09-11:
Hi Jose,

The CPU should be fine there are no differences apart the form factor.
The main difference is the PSRAM size. Maybe if the image size > 2Mb (pixels) things go wrong...
There is an option not to use PSRAM, this can be tested. Maybe the API assumes 4Mb?

I myself have experienced issues with single core (when buying dual core) boards or 150 instead of 240Mhz boards.
Maybe something similar?

Rene

bandaangosta, 2023-09-14:
Hi, rzeldent. Thanks for your reply. I tried with disabling the PSRAM, but the problem remains.
I ordered a couple of more units of the ESP32 that works and will try again.  Will comment again in a few weeks.

rzeldent, 2024-02-06:
There are now easy configurable json definitions so you can fiddle/tweak the settings. This is now available in the develop branch.
Maybe this works better for you...

bandaangosta, 2024-02-11:
Thanks, @rzeldent! It does work better and very stable with Onvier app as an RTSP camera.
I will close the issue. Just one question, did you remove support for the "flash" onboard LED?

closed 2024-02-11 by bandaangosta

rzeldent, 2024-02-11:
Hi,

Thanks for testing! The led might have dropped out when moving to separate json files.
You can try to add the led if you know the gpio, Let me know what works!