rzeldent/esp32cam-rtsp #95: Where to enable AP?

eSportler89, 2023-10-12:
Hey,
I followed the instructions in the README.md, however, no AP shows up for me.
Also a fixed setting of the wlan access data in the settings.h file does not work.

I have no idea what the problem could be....

Attached is the output of the monitor task:

```
 *  Executing task: platformio device monitor --environment esp32cam 

/Users/esportler/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
--- forcing DTR inactive
--- forcing RTS inactive

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Logging an output to /Users/esportler/Downloads/esp32cam-rtsp-1.0.5/logs/device-monitor-231012-154132.log
--- Terminal on /dev/cu.usbserial-110 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
15:41:32.311 > ets Jul 29 2019 12:21:46
15:41:32.311 > 
15:41:32.311 > rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
15:41:32.311 > configsip: 0, SPIWP:0xee
15:41:32.311 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:41:32.311 > mode:DIO, clock div:2
15:41:32.311 > load:0x3fff0030,len:1184
15:41:32.311 > load:0x40078000,len:13232
15:41:32.311 > load:0x40080400,len:3028
15:41:32.311 > entry 0x400805e4
15:41:32.311 > [     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...a�␞
15:41:32.311 > [     4][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
15:41:32.311 > [    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
15:41:32.716 > [   463][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
15:41:32.726 > [   473][V][e[   474][I][main.cpp:395] setup(): CPU Freq: 240 Mhz
15:41:32.731 > [   474][I][main.cpp:396] setup(): Free heap: 215092 bytes
15:41:32.734 > [   475][I][main.cpp:397] setup(): SDK version: v4.4.5
15:41:32.740 > [   478][I][main.cpp:398] setup(): Starting ESP32CAM-RTSP...
15:41:32.745 > Config version: 1.5
15:41:32.748 > Config size: 455
15:41:32.748 > Wrong config version. Applying defaults.
15:41:32.754 > [iwcAll]
15:41:32.754 > |-- [iwcSys]
15:41:32.754 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
15:41:32.759 > |   |-- 'iwcApPassword' with value: <hidden>
15:41:32.765 > |   |-- [iwcWifi0]
15:41:32.765 > |   |   |-- 'iwcWifiSsid' with value: ''
15:41:32.768 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
15:41:32.773 > |   \-- 'iwcApTimeout' with value: '30'
15:41:32.776 > |-- [iwcCustom]
15:41:32.779 > |   |-- [board]
15:41:32.779 > |   |   \-- 'bt' with value: 'AI THINKER'
15:41:32.784 > |   |-- [camera]
15:41:32.784 > |   |   |-- 'fd' with value: '100'
15:41:32.787 > |   |   |-- 'fs' with value: 'SVGA (800x600)'
15:41:32.793 > |   |   |-- 'q' with value: '12'
15:41:32.795 > |   |   |-- 'eps' with value: '0'
15:41:32.798 > |   |   |-- 'fb' with value: '1'
15:41:32.801 > |   |   |-- 'b' with value: '0'
15:41:32.804 > |   |   |-- 'c' with value: '0'
15:41:32.806 > |   |   |-- 's' with value: '0'
15:41:32.809 > |   |   |-- 'e' with value: 'Normal'
15:41:32.812 > |   |   |-- 'wb' with value: '1'
15:41:32.815 > |   |   |-- 'awbg' with value: '1'
15:41:32.821 > |   |   |-- 'wbm' with value: 'Auto'
15:41:32.823 > |   |   |-- 'ec' with value: '1'
15:41:32.826 > |   |   |-- 'aec2' with value: '1'
15:41:32.829 > |   |   |-- 'ael' with value: '0'
15:41:32.831 > |   |   |-- 'aecv' with value: '300'
15:41:32.834 > |   |   |-- 'gc' with value: '1'
15:41:32.837 > |   |   |-- 'agcg' with value: '0'
15:41:32.840 > |   |   |-- 'gcl' with value: '2X'
15:41:32.845 > |   |   |-- 'bpc' with value: '0'
15:41:32.848 > |   |   |-- 'wpc' with value: '1'
15:41:32.851 > |   |   |-- 'rg' with value: '1'
15:41:32.854 > |   |   |-- 'lenc' with value: '1'
15:41:32.856 > |   |   |-- 'hm' with value: '0'
15:41:32.859 > |   |   |-- 'vm' with value: '0'
15:41:32.862 > |   |   |-- 'dcw' with value: '1'
15:41:32.865 > |   |   \-- 'cb' with value: '0'
15:41:32.868 > |   \-- [io]
15:41:32.870 > |       \-- 'li' with value: '0'
15:41:32.873 > \-- [hidden]
15:41:32.873 > [   616][V][main.cpp:278] initialize_camera(): initialize_camera
15:41:32.879 > [   617][I][main.cpp:279] initialize_camera(): Camera config: AI THINKER
15:41:32.887 > [   623][I][main.cpp:285] initialize_camera(): Frame size: SVGA (800x600)
15:41:32.893 > [   630][I][main.cpp:287] initialize_camera(): JPEG quality: 12
15:41:32.898 > [   635][I][main.cpp:288] initialize_camera(): Frame duration: 100 ms
15:41:32.904 > [   642][I][main.cpp:292] initialize_camera(): Enable PSRAM: 0
15:41:32.909 > [   647][I][main.cpp:293] initialize_camera(): Frame buffers: 1
15:41:32.915 > [   653][I][main.cpp:304] initialize_camera(): PSRAM disabled
15:41:33.152 > AP password was not set in configuration
15:41:33.155 > State changing from: 0 to 1
15:41:33.162 > E (2105) wifi:Expected to init 4 rx buffer, actual is 3
15:41:33.165 > [   909][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 257
15:41:33.171 > Setting up AP: ESP32CAM-RTSP
15:41:33.174 > With default password: <hidden>
15:41:33.177 > E (2122) wifi:Expected to init 4 rx buffer, actual is 0
15:41:33.182 > [   926][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 257
15:41:33.188 > [   927][E][WiFiAP.cpp:154] softAP(): enable AP first!
15:41:33.193 > AP IP address: 0.0.0.0
15:41:33.193 > AP password was not set.
15:41:33.196 > WiFi SSID was not set.
15:41:33.199 > Will stay in AP mode.
15:41:33.204 > State changed from: 0 to 1
```

rzeldent, 2023-10-12:
Hi eSportler89,

Maybe the configuration is not completely wiped. Try "erase flash" from the PlatformIO menu and then program again...

eSportler89, 2023-10-13:
Hey @rzeldent ,

already tried flashing yesterday with Arduino IDE, webesp etc.
Right now I tried it with Platformio within VSCode but without success :/

Output of the whole process following up (Erase, Build, Upload, Monitor)

Erase:
`|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
Looking for serial port...
Auto-detected: /dev/cu.usbserial-110
Erasing...
esptool.py v4.5.1
Serial port /dev/cu.usbserial-110
Connecting.......
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a0:b7:65:4e:4f:bc
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.3s
Hard resetting via RTS pin...
================================== [SUCCESS] Took 14.42 seconds ==================================
 *  Terminal will be reused by tasks, press any key to close it. `

Build:
` *  Executing task: platformio run 

/Users/esportler/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
--------------------------------------------------------------------------------------------------
/Users/esportler/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (6.4.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
Compiling .pio/build/esp32cam/src/main.cpp.o
Building .pio/build/esp32cam/bootloader.bin
Generating partitions .pio/build/esp32cam/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio/build/esp32cam/lib11f/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32cam/lib11f/WiFi/WiFiAP.cpp.o
Compiling .pio/build/esp32cam/lib11f/WiFi/WiFiClient.cpp.o
[... 312 lines cut ...]
07:06:12.753 > |   \-- [io]
07:06:12.756 > |       \-- 'li' with value: '0'
07:06:12.759 > \-- [hidden]
07:06:12.759 > [   616][V][main.cpp:278] initialize_camera(): initialize_camera
07:06:12.765 > [   617][I][main.cpp:279] initialize_camera(): Camera config: AI THINKER
07:06:12.773 > [   623][I][main.cpp:285] initialize_camera(): Frame size: SVGA (800x600)
07:06:12.778 > [   630][I][main.cpp:287] initialize_camera(): JPEG quality: 12
07:06:12.784 > [   635][I][main.cpp:288] initialize_camera(): Frame duration: 100 ms
07:06:12.789 > [   642][I][main.cpp:292] initialize_camera(): Enable PSRAM: 0
07:06:12.795 > [   647][I][main.cpp:293] initialize_camera(): Frame buffers: 1
07:06:12.801 > [   653][I][main.cpp:304] initialize_camera(): PSRAM disabled
07:06:13.039 > AP password was not set in configuration
07:06:13.041 > State changing from: 0 to 1
07:06:13.046 > E (2105) wifi:Expected to init 4 rx buffer, actual is 3
07:06:13.049 > [   909][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 257
07:06:13.058 > Setting up AP: ESP32CAM-RTSP
07:06:13.058 > With default password: <hidden>
07:06:13.063 > E (2122) wifi:Expected to init 4 rx buffer, actual is 0
07:06:13.066 > [   926][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 257
07:06:13.074 > [   927][E][WiFiAP.cpp:154] softAP(): enable AP first!
07:06:13.077 > AP IP address: 0.0.0.0
07:06:13.080 > AP password was not set.
07:06:13.083 > WiFi SSID was not set.
07:06:13.085 > Will stay in AP mode.
07:06:13.085 > State changed from: 0 to 1`

rzeldent, 2023-10-15:
AH, I think I understand the question; you need to configure the AP it makes a connection to.
To do this, by default is behaves like an Access Point (AP). You make a connection to the cam, selecting it from the WiFi AP's and configure it there! TLDR

eSportler89, 2023-10-15:
Hey, unfortunately not correctly.
I would like to enter the access data for my access point (normal house WLAN), but I can not get to the "Setup" access point generated by ESP.
There is simply no accesspoint visible from ESP. No SSID, also no hidden SSID.

eSportler89, 2023-10-15:
Unfortunately still not properly understood ^^
I have flashed your program on the ESP and after the restart I see no access point from the ESP.
I can not connect to the ESP (except serial console).

rzeldent, 2023-10-15:
Do you have a version with antenna  without having it connected?
Broken?
Are you using 2.4G (does not work on 5G) Wifi?

eSportler89, 2023-10-15:
Neither.
Version without external antennas and works fine with other images.
My laptop searches for both 2,4ghz and 5ghz wlans, but does not find any from the esp, respectively from your image.
With other images the setup AP works.

rzeldent, 2023-10-15:
Will retry setup here with an AI-Thinker

rzeldent, 2023-10-15:
My output:

21:26:58.663 > et.R�n��␂�0�␓��:&��7��HH��i0�LB␂=]ER�}IESQ␕�I��t���3 �*��F�*�e1�SH%=�T�C��n�Z֥�� 0��AIW���5
21:26:58.663 > ��}�r�'��0)}�rv'�A0��E�v���0,�.����:0�0  ��d����i��d�␗����C�둕:DR�       ␚��c�"�v:�H�녑:����03�ɕn:�L�*
21:26:58.680 > 녑:�Ѱ�0�lYK�3�NfH��ad'�Q0����0�+�i3�&j
21:26:58.687 > e�␗� 0�0�e�C�[     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
21:26:59.114 > ;␂␂␂��mYum]��M��ٕɹcpp:87] WebServer(): WebServer::Webserver(port=80)
21:26:59.114 > [    11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
21:26:59.565 > [   463][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
21:26:59.591 > [   483][V][[   484][I][main.cpp:403] setup(): CPU Freq: 240 Mhz, 2 core(s)
21:26:59.597 > [   485][I][main.cpp:404] setup(): Free heap: 249076 bytes
21:26:59.598 > [   485][I][main.cpp:405] setup(): SDK version: v4.4.4
21:26:59.598 > [   489][I][main.cpp:406] setup(): Starting ESP32CAM-RTSP...
21:26:59.598 > [   494][V][main.cpp:411] setup(): PSRAM found and initialized
21:26:59.616 > Config version: 1.5
21:26:59.619 > Config size: 455
21:26:59.622 > Wrong config version. Applying defaults.
21:26:59.624 > [iwcAll]
21:26:59.624 > |-- [iwcSys]
21:26:59.624 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-4023a7f23a08'
21:26:59.634 > |   |-- 'iwcApPassword' with value: <hidden>
21:26:59.636 > |   |-- [iwcWifi0]
21:26:59.638 > |   |   |-- 'iwcWifiSsid' with value: ''
21:26:59.641 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
21:26:59.646 > |   \-- 'iwcApTimeout' with value: '30'
21:26:59.649 > |-- [iwcCustom]
21:26:59.651 > |   |-- [board]
21:26:59.651 > |   |   \-- 'bt' with value: 'AI THINKER'
21:26:59.651 > |   |-- [camera]
21:26:59.651 > |   |   |-- 'fd' with value: '200'
21:26:59.651 > |   |   |-- 'fs' with value: 'VGA (640x480)'
21:26:59.663 > |   |   |-- 'q' with value: '14'
21:26:59.663 > |   |   |-- 'eps' with value: '0'
21:26:59.663 > |   |   |-- 'fb' with value: '1'
21:26:59.663 > |   |   |-- 'b' with value: '0'
21:26:59.663 > |   |   |-- 'c' with value: '0'
21:26:59.680 > |   |   |-- 's' with value: '0'
21:26:59.680 > |   |   |-- 'e' with value: 'Normal'
21:26:59.680 > |   |   |-- 'wb' with value: '1'
21:26:59.680 > |   |   |-- 'awbg' with value: '1'
21:26:59.691 > |   |   |-- 'wbm' with value: 'Auto'
21:26:59.691 > |   |   |-- 'ec' with value: '1'
21:26:59.697 > |   |   |-- 'aec2' with value: '1'
21:26:59.697 > |   |   |-- 'ael' with value: '0'
21:26:59.697 > |   |   |-- 'aecv' with value: '300'
21:26:59.697 > |   |   |-- 'gc' with value: '1'
21:26:59.697 > |   |   |-- 'agcg' with value: '0'
21:26:59.714 > |   |   |-- 'gcl' with value: '2X'
21:26:59.714 > |   |   |-- 'bpc' with value: '0'
21:26:59.714 > |   |   |-- 'wpc' with value: '1'
21:26:59.714 > |   |   |-- 'rg' with value: '1'
21:26:59.714 > |   |   |-- 'lenc' with value: '1'
21:26:59.730 > |   |   |-- 'hm' with value: '0'
21:26:59.731 > |   |   |-- 'vm' with value: '0'
21:26:59.731 > |   |   |-- 'dcw' with value: '1'
21:26:59.738 > |   |   \-- 'cb' with value: '0'
21:26:59.741 > |   \-- [io]
21:26:59.741 > |       \-- 'li' with value: '0'
21:26:59.744 > \-- [hidden]
21:26:59.747 > [   632][V][main.cpp:289] initialize_camera(): initialize_camera
21:26:59.752 > [   634][I][main.cpp:290] initialize_camera(): Camera config: AI THINKER
21:26:59.756 > [   641][I][main.cpp:296] initialize_camera(): Frame size: VGA (640x480)
21:26:59.764 > [   647][I][main.cpp:298] initialize_camera(): JPEG quality: 14
21:26:59.771 > [   653][I][main.cpp:299] initialize_camera(): Frame duration: 200 ms
21:26:59.777 > [   659][I][main.cpp:303] initialize_camera(): Enable PSRAM: 0
21:26:59.783 > [   665][I][main.cpp:304] initialize_camera(): Frame buffers: 1
21:26:59.785 > [   670][I][main.cpp:315] initialize_camera(): PSRAM disabled
21:26:59.999 > [   904][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 4 - Using Channel 15, Value = 0
21:27:00.015 > AP password was not set in configuration
21:27:00.015 > State changing from: 0 to 1
21:27:00.033 > [   925][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
21:27:00.114 > Setting up AP: ESP32CAM-RTSP-4023a7f23a08
21:27:00.114 > With default password: <hidden>
21:27:00.129 > [  1015][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
21:27:00.130 > AP IP address: [  1021][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
21:27:00.130 > [  1023][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
21:27:00.148 > 192.[  1031][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
21:27:00.153 > [  1031][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_STOP
21:27:00.162 > [  1044][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
21:27:00.168 > 168.4.1
21:27:00.170 > AP password was not set.
21:27:00.170 > WiFi SSID was not set.
21:27:00.170 > Will stay in AP mode.
21:27:00.170 > State changed from: 0 to 1

The ESP32 is visible as an AP.

What's different is the line in your output:

You have a newer version of the SDK (4.4.5) and no PSRAM?

07:06:13.063 > E (2122) wifi:Expected to init 4 rx buffer, actual is 0
07:06:13.066 > [   926][E][WiFiGeneric.cpp:680] wifiLowLevelInit(): esp_wifi_init 257

rzeldent, 2023-10-15:
Hi, I updated the platform version to 6.4.0 and the SDK versions match. I ran the application again:

```
22:20:10.844 > �e�A
22:20:10.844 > �%��5ي��ٹv�'��␚��␕���ٶ������W�����Eɶ'�A�I��.␗¸¬���␖�y�a���ߖ��a�i�ߖ���a0␃A��e␇$�Q ,�ͻ�嶧j����'�M��A�&b�Y�ʱҍ!녤'�i�ڸ°��nG�2�*
22:20:10.844 > ��d␇�q�␓��[��&��Ca�Kɹ��␓������[     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
22:20:10.844 > [␀␘␂␂␂��mYum]��M��ٕɹcpp:87] WebServer(): WebServer::Webserver(port=80)
22:20:10.844 > [    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
�ʹ␗�E�10u�E��}�Ej�U�b�K�m ��je��P�%�O�i␕�T�I␕�T�)%%�_�Q꥕���)������␖��r��k�l�XV�eW�J�␋␂� �������ҕSH�
22:20:21.815 > �.W�x��UI��%␕SQU�)���␇�œ��␕�FP*�e1���␅��U��c�˥g.␋i��
22:20:21.821 > �]��eC�␌�}��W��A�)}Ѳ���A��}d��������_��W�x�b��%ɶ␇������V�i�A�␃���'T�ʂc�k�A�:W�*��␋�i�M��␖��9K���S�l␋�i�Q�����,�␙�ʓ��j
22:20:21.834 > �+�:��0�������i�Ҙ#!���y��Q�°͖*�[     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
22:20:22.267 > [␀␘␂␂␂��mYum]��M��ٕɹcpp:87] WebServer(): WebServer::Webserver(port=80)
22:20:22.273 > [    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
22:20:22.710 > [   463][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
22:20:22.738 > [   480][V][e[   481][I][main.cpp:403] setup(): CPU Freq: 240 Mhz, 2 core(s)
22:20:22.743 > [   481][I][main.cpp:404] setup(): Free heap: 214876 bytes
22:20:22.747 > [   482][I][main.cpp:405] setup(): SDK version: v4.4.5
22:20:22.747 > [   486][I][main.cpp:406] setup(): Starting ESP32CAM-RTSP...
22:20:22.747 > [   491][V][main.cpp:411] setup(): PSRAM found and initialized
22:20:22.766 > Config version: 1.5
22:20:22.766 > Config size: 455
22:20:22.769 > Wrong config version. Applying defaults.
22:20:22.770 > [iwcAll]
22:20:22.770 > |-- [iwcSys]
22:20:22.770 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-4023a7f23a08'
22:20:22.779 > |   |-- 'iwcApPassword' with value: <hidden>
22:20:22.782 > |   |-- [iwcWifi0]
22:20:22.784 > |   |   |-- 'iwcWifiSsid' with value: ''
22:20:22.784 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
22:20:22.784 > |   \-- 'iwcApTimeout' with value: '30'
22:20:22.784 > |-- [iwcCustom]
22:20:22.799 > |   |-- [board]
22:20:22.799 > |   |   \-- 'bt' with value: 'AI THINKER'
22:20:22.802 > |   |-- [camera]
22:20:22.802 > |   |   |-- 'fd' with value: '200'
22:20:22.802 > |   |   |-- 'fs' with value: 'VGA (640x480)'
22:20:22.802 > |   |   |-- 'q' with value: '14'
22:20:22.815 > |   |   |-- 'eps' with value: '0'
22:20:22.819 > |   |   |-- 'fb' with value: '1'
22:20:22.819 > |   |   |-- 'b' with value: '0'
22:20:22.819 > |   |   |-- 'c' with value: '0'
22:20:22.819 > |   |   |-- 's' with value: '0'
22:20:22.830 > |   |   |-- 'e' with value: 'Normal'
22:20:22.832 > |   |   |-- 'wb' with value: '1'
22:20:22.835 > |   |   |-- 'awbg' with value: '1'
22:20:22.837 > |   |   |-- 'wbm' with value: 'Auto'
22:20:22.837 > |   |   |-- 'ec' with value: '1'
22:20:22.846 > |   |   |-- 'aec2' with value: '1'
22:20:22.849 > |   |   |-- 'ael' with value: '0'
22:20:22.851 > |   |   |-- 'aecv' with value: '300'
22:20:22.851 > |   |   |-- 'gc' with value: '1'
22:20:22.851 > |   |   |-- 'agcg' with value: '0'
22:20:22.860 > |   |   |-- 'gcl' with value: '2X'
22:20:22.863 > |   |   |-- 'bpc' with value: '0'
22:20:22.865 > |   |   |-- 'wpc' with value: '1'
22:20:22.868 > |   |   |-- 'rg' with value: '1'
22:20:22.868 > |   |   |-- 'lenc' with value: '1'
22:20:22.877 > |   |   |-- 'hm' with value: '0'
22:20:22.879 > |   |   |-- 'vm' with value: '0'
22:20:22.882 > |   |   |-- 'dcw' with value: '1'
22:20:22.884 > |   |   \-- 'cb' with value: '0'
22:20:22.884 > |   \-- [io]
22:20:22.884 > |       \-- 'li' with value: '0'
22:20:22.884 > \-- [hidden]
22:20:22.884 > [   629][V][main.cpp:289] initialize_camera(): initialize_camera
22:20:22.899 > [   631][I][main.cpp:290] initialize_camera(): Camera config: AI THINKER
22:20:22.901 > [   638][I][main.cpp:296] initialize_camera(): Frame size: VGA (640x480)
22:20:22.901 > [   644][I][main.cpp:298] initialize_camera(): JPEG quality: 14
22:20:22.917 > [   650][I][main.cpp:299] initialize_camera(): Frame duration: 200 ms
[... 12 lines cut ...]
22:20:23.296 > [  1038][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
22:20:23.296 > [  1042][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 11 - AP_STOP
22:20:23.317 > [  1043][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
22:20:23.320 > [  1055][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 10 - AP_START
22:20:23.331 > AP password was not set.
22:20:23.331 > WiFi SSID was not set.
22:20:23.333 > Will stay in AP mode.
22:20:23.334 > State changed from: 0 to 1
```

All wotks fine, I see the AP in my list.

I do not have this error lines:
```
E (2122) wifi:Expected to init 4 rx buffer, actual is 0
```
It might be that memory is missing and this is not needed in other programs. Another hint might be the
```
 491][V][main.cpp:411] setup(): PSRAM found and initialized
```
That's available in my AI thinker ESP cam but lacks in your output.

It might be possible that the PSRAM was not in the module to save money (or faulty modules). You can try to remove those lines to detect/enable the PSRAM. Normally for higher resolutions this ram is required.

Have you already tried another module from another vendor/batch?

eSportler89, 2023-10-16:
PSRAM module is there, but maybe faulty.
I'll buy an other one and give it a new try.

Thanks for your help till now! :)

I'll give feedback when the new esp arrived!

rzeldent, 2023-10-16:
Yes, let me know!

eSportler89, 2023-10-31:
Hey, got two new ESPs and both of them got the same problems.
This seems to be a problem of your script, because AI on the Edge recognize the PSRAM and use it corretly.

```
I (853) quad_psram: This chip is ESP32-D0WD
I (853) esp_psram: Found 8MB PSRAM device
I (853) esp_psram: Speed: 40MHz
I (853) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (860) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
```

rzeldent, 2023-10-31:
Hi Esportler89,

I checked the difference in the code regarding the PSRAM. AI on the edge used a different toolkit: IDF and this esp32-rtsp uses arduino. In the arduino libraries the same initialization is done. This is unlikely to be an issue.
Additionally, I think PSRAM is not used for WiFi but the main RAM.

There are slight differences in the available memory at the start:
Yours: 07:06:12.617 > [   474][I][main.cpp:396] setup(): Free heap: 215092 bytes
Mine: 22:20:22.743 > [   481][I][main.cpp:404] setup(): Free heap: 214876 bytes

This is immediately at startup, nothing has been done. You have even more RAM! Not much but it is a difference.
This looks like there are differences in compiling source code.

I checked the packages and dependencies, they're identical.

After compiling
Yours:
RAM:   [===       ]  26.7% (used 87384 bytes from 327680 bytes)
Flash: [=====     ]  55.0% (used 1080969 bytes from 1966080 bytes)
Mine:
RAM:   [===       ]  26.7% (used 87368 bytes from 327680 bytes)
Flash: [=====     ]  53.6% (used 1052837 bytes from 1966080 bytes)

There are some differences. Your flash is larger....

The error suggests a memory issue. All that I could find: https://github.com/espressif/esp-idf/issues/10209

Besides that, no clue yet...

Anyone?

mmaravillo, 2023-11-13:
Hi @eSportler89,

Yours:
15:41:32.726 > [   473][V][e[   474][I][main.cpp:395] setup(): CPU Freq: 240 Mhz
Mine:
6:24:43.836 > [   471][V][e[   472][I][main.cpp:403] setup(): CPU Freq: 240 Mhz, 2 core(s)

I wonder how outdated your copy of the code is, but it is older than a711eec80c06c324b78348fe5b6feca09ceb9827 from 6 months ago.  Way old and possibly missing bugfixes.

rzeldent, 2024-02-02:
Please pull latest version

eSportler89, 2024-02-13:
Sadly also doens't work for me :/

rzeldent, 2024-02-13:
Please try the develop, there is much going on!