rzeldent/esp32cam-rtsp #96: Build error: Platform IO/VS Code

bjx9, 2023-10-21:
When I try to build or upload the code, I get the following error.  I am using Platform IO on VS Code on Windows 11

Building in release mode
prepare_file([".pio\build\esp32cam\index.min.html.txt.o"], ["html\index.min.html"])
prepare_file([".pio\build\esp32cam\restart.min.html.txt.o"], ["html\restart.min.html"])
*** [.pio\build\esp32cam\index.min.html.txt.o] Bad file descriptor
*** [.pio\build\esp32cam\restart.min.html.txt.o] Bad file descriptor

Any help?

rzeldent, 2023-10-21:
Looks like the python script to minify the html has not ran.
No other errors?

bjx9, 2023-10-21:
Full report below...

Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (6.4.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
prepare_file([".pio\build\esp32cam\index.min.html.txt.o"], ["html\index.min.html"])
prepare_file([".pio\build\esp32cam\restart.min.html.txt.o"], ["html\restart.min.html"])
*** [.pio\build\esp32cam\index.min.html.txt.o] Bad file descriptor
*** [.pio\build\esp32cam\restart.min.html.txt.o] Bad file descriptor
================================================================ [FAILED] Took 3.98 seconds ================================================================

 *  The terminal process "C:\Users\xxx\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

rzeldent, 2023-10-21:
Hi bjx9,

Strange, looks like your configuration is different. Used Windows 11, Platform IO.
I do not have an answer for that but suggest reinstalling...

My output:

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
prepare_file([".pio\build\esp32cam\index.min.html.txt.o"], ["html\index.min.html"])
**Converting .pio\build\esp32cam\index.min.html.txt.o**
prepare_file([".pio\build\esp32cam\restart.min.html.txt.o"], ["html\restart.min.html"])
**Converting .pio\build\esp32cam\restart.min.html.txt.o**
revert_original_file([".pio\build\esp32cam\restart.min.html.txt.o"], ["html\restart.min.html"])
revert_original_file([".pio\build\esp32cam\index.min.html.txt.o"], ["html\index.min.html"])
Compiling .pio\build\esp32cam\src\main.cpp.o
Building .pio\build\esp32cam\bootloader.bin
Generating partitions .pio\build\esp32cam\partitions.bin
esptool.py v4.5.1

rzeldent, 2024-02-02:
Does it works now?

bjx9, 2024-02-06:
Hi,

 

I stopped with the project.

rzeldent, 2024-02-06:
Sad to hear that...

Is your environment up to date?
In the pio command shell, try

```
pio pkg update 
```

Good luck!

closed 2024-02-06 by rzeldent