rzeldent/esp32cam-rtsp #97: Can't boot: "invalid header: 0x400d3f64

unfa, 2023-11-09:
I have been flashing the firmware.bin file many times and all I got was an endless stream of these messages:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x400d3f64
invalid header: 0x400d3f64
invalid header: 0x400d3f64
invalid header: 0x400d3f64
invalid header: 0x400d3f64
invalid header: 0x400d3f64
invalid header: 0x400d3f64
```

I used [ESPhome web tool](https://web.esphome.io/) since that's the easiest way for me to flash an ESP32 module.

Flashing the "prepare for first use" option works fine and I get proper logs from the device afterwards.

I've downloaded the firmware from previous release to test - same thing. I have taken out the microSD card I've put in - same thing.

I had 2 instances where something happened with the while LED - one time it flashed a short sequence, other time it became enabled (on low power, so indicator mode, not torch).

The module I use is a standard ESP32-CAM one with a dedicated MB programmer
I've flashed the firmware generated by ESPHome (inside Home Assistant) and it works first try.

Have anybody had something like this happening before? I couldn't find anything relevant online.

Thank you for creating this project :) I hope I'll be able to use it soon!

rzeldent, 2023-11-09:
Hi unfa,

If you use upload in platformio it will (build and) upload automatically!

unfa, 2023-11-10:
Yes, using PlatformIO it did work.

I did install it on Arch Linux from the official repositories:
```
sudo pacman -Sy platformio
```
After that following your guide was incredibly easy, I just cloned the repo, run two commands using the PlatformIO (`pio`) executable and... it works!

This gives me ideas to maybe use this for maybe modifying the code myself in the future, maybe I'd be able to commit something useful.

One thing in my mind is making the LED turn on based on ambient light, so instead of just increasing camera gain, it could try to increase it less, but also increase the light. But I digress.

Thanks!

I don't know why it doesn't work with ESPHome, but I guess ESPhome web flasher is made for ESPhome-made firmware binaries.

closed 2023-11-10 by unfa

rzeldent, 2023-11-11:
Hi Unfa,

Glad to hear it worked out!

Using the CDS resistor and adapting the backlight to the surrounding light is exactly why it's there!
Be aware that later boards do not have the CDS anymore on the board...

Kind regards,

Rene