rzeldent/esp32cam-rtsp #98: Using rtsp is very laggy, only the jpeg stream is normal

yygcom, 2023-11-11:
Frame rate:
40 ms (25.0 f/s)
Frame size:
VGA (640x480)
JPEG quality:
20 [1-100]
Enable PSRAM:
Enabled
Number of frame buffers:
1
Brightness:
0 [-2,2]
Contrast:
0 [-2,2]
Saturation:
0 [-2,2]
Special effect:
Normal
White balance:
Auto
AWB gain:
Auto
WB mode:
Auto
Exposure control:
Auto
Auto exposure control (dsp):
Enabled
Auto Exposure level:
0
Manual exposure value:
300
Gain control:
Auto
AGC gain:
0
Gain ceiling:
2X
Black pixel correct:
Manual
White pixel correct:
Auto
Gamma correct:
Enabled
Lens correction:
Enabled
Horizontal mirror:
Normal
Vertical flip:
Normal
Downsize enable:
Enabled
Color bar:
Camera

rzeldent, 2023-11-11:
Hi Liu,

Thanks for submitting this issue. This has been noticed in the past.
This is caused because of buffering in the RTSP client (VLC for example).
VLC has options to decrease the buffer so will leg much less...

Hope this helps,

Rene

rzeldent, 2026-08-28:
This is a problem of the WiFi throughput; that's limited to 1MB/s or so.
I made a new RTSP streamer. Currently in develop brach. Maybe this fixes your problem