rzeldent/esp32cam-rtsp #101: Authentication impossible

kapabel, 2023-12-21:
Hello,
I am sure I am using it wrong but I cant get the authentication to work.
When setting up the device I define a AP password.

Later on when I want to change the settings it asks me for a username and password. I have tried all combinations but I can't get through. I have reflashed my ESP32CAM twice to make sure it wasn't a typo.

So whats the username and password to change settings? Is it the AP password I have defined on first setup or something completely different?

Many thanks!
Bernhard

rzeldent, 2023-12-22:
Hi Bernhard,

It is in the manual.
[image]

So admin and the AP Password

kapabel, 2023-12-22:
Ha! There it was. Many thanks!

rzeldent, 2023-12-22:
👍

closed 2023-12-22 by rzeldent