rzeldent/esp32cam-rtsp #102: Trying to compile Seeed Xiao ESP32S3 Sense, get AttributeError

heyitsyang, 2024-01-06:
I am using the following in pio:

[env:seeed_xiao_esp32s3]
platform = https://github.com/platformio/platform-espressif32.git
board = seeed_xiao_esp32s3
build_flags = 
  -D BOARD_SEEED_XIAO_ESP32S3_SENSE

But I get this error.  Any ideas?
```
Resolving seeed_xiao_esp32s3 dependencies...
AttributeError: Traceback (most recent call last):
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 105, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 107, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 149, in _install_project_env_platform
    PlatformPackageManager().install(
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 52, in install
    p = PlatformFactory.new(pkg)
  File "C:\Users\Yang\.platformio\penv\lib\site-packages\platformio\platform\factory.py", line 79, in new
    platform_cls = getattr(
AttributeError: module 'platformio.platform.platform' has no attribute 'PlatformPlatform'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

rzeldent, 2024-01-07:
Hi heyitsyang,

This looks like something goes wrong in the python scripts of platformIO.
Maybe you're using a new prerelease or something. Try to update to the lastest version (or resinstall?)

A difference I see is the platform. I use platform = espressif32

Hope this helps

Rene

heyitsyang, 2024-01-07:
@rzeldent, thanks for your reply.  

If I use platform = espressif32, I get:  fatal error: pins_arduino.h: No such file or directory

Using the upstream platform = https://github.com/platformio/platform-espressif32.git makes the pins_arduino.h error go away, but I get the AttributeError previously shown.  I have tried re-installing PIO with same results.  I can compile all of my old projects with the same PIO setup.

I should mention I had to add seeed_xiao_esp32s3.json to my .platformio/boards directory to get this far.

heyitsyang, 2024-01-07:
I'm not sure what I did, but it caused Intellisense to run and fix my Arduino.h problem.  Everything compiles as expected now.   Thanks for your help.

closed 2024-01-07 by heyitsyang