rzeldent/esp32cam-rtsp #103: Can't compile code on PlatformIO

yanxke, 2024-01-27:
I'm getting the error

```
In file included from src/main.cpp:10:0:
include/camera_config.h:29:20: error: 'CAMERA_FB_IN_DRAM' was not declared in this scope
     .fb_location = CAMERA_FB_IN_DRAM,
                    ^
```

Board is esp32cam

rzeldent, 2024-01-27:
Hi Yanxke,

I cannot find this line. Maybe you need to update the branch (git update)?
The CI is passing: https://github.com/rzeldent/esp32cam-rtsp/actions/runs/7254756448

yanxke, 2024-01-27:
It's right here

https://github.com/rzeldent/esp32cam-rtsp/blob/develop/include/camera_config.h#L29

[image]

yanxke, 2024-01-27:
Oh, I finally figured it out.  I had an old version of the esp32 framework installed on PlatformIO.  After I upgraded that, it compiled okay.

closed 2024-01-27 by yanxke