rzeldent/esp32cam-rtsp #104: Arduino version??

cohen-sydney, 2024-01-28:
Hi I have not mastered Platform IO, any chance that there is an Arduino version??

I am trying to load this onto a M5Stack UnitCam S3 to stream to HomeKit via HomeBrigde FFMPEG.

thanks.

cohen-sydney, 2024-01-28:
I have got the Platform IO working for this but having issues building for the
M5Satck CAMS3 Unit. It builds but doe snot find the camera on running. 
any tips on what to hang in INI or PlatformIO to get it to build??

rzeldent, 2024-01-28:
Hi cohen,

Use the develop version. 
I do not own such a camera but take a look at the include/camera_config.h. There are the pins defined to connect to the camera. Something might be wrong there...

cohen-sydney, 2024-01-28:
I did use the Develop version. ill look into the Camera config for the pins.
thanks

rzeldent, 2024-01-28:
Yes, good luck, the develop version is an attempt to get it also running on other platforms. This was required for the seeed_xiao_esp32s3 that uses the esp32s3. Not everything has been tested so help is very welcome.
Make a PR if you have suggestions!

cohen-sydney, 2024-01-28:
will do nice code by the way.

cohen-sydney, 2024-01-28:
so I use this Env
default_envs = seeed_xiao_esp32s3

it compiles and loads for my M5camS3 Unit, I can get the config page and log into my wifi, but as you say the Camera Config might be incorrect.

I looked at the pins and the M5 docs

https://docs.m5stack.com/en/unit/Unit-CamS3
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/unit/OV2640DS_en.pdf
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/unit/Unit-CamS3/ESP32-S3-WROOM-1-N16R8.pdf

but I really have no idea what to change on this:-
any help??? would be appreciated, thanks 
constexpr camera_config_t xiao_esp32s3_camera_settings = {
    .pin_pwdn = -1,
    .pin_reset = -1,
    .pin_xclk = 10,
    .pin_sscb_sda = 40,
    .pin_sscb_scl = 39,
    .pin_d7 = 48,
    .pin_d6 = 11,
    .pin_d5 = 12,
    .pin_d4 = 14,
    .pin_d3 = 16,
    .pin_d2 = 18,
    .pin_d1 = 17,
    .pin_d0 = 15,
    .pin_vsync = 38,
    .pin_href = 47,
    .pin_pclk = 13,
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,
    .pixel_format = PIXFORMAT_JPEG, // for streaming
    .frame_size = FRAMESIZE_UXGA,
    .jpeg_quality = 12,
    .fb_count = 2,
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_LATEST};

rzeldent, 2024-01-28:
Hiconstexpr camera_config_t m5stack_camera_settings = {
I added the settings. Just do a pull and they should be there if you build the m5stack

cohen-sydney, 2024-01-28:
Thanks. Just to confirm I am using the Env constexpr camera_config_t xiao_esp32s3_camera_settings = {
as this is a ESP32S3 the only one that works for my device. 
If you change the other one, will it help? I downloaded the new file and it did not help??

cohen-sydney, 2024-01-28:
I built using the default_envs = m5stack_esp32cam in the Platformio . ini 
but got this error 
A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
only the default_envs = seeed_xiao_esp32s3 is esp32s3 the tip my camera has. thanks

rzeldent, 2024-01-28:
Maybe board is off... I think this is a new version and needs to be added
What do you use for the board?

cohen-sydney, 2024-01-28:
It's all one device. Board and camera together. If I understand your question.

I want to use it as home security camera. It's small.

rzeldent, 2024-01-28:
Added it to the project, development branch, just uncomment in the platformio.ini the board...

cohen-sydney, 2024-01-28:
thanks ill try it out in the morning.
appreciate it.

rzeldent, 2024-01-28:
No guarantees, but maybe your fiddling will make it work?

cohen-sydney, 2024-01-28:
Thanks for your help. I get a panic after loading.
I set the ini file to "default_envs = m5stack_unitcams3"
it complies and transfers OK. I set the device target to esp32s3. 
this is the panic from the Serial Monitor

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4209ed52
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Config version: 1.6
Config size: 271
Loading configurations
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-3882503bdaec'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'BuzzIoT'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   \-- [camera]
|       |-- 'fd' with value: '200'
|       |-- 'fs' with value: 'XGA (1024x768)'
|       |-- 'q' with value: '10'
|       |-- 'b' with value: '0'
|       |-- 'c' with value: '0'
|       |-- 's' with value: '0'
|       |-- 'e' with value: 'Normal'
|       |-- 'wb' with value: '1'
|       |-- 'awbg' with value: '1'
|       |-- 'wbm' with value: 'Auto'
|       |-- 'ec' with value: '1'
|       |-- 'aec2' with value: '1'
|       |-- 'ael' with value: '0'
|       |-- 'aecv' with value: '300'
|       |-- 'gc' with value: '1'
|       |-- 'agcg' with value: '0'
|       |-- 'gcl' with value: '2X'
|       |-- 'bpc' with value: '0'
|       |-- 'wpc' with value: '1'
|       |-- 'rg' with value: '1'
|       |-- 'lenc' with value: '1'
|       |-- 'hm' with value: '0'
|       |-- 'vm' with value: '0'
|       |-- 'dcw' with value: '1'
|       \-- 'cb' with value: '0'
\-- [hidden]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4204ab86  PS      : 0x00060830  A0      : 0x82043cc4  A1      : 0x3fcebcf0  
A2      : 0x3fcab4a0  A3      : 0x3fc9cc80  A4      : 0xffff8fff  A5      : 0x3c0c38c8  
A6      : 0x00001000  A7      : 0x3c0c3920  A8      : 0x00000000  A9      : 0x3fcebcd0  
A10     : 0x00000080  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3fcab4fc  
A14     : 0x00000008  A15     : 0x00000005  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4204ab83:0x3fcebcf0 0x42043cc1:0x3fcebd40 0x42043812:0x3fcebd70 0x4201299d:0x3fcebdc0 0x42005a66:0x3fcebde0 0x42005ea8:0x3fcebee0 0x420160e2:0x3fcebf70

ELF 

end of error.
[... 69 lines cut ...]
Writing at 0x000ab799... (67 %)
Writing at 0x000b0c2e... (70 %)
Writing at 0x000b6068... (72 %)
Writing at 0x000bb7ed... (75 %)
Writing at 0x000c1294... (78 %)
Writing at 0x000c683c... (81 %)
Writing at 0x000cbcea... (83 %)
Writing at 0x000d4931... (86 %)
Writing at 0x000dc918... (89 %)
Writing at 0x000e2aa4... (91 %)
Writing at 0x000e8047... (94 %)
Writing at 0x000ed811... (97 %)
Writing at 0x000f362a... (100 %)
Wrote 935488 bytes (592595 compressed) at 0x00010000 in 6.8 seconds (effective 1100.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================== [SUCCESS] Took 12.59 seconds ======================================================

Environment        Status    Duration
-----------------  --------  ------------
m5stack_unitcams3  SUCCESS   00:00:12.590
====================================================== 1 succeeded in 00:00:12.590 ======================================================
 *  Terminal will be reused by tasks, press any key to close it.

rzeldent, 2024-01-28:
Could you add the exception decoder in the platformio.ini?
https://community.platformio.org/t/how-do-i-use-esp32-exception-decoder/22789/2

cohen-sydney, 2024-01-28:
will do.

cohen-sydney, 2024-01-28:
Its in the INI file already

line 36 of the ini 
monitor_filters = log2file, time, default, esp32_exception_decoder

cohen-sydney, 2024-01-28:
I get the following Build warning.
/Users/andrew/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c: In function 'uartSetPins':
/Users/andrew/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
/Users/andrew/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:149:6: note: declared here
 bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)

cohen-sydney, 2024-01-28:
[env:m5stack_unitcams3]
board = esp32s3box
build_flags = 
  -DBOARD_M5STACK_UNITCAMS3
  -DBOARD_HAS_PSRAM
  -DCORE_DEBUG_LEVEL=3

tried to set a debug level not sure where it prints out?

cohen-sydney, 2024-01-28:
Here is the latest Paid message with little more detail.
Just after this command it fails,

[   235][I][main.cpp:232] initialize_camera(): Frame duration: 200 ms

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4209f93a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   205][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   224][I][main.cpp:311] setup(): CPU Freq: 240 Mhz, 2 core(s)
[   225][I][main.cpp:312] setup(): Free heap: 289836 bytes
[   225][I][main.cpp:313] setup(): SDK version: v4.4.6-dirty
[   226][I][main.cpp:314] setup(): Board: M5Stack UnitCamS3
[   226][I][main.cpp:315] setup(): Starting ESP32CAM-RTSP...
Config version: 1.6
Config size: 271
Loading configurations
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-3882503bdaec'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'BuzzIoT'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   \-- [camera]
|       |-- 'fd' with value: '200'
|       |-- 'fs' with value: 'XGA (1024x768)'
|       |-- 'q' with value: '10'
|       |-- 'b' with value: '0'
|       |-- 'c' with value: '0'
|       |-- 's' with value: '0'
|       |-- 'e' with value: 'Normal'
|       |-- 'wb' with value: '1'
|       |-- 'awbg' with value: '1'
|       |-- 'wbm' with value: 'Auto'
|       |-- 'ec' with value: '1'
|       |-- 'aec2' with value: '1'
|       |-- 'ael' with value: '0'
|       |-- 'aecv' with value: '300'
|       |-- 'gc' with value: '1'
|       |-- 'agcg' with value: '0'
|       |-- 'gcl' with value: '2X'
|       |-- 'bpc' with value: '0'
|       |-- 'wpc' with value: '1'
|       |-- 'rg' with value: '1'
|       |-- 'lenc' with value: '1'
|       |-- 'hm' with value: '0'
|       |-- 'vm' with value: '0'
|       |-- 'dcw' with value: '1'
|       \-- 'cb' with value: '0'
\-- [hidden]
[   234][I][main.cpp:229] initialize_camera(): Frame size: XGA (1024x768)
[   234][I][main.cpp:231] initialize_camera(): JPEG quality: 10
[   235][I][main.cpp:232] initialize_camera(): Frame duration: 200 ms
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4204b676  PS      : 0x00060830  A0      : 0x820447b4  A1      : 0x3fcebce0  
A2      : 0x3fcab4a0  A3      : 0x3fc9cc80  A4      : 0xffff8fff  A5      : 0x3c0c42d0  
A6      : 0x00001000  A7      : 0x3c0c4328  A8      : 0x00000000  A9      : 0x3fcebcc0  
A10     : 0x00000080  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3fcab4fc  
A14     : 0x00000008  A15     : 0x00000005  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4204b673:0x3fcebce0 0x420447b1:0x3fcebd30 0x42044302:0x3fcebd60 0x420131dd:0x3fcebdb0 0x42005cda:0x3fcebdd0 0x4200628c:0x3fcebee0 0x42016b5e:0x3fcebf70

cohen-sydney, 2024-01-29:
Panic with Verbose Debug on.

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420a1792
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   196][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[   196][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
[   209][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   228][I][main.cpp:311] setup(): CPU Freq: 240 Mhz, 2 core(s)
[   229][I][main.cpp:312] setup(): Free heap: 289836 bytes
[   230][I][main.cpp:313] setup(): SDK version: v4.4.6-dirty
[   230][I][main.cpp:314] setup(): Board: M5Stack UnitCamS3
[   231][I][main.cpp:315] setup(): Starting ESP32CAM-RTSP...
Config version: 1.6
Config size: 271
Loading configurations
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-3882503bdaec'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'BuzzIoT'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   \-- [camera]
|       |-- 'fd' with value: '200'
|       |-- 'fs' with value: 'XGA (1024x768)'
|       |-- 'q' with value: '10'
|       |-- 'b' with value: '0'
|       |-- 'c' with value: '0'
|       |-- 's' with value: '0'
|       |-- 'e' with value: 'Normal'
|       |-- 'wb' with value: '1'
|       |-- 'awbg' with value: '1'
|       |-- 'wbm' with value: 'Auto'
|       |-- 'ec' with value: '1'
|       |-- 'aec2' with value: '1'
|       |-- 'ael' with value: '0'
|       |-- 'aecv' with value: '300'
|       |-- 'gc' with value: '1'
|       |-- 'agcg' with value: '0'
|       |-- 'gcl' with value: '2X'
|       |-- 'bpc' with value: '0'
|       |-- 'wpc' with value: '1'
|       |-- 'rg' with value: '1'
|       |-- 'lenc' with value: '1'
|       |-- 'hm' with value: '0'
|       |-- 'vm' with value: '0'
|       |-- 'dcw' with value: '1'
|       \-- 'cb' with value: '0'
\-- [hidden]
[   238][V][main.cpp:224] initialize_camera(): initialize_camera
[   239][I][main.cpp:229] initialize_camera(): Frame size: XGA (1024x768)
[   239][I][main.cpp:231] initialize_camera(): JPEG quality: 10
[   240][I][main.cpp:232] initialize_camera(): Frame duration: 200 ms
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4204d476  PS      : 0x00060e30  A0      : 0x820465b4  A1      : 0x3fcebce0  
A2      : 0x3fcab4a0  A3      : 0x3fc9cc80  A4      : 0xffff8fff  A5      : 0x3c0c55cc  
A6      : 0x00001000  A7      : 0x3c0c5624  A8      : 0x00000000  A9      : 0x3fcebcc0  
A10     : 0x00000080  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3fcab4fc  
A14     : 0x00000008  A15     : 0x00000005  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4204d473:0x3fcebce0 0x420465b1:0x3fcebd30 0x42046102:0x3fcebd60 0x42014ef9:0x3fcebdb0 0x4200608e:0x3fcebdd0 0x420066f0:0x3fcebee0 0x4201891a:0x3fcebf70

rzeldent, 2024-01-31:
Hi cohen,

You need to add the line for the exception decoder. This will make the backtrace readable..

cohen-sydney, 2024-01-31:
can you advise what to add and where? is that in ini file or somewhere else?
sorry never done this/
thanks

rzeldent, 2024-01-31:
Add the line:

```
monitor_filters = esp32_exception_decoder
```

cohen-sydney, 2024-01-31:
Its in the INI file already

line 36 of the ini
monitor_filters = log2file, time, default, esp32_exception_decoder

only thing did not do is.

set build type to debug instead of release by adding build_type = debug to your platformio.ini (https://community.platformio.org/t/cannot-get-esp32-exception-decoder-to-work/37588)

rzeldent, 2024-01-31:
I think it only works if the other filters are not present...

cohen-sydney, 2024-02-01:
Tried to Build with Debug. this is what is see in the Monitor- I note the wrong config version note sure what that related too.

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420a49aa
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Config version: 1.6
Config size: 271
Wrong config version. Applying defaults.
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-3882503bdaec'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: ''
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   \-- [camera]
|       |-- 'fd' with value: '200'
|       |-- 'fs' with value: 'VGA (640x480)'
|       |-- 'q' with value: '14'
|       |-- 'b' with value: '0'
|       |-- 'c' with value: '0'
|       |-- 's' with value: '0'
|       |-- 'e' with value: 'Normal'
|       |-- 'wb' with value: '1'
|       |-- 'awbg' with value: '1'
|       |-- 'wbm' with value: 'Auto'
|       |-- 'ec' with value: '1'
|       |-- 'aec2' with value: '1'
|       |-- 'ael' with value: '0'
|       |-- 'aecv' with value: '300'
|       |-- 'gc' with value: '1'
|       |-- 'agcg' with value: '0'
|       |-- 'gcl' with value: '2X'
|       |-- 'bpc' with value: '0'
|       |-- 'wpc' with value: '1'
|       |-- 'rg' with value: '1'
|       |-- 'lenc' with value: '1'
|       |-- 'hm' with value: '0'
|       |-- 'vm' with value: '0'
|       |-- 'dcw' with value: '1'
|       \-- 'cb' with value: '0'
\-- [hidden]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42050196  PS      : 0x00060830  A0      : 0x820492d4  A1      : 0x3fcebb20  
A2      : 0x3fcab4a0  A3      : 0x3fc9cc80  A4      : 0xffff8fff  A5      : 0x3c0c1bc4  
A6      : 0x00001000  A7      : 0x3c0c1c1c  A8      : 0x00000000  A9      : 0x3fcebb00  
A10     : 0x00000080  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3fcab4fc  
A14     : 0x00000008  A15     : 0x00000005  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42050193:0x3fcebb20 0x420492d1:0x3fcebb70 0x42048e22:0x3fcebba0 0x42017755:0x3fcebbf0 0x420093f6:0x3fcebc10 0x42009874:0x3fcebd10 0x4201b582:0x3fcebd90

rzeldent, 2024-02-05:
Hi cohen,

Updated the develop branch with new types: https://github.com/rzeldent/esp32cam-rtsp/tree/develop
Configuration has been taken out to seaparate files.
M5STACK_UNITCAMS3 has been added but untested.

cohen-sydney, 2024-02-05:
I'll give it a try later today. thanks for the hard work on this, it's a very helpful code.

cohen-sydney, 2024-02-05:
I ran the new version but using the Definition for "default_envs = esp32cam_m5stack_unitcams3" it has panic and reboot.
Interestingly I tried "#default_envs = esp32cam_seeed_xiao_esp32s3_sense" this does run successfully but the camera does not work. why?? and does this help to fix the unitcam3??
thanks

cohen-sydney, 2024-02-06:
the following file of ESP32CAM_M5STACK_UNITCAMS3, does not crash but the camera does not work. the Webpage is available via 192.168.4.1 with a camera error.
Here is the json file.

{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_8MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "'-D ESP32CAM_M5STACK_UNITCAMS3'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D LED_GPIO=14'",
      "'-D CAMERA_CONFIG_PIN_PWDN=GPIO_NUM_NC'",
      "'-D CAMERA_CONFIG_PIN_RESET=GPIO_NUM_NC'",
      "'-D CAMERA_CONFIG_PIN_XCLK=15'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SDA=25'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SCL=23'",
      "'-D CAMERA_CONFIG_PIN_Y9=16'",
      "'-D CAMERA_CONFIG_PIN_Y8=17'",
      "'-D CAMERA_CONFIG_PIN_Y7=18'",
      "'-D CAMERA_CONFIG_PIN_Y6=12'",
      "'-D CAMERA_CONFIG_PIN_Y5=10'",
      "'-D CAMERA_CONFIG_PIN_Y4=8'",
      "'-D CAMERA_CONFIG_PIN_Y3=9'",
      "'-D CAMERA_CONFIG_PIN_Y2=11'",
      "'-D CAMERA_CONFIG_PIN_VSYNC=6'",
      "'-D CAMERA_CONFIG_PIN_HREF=7'",
      "'-D CAMERA_CONFIG_PIN_PCLK=13'",
      "'-D CAMERA_CONFIG_CLK_FREQ_HZ=20000000'",
      "'-D CAMERA_CONFIG_LEDC_TIMER=LEDC_TIMER_0'",
      "'-D CAMERA_CONFIG_LEDC_CHANNEL=LEDC_CHANNEL_0'",
      "'-D CAMERA_CONFIG_FB_COUNT=2'",
      "'-D CAMERA_CONFIG_FB_LOCATION=CAMERA_FB_IN_DRAM'",
      "'-D I2C_MEMS_SDA=48'",
      "'-D I2C_MEMS_SCL=47'"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x2886",
        "0x0056"
      ],
      [
        "0x2886",
        "0x8056"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "bluetooth",
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "M5STACK UNITCAMS3",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.m5stack.com/en/unit/Unit-CamS3",
  "vendor": "M5STACK"
}

cohen-sydney, 2024-02-06:
Here is my latest try as the Json file, it runs but the camera does not start. also included the camera Schematics. see the U2 MAPPINGS FOR THE Y CODES. I think the error is there.

{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_8MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "'-D ESP32CAM_M5STACK_UNITCAMS3'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D LED_GPIO=14'",
      "'-D CAMERA_CONFIG_PIN_PWDN=GPIO_NUM_NC'",
      "'-D CAMERA_CONFIG_PIN_RESET=GPIO_NUM_NC'",
      "'-D CAMERA_CONFIG_PIN_XCLK=11'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SDA=25'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SCL=23'",
      "'-D CAMERA_CONFIG_PIN_Y9=13'",
      "'-D CAMERA_CONFIG_PIN_Y8=4'",
      "'-D CAMERA_CONFIG_PIN_Y7=10'",
      "'-D CAMERA_CONFIG_PIN_Y6=12'",
      "'-D CAMERA_CONFIG_PIN_Y5=7'",
      "'-D CAMERA_CONFIG_PIN_Y4=16'",
      "'-D CAMERA_CONFIG_PIN_Y3=15'",
      "'-D CAMERA_CONFIG_PIN_Y2=6'",
      "'-D CAMERA_CONFIG_PIN_VSYNC=42'",
      "'-D CAMERA_CONFIG_PIN_HREF=18'",
      "'-D CAMERA_CONFIG_PIN_PCLK=12'",
      "'-D CAMERA_CONFIG_CLK_FREQ_HZ=20000000'",
      "'-D CAMERA_CONFIG_LEDC_TIMER=LEDC_TIMER_0'",
      "'-D CAMERA_CONFIG_LEDC_CHANNEL=LEDC_CHANNEL_0'",
      "'-D CAMERA_CONFIG_FB_COUNT=2'",
      "'-D CAMERA_CONFIG_FB_LOCATION=CAMERA_FB_IN_DRAM'",
      "'-D I2C_MEMS_SDA=48'",
      "'-D I2C_MEMS_SCL=47'"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x2886",
        "0x0056"
      ],
      [
        "0x2886",
        "0x8056"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "bluetooth",
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "M5STACK UNITCAMS3",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.m5stack.com/en/unit/Unit-CamS3",
  "vendor": "M5STACK"
}

[image]

cohen-sydney, 2024-02-06:
Thanks I’ll try it later. Is the Flash Size correct at 8 ?

cohen-sydney, 2024-02-07:
Unfortunately its still erroring and rebooting.
here is the output on the Monitor.

ELF file SHA256: 947182b7746a3913

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420a0f4e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   196][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[   196][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
[   209][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5228][I][main.cpp:335] setup(): Core debug level: 5
[  5229][I][main.cpp:336] setup(): CPU Freq: 240 Mhz, 2 core(s)
[  5230][I][main.cpp:337] setup(): Free heap: 289572 bytes
[  5230][I][main.cpp:338] setup(): SDK version: v4.4.6-dirty
[  5231][I][main.cpp:339] setup(): Board: esp32cam_m5stack_unitcams3
[  5231][I][main.cpp:340] setup(): Starting ESP32CAM-RTSP...
Config version: 1.6
Config size: 333
Loading configurations
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-3882503bdaec'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'BuzzIoT'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   \-- [camera]
|       |-- 'fd' with value: '200'
|       |-- 'fs' with value: 'XGA (1024x768)'
|       |-- 'q' with value: '10'
|       |-- 'b' with value: '0'
|       |-- 'c' with value: '0'
|       |-- 's' with value: '0'
|       |-- 'e' with value: 'Normal'
|       |-- 'wb' with value: '1'
|       |-- 'awbg' with value: '1'
|       |-- 'wbm' with value: 'Auto'
|       |-- 'ec' with value: '1'
|       |-- 'aec2' with value: '1'
|       |-- 'ael' with value: '0'
|       |-- 'aecv' with value: '300'
|       |-- 'gc' with value: '1'
|       |-- 'agcg' with value: '0'
|       |-- 'gcl' with value: '2X'
|       |-- 'bpc' with value: '0'
|       |-- 'wpc' with value: '1'
|       |-- 'rg' with value: '1'
|       |-- 'lenc' with value: '1'
|       |-- 'hm' with value: '0'
|       |-- 'vm' with value: '0'
|       |-- 'dcw' with value: '1'
|       \-- 'cb' with value: '0'
\-- [hidden]
[  5239][V][main.cpp:214] initialize_camera(): initialize_camera
[  5239][I][main.cpp:217] initialize_camera(): Frame size: XGA (1024x768)
[  5240][I][main.cpp:219] initialize_camera(): JPEG quality: 10
[  5241][I][main.cpp:221] initialize_camera(): Frame duration: 200 ms
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4204d3c5  PS      : 0x00060830  A0      : 0x8204653c  A1      : 0x3fcebc70  
A2      : 0x3fcab638  A3      : 0x3fc9d520  A4      : 0xffff8fff  A5      : 0x3c0c57f8  
A6      : 0x0000005c  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3fcebc90  
A10     : 0x3fcab638  A11     : 0x00000000  A12     : 0x8037dd28  A13     : 0x3fc98350  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4204d3c2:0x3fcebc70 0x42046539:0x3fcebcc0 0x4204608a:0x3fcebcf0 0x42014b4d:0x3fcebd40 0x42005e96:0x3fcebd60 0x42006510:0x3fcebe70 0x42017922:0x3fcebee0

rzeldent, 2024-02-07:
With the information I can find these seem to be the right mappings. Unfortunately I do not know exactly what the device is; and if these pins are correct. The sensor might also be different than the specs.
You can try to try some different pin mappings but I need more information...

sheedje, 2024-02-15:
I got the cam working!!
There are the pin definitions from the json file (esp32cam_m5stack_unitcams3.json) of the main branch

`    "extra_flags": [
      "'-D ESP32CAM_M5STACK_UNITCAMS3'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D USER_LED_GPIO=14'",
      "'-D USER_LED_ON_LEVEL=HIGH'",
      "'-D CAMERA_CONFIG_PIN_PWDN=GPIO_NUM_NC'",
      "'-D CAMERA_CONFIG_PIN_RESET=21'",
      "'-D CAMERA_CONFIG_PIN_XCLK=11'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SDA=17'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SCL=41'",
      "'-D CAMERA_CONFIG_PIN_Y9=13'",
      "'-D CAMERA_CONFIG_PIN_Y8=4'",
      "'-D CAMERA_CONFIG_PIN_Y7=10'",
      "'-D CAMERA_CONFIG_PIN_Y6=5'",
      "'-D CAMERA_CONFIG_PIN_Y5=7'",
      "'-D CAMERA_CONFIG_PIN_Y4=16'",
      "'-D CAMERA_CONFIG_PIN_Y3=15'",
      "'-D CAMERA_CONFIG_PIN_Y2=6'",
      "'-D CAMERA_CONFIG_PIN_VSYNC=42'",
      "'-D CAMERA_CONFIG_PIN_HREF=18'",
      "'-D CAMERA_CONFIG_PIN_PCLK=12'",
      "'-D CAMERA_CONFIG_CLK_FREQ_HZ=20000000'",
      "'-D CAMERA_CONFIG_LEDC_TIMER=LEDC_TIMER_0'",
      "'-D CAMERA_CONFIG_LEDC_CHANNEL=LEDC_CHANNEL_0'",
      "'-D CAMERA_CONFIG_FB_COUNT=2'",
      "'-D CAMERA_CONFIG_FB_LOCATION=CAMERA_FB_IN_DRAM'",
      "'-D I2C_MEMS_SDA=17'",
      "'-D I2C_MEMS_SCL=41'",
      "'-D TF_CS=9'",
      "'-D TF_MOSI=38'",
      "'-D TF_CLK=39'",
      "'-D TF_MISO=40'"
    ],`

cohen-sydney, 2024-02-15:
That's great news I'll give it a go. thanks for sharing.

cohen-sydney, 2024-02-15:
@sheedje well done, can confirm that got it. 
It's a little temperamental but it works as RTSP. thanks for the help here.

virgileMaon, 2024-02-25:
@sheedje thank you very mutch, good job.
@rzeldent do you back port it in develop branch?

Iondream, 2024-03-17:
odd. I'm still getting panics with this config if I try to stream via web or rtsp. 

`17:43:09.316 > [ 47691][V][Parsing.cpp:254] _parseArguments(): args: 
17:43:09.317 > [ 47692][V][Parsing.cpp:237] _parseRequest(): Request: /stream
17:43:09.317 > [ 47692][V][Parsing.cpp:238] _parseRequest():  Arguments: 
17:43:09.318 > [ 47693][V][main.cpp:185] handle_stream(): handle_stream
17:43:09.319 > [ 47694][V][main.cpp:192] handle_stream(): starting streaming
17:43:17.576 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
17:43:17.576 > 
17:43:17.576 > Core  1 register dump:
17:43:17.576 > PC      : 0x42014b37  PS      : 0x00060b30  A0      : 0x820052a0  A1      : 0x3fcebca0  
17:43:17.576 > A2      : 0x00000000  A3      : 0x3c0b2dec  A4      : 0x0000000b  A5      : 0x000000c0  
17:43:17.576 > A6      : 0x3c0b3733  A7      : 0x3fceba80  A8      : 0x82014b34  A9      : 0x3fcebc80  
17:43:17.576 > A10     : 0x3fc9d520  A11     : 0x3c0b2dec  A12     : 0x3fcebc74  A13     : 0x00000033  
17:43:17.576 > A14     : 0x3fcebcd4  A15     : 0x00080000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
17:43:17.576 > EXCVADDR: 0x00000004  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
17:43:17.576 > 
17:43:17.576 > 
17:43:17.576 > Backtrace: 0x42014b34:0x3fcebca0 0x4200529d:0x3fcebcc0 0x4209f5a5:0x3fcebd30 0x4200f462:0x3fcebd50 0x4200f509:0x3fcebd70 0x4200f5e6:0x3fcebdb0 0x4200f809:0x3fcebe20 0x4209fc41:0x3fcebe80 0x4201411b:0x3fcebea0 0x420066f6:0x3fcebec0 0x42017945:0x3fcebee0`

this is my config

`    "core": "esp32",
    "extra_flags": [
      "'-D ESP32CAM_M5STACK_UNITCAMS3'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D USER_LED_GPIO=14'",
      "'-D USER_LED_ON_LEVEL=HIGH'",
      "'-D CAMERA_CONFIG_PIN_PWDN=GPIO_NUM_NC'",
      "'-D CAMERA_CONFIG_PIN_RESET=21'",
      "'-D CAMERA_CONFIG_PIN_XCLK=11'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SDA=17'",
      "'-D CAMERA_CONFIG_PIN_SCCB_SCL=41'",
      "'-D CAMERA_CONFIG_PIN_Y9=13'",
      "'-D CAMERA_CONFIG_PIN_Y8=4'",
      "'-D CAMERA_CONFIG_PIN_Y7=10'",
      "'-D CAMERA_CONFIG_PIN_Y6=5'",
      "'-D CAMERA_CONFIG_PIN_Y5=7'",
      "'-D CAMERA_CONFIG_PIN_Y4=16'",
      "'-D CAMERA_CONFIG_PIN_Y3=15'",
      "'-D CAMERA_CONFIG_PIN_Y2=6'",
      "'-D CAMERA_CONFIG_PIN_VSYNC=42'",
      "'-D CAMERA_CONFIG_PIN_HREF=18'",
      "'-D CAMERA_CONFIG_PIN_PCLK=21'",
      "'-D CAMERA_CONFIG_CLK_FREQ_HZ=20000000'",
      "'-D CAMERA_CONFIG_LEDC_TIMER=LEDC_TIMER_0'",
      "'-D CAMERA_CONFIG_LEDC_CHANNEL=LEDC_CHANNEL_0'",
      "'-D CAMERA_CONFIG_FB_COUNT=2'",
      "'-D CAMERA_CONFIG_FB_LOCATION=CAMERA_FB_IN_DRAM'",
      "'-D SCCB_I2C_PORT=I2C_NUM_0'",
      "'-D I2C_MEMS_SDA=17'",
      "'-D I2C_MEMS_SCL=41'",
      "'-D TF_CS=9'",
      "'-D TF_MOSI=38'",
      "'-D TF_CLK=39'",
      "'-D TF_MISO=40'"
    ]`

Am I doing something wrong?

rzeldent, 2024-03-19:
Updated the settings in develop: fe7195c23272c4f733441381207cc02d51f0e6eb

Thanks guys!

theschles, 2024-03-20:
FYI ran `pio run -e esp32cam_ai_thinker` on MacOS 14 with PlatformIO Core, version 6.1.13 on commit:

```
commit 312d916647b22a233b61d232fe2d2a196940a301 (HEAD -> main, origin/main, origin/HEAD)
Author: Rene Zeldenthuis <rzeldent@outlook.com>
Date:   Sun Mar 10 12:46:48 2024 +0100
```

[These warnings came up -- same as @cohen-sydney reported previously in this issue thread](https://github.com/rzeldent/esp32cam-rtsp/issues/104#issuecomment-1913746293):
```
/Users/foo/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c: In function 'uartSetPins':
/Users/foo/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
/Users/foo/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:149:6: note: declared here
 bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
      ^~~~~~~~~~~
```

rzeldent, 2024-03-20:
I enabled the warnings in the build flags (-Wall) to explicitly have all the warnings.
So these warnings are "normal" and in the framework...

theschles, 2024-03-21:
> I enabled the warnings in the build flags (-Wall) to explicitly have all the warnings. So these warnings are "normal" and in the framework...

Absolutely a good thing to do. 

FYI [comments on the same issue in the espressif / arduino-esp32 repo](https://github.com/espressif/arduino-esp32/issues/9047#issuecomment-1872369241) stated:

> PR https://github.com/espressif/arduino-esp32/pull/8644
> 
> Fixed in https://github.com/espressif/arduino-esp32/tree/release/v2.x

As shown above, I ran `pio pkg update -g -p espressif32` and it stated I'm fully up-to-date.

Is there another command I should be running to get the version of `framework-arduinoespressif32` that won't throw this compiler warning?