rzeldent/esp32cam-rtsp #110: Error compiler

djsaveRUS, 2024-02-18:
I do as in the instructions on the command line and get an unsatisfactory result. Is this really supposed to work?
[image]

djsaveRUS, 2024-02-18:
[image]

rzeldent, 2024-02-18:
Yes, it does work.

I suspect you forgot to include the boards submodule when cloning the repo.

djsaveRUS, 2024-02-18:
It is possible that the connection of the board item is not included in the instructions.

rzeldent, 2024-02-18:
you need to checkout with submodules

djsaveRUS, 2024-02-18:
How do I do this?

rzeldent, 2024-02-18:
Hi,

See the comment in the demo:

git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo.git

djsaveRUS, 2024-02-18:
[image]
[image]
[image]
[image]

What did I do wrong?

rzeldent, 2024-02-18:
I don't know exactly; the documentation might be a but outdated but I retested it using a PlatformIO CLI prompt and, of course, it worked on my computer; see output below.
The remark I made about including submodules was incorrect; the project does not use any submodules yet; However, this is something that happens often with another project.

You can also take a look at the git workflow used to build the project; https://github.com/rzeldent/esp32cam-rtsp/blob/main/.github/workflows/main.yml.

Normally I use it from Visual Studio Code and load the directory. Maybe this gives some additional information.
Do you have Python installed and python3-venv?

```
Compiling .pio/build/esp32cam_ttgo_t_journal/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32cam_ttgo_t_journal/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32cam_ttgo_t_journal/libFrameworkArduino.a
Indexing .pio/build/esp32cam_ttgo_t_journal/libFrameworkArduino.a
Linking .pio/build/esp32cam_ttgo_t_journal/firmware.elf
Retrieving maximum program size .pio/build/esp32cam_ttgo_t_journal/firmware.elf
Checking size .pio/build/esp32cam_ttgo_t_journal/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  26.0% (used 85132 bytes from 327680 bytes)
Flash: [=====     ]  51.6% (used 1014069 bytes from 1966080 bytes)
Building .pio/build/esp32cam_ttgo_t_journal/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 26 ELF sections
Successfully created esp32 image.
================================================================ [SUCCESS] Took 7.66 seconds ================================================================

Environment                            Status    Duration
-------------------------------------  --------  ------------
esp32cam_ai_thinker                    SUCCESS   00:00:01.473
esp32cam_espressif_esp_eye             SUCCESS   00:00:01.419
esp32cam_espressif_esp32s2_cam_board   SUCCESS   00:00:01.458
esp32cam_espressif_esp32s2_cam_header  SUCCESS   00:00:01.415
esp32cam_espressif_esp32s3_cam_lcd     SUCCESS   00:00:01.504
esp32cam_espressif_esp32s3_eye         SUCCESS   00:00:04.619
esp32cam_freenove_wrover_kit           SUCCESS   00:00:08.178
esp32cam_m5stack_camera_psram          SUCCESS   00:00:07.725
esp32cam_m5stack_camera                SUCCESS   00:00:07.728
esp32cam_m5stack_esp32cam              SUCCESS   00:00:07.692
esp32cam_m5stack_unitcam               SUCCESS   00:00:07.749
esp32cam_m5stack_unitcams3             SUCCESS   00:00:08.132
esp32cam_m5stack_wide                  SUCCESS   00:00:07.716
esp32cam_seeed_xiao_esp32s3_sense      SUCCESS   00:00:07.747
esp32cam_ttgo_t_camera                 SUCCESS   00:00:07.702
esp32cam_ttgo_t_journal                SUCCESS   00:00:07.661
=============================================================== 16 succeeded in 00:01:29.919 ===============================================================
```

JDepto, 2024-02-19:
Check which version of Espressif 32 you have installed (PIO Home -> Platforms).   I had version 3.x something installed and was getting the same errors about the GPIO_NUM_NC and such.  I was able to get it upgraded to 6.5.0 and everything is compiling now.

jeffehobbs, 2024-03-10:
I had the same issue:

[image]

...but then I updated the Espressif32 libraries as per the advice [here](https://community.platformio.org/t/updating-espressif32-version/34596/2)...

[image]

...much better result. I'd suggest you add a step to update those libraries:

`pio pkg update -g -p espressif32`

... before the `pio run` step.

rzeldent, 2024-03-10:
Have updated the documentation for this. Thanks for the feedback!

closed 2024-06-14 by rzeldent