rzeldent/esp32cam-rtsp #117: restart and flash missing?

kvikindi, 2024-03-17:
I might be taking crazy pills here, but going by the commit history, it seems that all references to the LED flash and restart api have been removed in versions since at least 1.0.6?

Also I'm seeing references to the "version 2" branch under releases, but there's none to be found?

Please advise, as reality is already kind of funky these days 😄

ulab, 2024-03-19:
I'd think the "version 2" branch is "develop", but I agree, this is confusing.

rzeldent, 2024-04-16:
Hi Ragnar,

No, you did not take any crazy pills! Version 2 was created to support more devices. This unfortunately required a common base and LED flash was removed. Restart was removed because this was only available after settings were changed and (if the type of the board is correct), restarting was no longer required.

jsnyder, 2024-04-18:
On the Seeed ESP32S3 Sense the settings don't seem to take effect without a restart though (at least resolution doesn't) or am I missing something?

acongdon, 2024-07-30:
Same on AI Thinker, change resolution settings OK but doesn't reboot to apply them...  up the ladder again...

jeremysherriff, 2024-12-24:
Confirming on current build of `development` - with the device plugged into my PC and running vscode/platform.io I can see the settings get saved by no trigger to reload or reset (AI Thinker).  @rzeldent can we please have `/restart` back or a fix to correctly trigger the settings reload?

closed 2026-09-04 by github-actions[bot]