rzeldent/esp32cam-rtsp #118: ESP32-CAM MB Error 261

thorstensd, 2024-03-23:
Hello,

I have just stumbled across this error and cannot explain it. I have already successfully installed an ESP32-S before. However, I have no idea what could be the problem with this ESP32-CAM-MB.

Failed to initialize the camera!
Result: 261 (ESP_ERR_NOT_FOUND)
Please check hardware or correct the camera settings and restart.

Upload Log:
esp32cam_espressif_esp_eye             SUCCESS   00:00:26.547
esp32cam_espressif_esp32s2_cam_board   FAILED    00:00:08.169
esp32cam_espressif_esp32s2_cam_header  FAILED    00:00:08.211
esp32cam_espressif_esp32s3_cam_lcd     FAILED    00:00:10.270
esp32cam_espressif_esp32s3_eye         FAILED    00:00:12.324
esp32cam_freenove_wrover_kit           SUCCESS   00:00:26.824
esp32cam_m5stack_camera_psram          SUCCESS   00:00:27.230
esp32cam_m5stack_camera                SUCCESS   00:00:26.396
esp32cam_m5stack_esp32cam              SUCCESS   00:00:26.992
esp32cam_m5stack_unitcam               SUCCESS   00:00:26.737
esp32cam_m5stack_unitcams3             FAILED    00:00:10.760
esp32cam_m5stack_wide                  SUCCESS   00:00:26.226
esp32cam_seeed_xiao_esp32s3_sense      FAILED    00:00:10.864
esp32cam_ttgo_t_camera                 SUCCESS   00:00:26.513
esp32cam_ttgo_t_journal                SUCCESS   00:00:26.535

Did somebody have a Idea how to fix it?

BR, Thorsten

rzeldent, 2024-03-23:
All the firmwares are built. You nned to install the correct one for your device.
Looking at the logging above, I suspect the last one is uploaded (esp32cam_ttgo_t_journal SUCCESS 00:00:26.535)
From the PIO menu select the device your're using...

thorstensd, 2024-03-24:
Ok, I will check. I´m not very familiar with this, so sorry for my questions. I need no to learn how to do it with visual code and how to find out which ESP I have. Its only written ESP32S and the ESP32MB Board.

Thank you so far.

rzeldent, 2024-03-24:
Take a look at the documentation, the pictures to know what board you have...
Then, in VIsual studio select in the PlatformIO the board you're using and use that to only build or upload the board you're using...

Hope this helps.

[image]

closed 2024-03-24 by thorstensd

rzeldent, 2024-03-24:
Glad you managed to solve this and indeed, using the command line is maybe even easier!

Enjoy!